home-assistant/operating-system #771: Odroid N2 - No Boot

fbaccretio, 2020-07-14:
Odroid N2 with Emmc does not boot. (no blue light)
I flashed the eMMC with the usb writer from hardkernel.

According to hardkernel the boot issue has nothing to do with the hidden boot sectors of the emmcs:
[All of Amlogic SoC on the C1/C2/N2/C4 boards don't use the hidden partition for the booting process.
](https://forum.odroid.com/viewtopic.php?f=181&p=299102#p299102)

```bl2_stage_init 0x81
hw id: 0x0000 - pwm id 0x01
bl2_stage_init 0xc1
bl2_stage_init 0x02

L0:00000000
L1:00000703
L2:0000c067
L3:14000020
B2:00402000
B1:e0f83180

TE: 196399

BL2 Built : 06:17:13, Jun 28 2019. g12b gf0505d7-dirty - qi.duan@droid13

Board ID = 4
Set A53 clk to 24M
Set A73 clk to 24M
Set clk81 to 24M
A53 clk: 1200 MHz
A73 clk: 1200 MHz
CLK81: 166.6M
smccc: 00034763
eMMC boot @ 0
sw8 s
DDR driver_vesion: LPDDR4_PHY_V_0_1_14 build time: Jun 28 2019 06:17:09
board id: 4
Load FIP HDR from eMMC, src: 0x00010200, des: 0xfffd0000, size: 0x00004000, part                                                                                                                                                             : 0
fw parse done
Load ddrfw from eMMC, src: 0x00054200, des: 0xfffd0000, size: 0x0000c000, part:                                                                                                                                                              0
Load ddrfw from eMMC, src: 0x0002c200, des: 0xfffd0000, size: 0x00004000, part:                                                                                                                                                              0
PIEI prepare done
fastboot data load
00000000
emmc switch 1 ok
00000000
emmc switch 2 ok
fastboot data verify
verify result: 255
Cfg max: 2, cur: 1. Board id: 255. Force loop cfg
DDR4 probe
ddr clk to 1320MHz
Load ddrfw from eMMC, src: 0x00014200, des: 0xfffd0000, size: 0x0000c000, part:                                                                                                                                                              0
00000000
emmc switch 0 ok
Check phy result
INFO : End of initialization
INFO : End of read enable training
INFO : End of fine write leveling
INFO : End of read dq deskew training
INFO : End of MPR read delay center optimization
INFO : End of Write leveling coarse delay
INFO : End of write delay center optimization
INFO : End of read delay center optimization
INFO : End of max read latency training
INFO : Training has run successfully!
1D training succeed
Load ddrfw from eMMC, src: 0x00020200, des: 0xfffd0000, size: 0x0000c000, part:                                                                                                                                                              0
Check phy result
INFO : End of initialization
INFO : End of 2D read delay Voltage center optimization
INFO : End of 2D write delay Voltage center optimization
INFO : Training has run successfully!

R0_RxClkDly_Margin==94 ps 8
R0_TxDqDly_Margi==106 ps 9

R1_RxClkDly_Margin==0 ps 0
R1_TxDqDly_Margi==0 ps 0

 dwc_ddrphy_apb_wr((0<<20)|(2<<16)|(0<<12)|(0xb0):0001
2D training succeed
auto size-- 65535DDR cs0 size: 2048MB
DDR cs1 size: 2048MB
DMC_DDR_CTRL: 00600024DDR size: 3928MB
cs0 DataBus test pass
cs1 DataBus test pass
cs0 AddrBus test pass
cs1 AddrBus test pass
 pre test  bdlr_100_average==450 bdlr_100_min==450 bdlr_100_max==450 bdlr_100_cu                                                                                                                                                             r==450
 aft test  bdlr_100_average==450 bdlr_100_min==450 bdlr_100_max==450 bdlr_100_cu                                                                                                                                                             r==450
non-sec scramble use zero key
ddr scramble enabled

100bdlr_step_size ps== 450
result report
boot times 0Enable ddr reg access
00000000
emmc switch 3 ok
Authentication key not yet programmed
get rpmb counter error 0x00000007
00000000
emmc switch 0 ok
Load FIP HDR from eMMC, src: 0x00010200, des: 0x01700000, size: 0x00004000, part                                                                                                                                                             : 0
Load BL3X from eMMC, src: 0x00060200, des: 0x01750000, size: 0x0007b400, part: 0
0.0;M3 CHK:0;cm4_sp_mode 0
E30HDR
MVN_1=0x00000000
MVN_2=0x00000000
[Image: g12b_v1.1.3375-8f9c8a7 2019-01-24 10:44:46 guotai.shen@droid11-sz]
OPS=0x40
ring efuse init
chipver efuse init
29 0a 40 00 01 17 13 00 00 01 38 34 48 56 4d 50
[0.019858 Inits done]
secure task start!
high task start!
low task start!
run into bl31
NOTICE:  BL31: v1.3(release):ab8811b
NOTICE:  BL31: Built : 15:03:31, Feb 12 2019
NOTICE:  BL31: G12A normal boot!
NOTICE:  BL31: BL33 decompress pass
ERROR:   Error initializing runtime service opteed_fast

U-Boot 2020.01 (Jul 03 2020 - 10:20:01 +0000) odroid-n2

DRAM:  3.8 GiB
MMC:   sd@ffe05000: 0, mmc@ffe07000: 1
In:    serial@3000
Out:   serial@3000
Err:   serial@3000
Net:   eth0: ethernet@ff3f0000
Card did not respond to voltage select!
unable to select a mode
switch to partitions #0, OK
mmc1(part 0) is current device
** No partition table - mmc 1 **
MMC Device 2 not found
no mmc device at slot 2
starting USB...
Bus usb@ff500000: Register 3000140 NbrPorts 3
Starting the controller
USB XHCI 1.10
scanning bus usb@ff500000 for devices... 1 USB Device(s) found
usb - USB sub-system

Usage:
usb start - start (scan) USB controller
usb reset - reset (rescan) USB controller
usb stop [f] - stop USB [f]=force stop
usb tree - show USB device tree
usb info [dev] - show available USB devices
usb test [dev] [port] [mode] - set USB 2.0 test mode
    (specify port 0 to indicate the device's upstream port)
    Available modes: J, K, S[E0_NAK], P[acket], F[orce_Enable]

ethernet@ff3f0000 Waiting for PHY auto negotiation to complete```

_Originally posted by @fbaccretio in https://github.com/home-assistant/operating-system/issues/675#issuecomment-657421772_

fbaccretio, 2020-07-14:
What i did so far:
- running etcher as administrator, 
- tried older versions of etcher (other threads mentioned this as a solution)
- I used Win32DiskImager

I also:
- bought a new emmcs module writer
- bought a new odroid n2 and emmc (32gb N2 linux/android version)
- bought a new Computer
- bought a new Notebook 

;-)

Willybits, 2020-07-14:
Strange, I am using the n2 as well and booting has never been a problem.
Is the switch in the MMC position?

fbaccretio, 2020-07-14:
Switch is to MMC Position

[image]

agners, 2020-07-14:
@fbaccretio it indeed looks like the issue #675. From what I can tell the boot ROM was able to load the boot loader/U-Boot, so that part is fine. But it seems that the partition table is not right, so I would guess a flashing issue.

The instructions say the eMMC should be flashed like a regular SD card:
https://github.com/home-assistant/operating-system/blob/dev/Documentation/boards/hardkernel/odroid-n2.md

Did you try the instructions from https://www.home-assistant.io/hassio/installation/, e.g. using balenaEtcher?

fbaccretio, 2020-07-14:
@agners 

Yes I flashed exacly like a regular SD card with balenaEtcher. (Also with Win32DiskImager)

When I flash the hassio image directly to the sd-card booting works.

I also flashed the original ubuntu image of hardkernel. (just to check the flashing process to emmc)
Also with this image there is absolutely no issue and the odroid is booting...
So i think it has something to do with hassio.

pepeEL, 2020-07-20:
I use eMMC on my Odroid N2 and flash as standard microSD but if i remember stwitch is setup to SPI.... I can check today.

fbaccretio, 2020-07-22:
@pepeEL 
If you wan't to boot from Emmc or SD-Card the mini switch should be to emmc position i guess.
It should only be in SPI Position if you want to boot from the internal SPI flash memory. 

https://wiki.odroid.com/odroid-n2/software/boot_sequence

agners, 2020-07-23:
@fbaccretio yes the switch should be in MMC position to boot from eMMC.

For me too 4.11 seems to just boot fine from a eMMC. This is the boot log for a working boot:

```
...
NOTICE:  BL31: v1.3(release):ab8811b
NOTICE:  BL31: Built : 15:03:31, Feb 12 2019
NOTICE:  BL31: G12A normal boot!
NOTICE:  BL31: BL33 decompress pass
ERROR:   Error initializing runtime service opteed_fast

U-Boot 2020.01 (Jul 03 2020 - 10:20:01 +0000) odroid-n2
DRAM:  3.8 GiB
MMC:   sd@ffe05000: 0, mmc@ffe07000: 1
In:    serial@3000
Out:   serial@3000
Err:   serial@3000
Net:   eth0: ethernet@ff3f0000
Card did not respond to voltage select!
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc 1:1...
Found U-Boot script /boot.scr
3053 bytes read in 0 ms
## Executing script at 08000000
switch to partitions #0, OK
mmc1(part 0) is current device
loading env...
MMC read: dev # 1, block # 1212432, count 16 ... 16 blocks read: OK
3092 bytes read in 1 ms (2.9 MiB/s)
starting USB...
Bus usb@ff500000: Register 3000140 NbrPorts 3
Starting the controller
USB XHCI 1.10
scanning bus usb@ff500000 for devices... 1 USB Device(s) found
38 bytes read in 0 ms
50922 bytes read in 2 ms (24.3 MiB/s)
Found valid slot A, 1 attempts remaining
storing env...
MMC write: dev # 1, block # 1212432, count 16 ... 16 blocks written: OK
Loading kernel
20724224 bytes read in 463 ms (42.7 MiB/s)
 Starting kernel
load_kernel=ext4load mmc 1:5 0x08080000 Image
bootargs=zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor systemd.machine_id=eee573f0241d4f528c6559ed58983d43 cgroup_enable=memory clk_ignore_unused hdmimode=1080p60hz cvbsmode=576cvbs max_freq_a53=1896 max_freq_a73=1800 maxcpus=6  voutmode=hdmi disablehpd="false" overscan=100 root=PARTUUID=48617373-06 rootfstype=squashfs ro rootwait rauc.slot=A console=tty0 console=ttyAML0,115200n8
## Flattened Device Tree blob at 08008000
   Booting using the fdt blob at 0x8008000
   Loading Device Tree to 00000000f2f4a000, end 00000000f2f596e9 ... OK
Starting kernel ...
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.7.7 (builder@494736eef542) (gcc version 9.3.0 (Buildroot 2020.02.2-gc751537f), GNU ld (GNU Binutils) 2.32) #1 SMP PREEMPT Fri Jul 3 10:22:10 UTC 2020
[    0.000000] Machine model: Hardkernel ODROID-N2
[
....
```
Since U-Boot (with the correct timestamp) actually appears, it seems to me that your eMMC has a working eMMC boot partition. I see that your eMMC used to be a Android eMMC, I use one which had Linux on it. The only thing I could guess is that your eMMC has some boot partition which is not readable, and confuses U-Boot (e.g. a backup GPT). Can you maybe try overwrite the complete eMMC block device somehow (e.g. `dd if=/dev/zero of=/dev/sdX bs=4M` in Linux).

fbaccretio, 2020-07-23:
@agners  Thank you very much for your help...

I'm not familiar with the dd command. The output of `sudo blkid` is:

```
/dev/sdd1: SEC_TYPE="msdos" LABEL="hassos-boot" UUID="BA25-827E" TYPE="vfat" PARTUUID="48617373-01"
/dev/sdd3: LABEL="hassos-overlay" UUID="07c14ca5-6788-4047-9982-013589c04a55" TYPE="ext4" PARTUUID="48617373-03"
/dev/sdd4: LABEL="hassos-data" UUID="1564c57a-da2f-4210-8802-8543c756f2cb" TYPE="ext4" PARTUUID="48617373-04"
/dev/sdd5: LABEL="hassos-kernel" UUID="1eb73711-0f88-417a-89aa-7446b99da9d1" TYPE="ext4" PARTUUID="48617373-05"
/dev/sdd6: TYPE="squashfs" PARTUUID="48617373-06"
/dev/sdd7: PARTUUID="48617373-07"
/dev/sdd8: PARTUUID="48617373-08"
/dev/sdd9: PARTUUID="48617373-09"
```

Should i delete all partitions?

Is this correct? `dd if=/dev/zero of=/dev/sdd1 bs=4M`

agners, 2020-07-23:
Better to first unmount all of them: `sudo umount /dev/sdd*`. Then use `dd` to erase the complete block device so without partition number: `dd if=/dev/zero of=/dev/sdd bs=4M`

agners, 2020-07-23:
@fbaccretio FWIW, that is what I usually use to flash, so after zeroing out the whole device, you can use this command to flash the image:

```
sudo umount /dev/sdX*
gunzip -c ~/Downloads/hassos_odroid-n2-4.11.img.gz | sudo dd of=/dev/sdX bs=4M status=progress
```

fbaccretio, 2020-07-23:
@agners 

I used dd to erease the block device. And flashed the hassos with gunzip command. But still there is no boot

```
root@server01:~# blkid /dev/sdd
/dev/sdd: PTUUID="48617373" PTTYPE="dos"
root@server01:~# blkid /dev/sdd*
/dev/sdd: PTUUID="48617373" PTTYPE="dos"
/dev/sdd1: SEC_TYPE="msdos" LABEL="hassos-boot" UUID="BA25-827E" TYPE="vfat" PARTUUID="48617373-01"
/dev/sdd2: PTTYPE="dos" PARTUUID="48617373-02"
/dev/sdd3: LABEL="hassos-overlay" UUID="07c14ca5-6788-4047-9982-013589c04a55" TYPE="ext4" PARTUUID="48617373-03"
/dev/sdd4: LABEL="hassos-data" UUID="1564c57a-da2f-4210-8802-8543c756f2cb" TYPE="ext4" PARTUUID="48617373-04"
/dev/sdd5: LABEL="hassos-kernel" UUID="1eb73711-0f88-417a-89aa-7446b99da9d1" TYPE="ext4" PARTUUID="48617373-05"
/dev/sdd6: TYPE="squashfs" PARTUUID="48617373-06"
/dev/sdd7: PARTUUID="48617373-07"
/dev/sdd8: PARTUUID="48617373-08"
/dev/sdd9: PARTUUID="48617373-09"
root@server01:~# fdisk -l /dev/sdd
Disk /dev/sdd: 29.1 GiB, 31268536320 bytes, 61071360 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x48617373

Device     Boot   Start     End Sectors  Size Id Type
/dev/sdd1  *      16384   65537   49154   24M  c W95 FAT32 (LBA)
/dev/sdd2         65539 1228814 1163276  568M  5 Extended
/dev/sdd3       1228816 1425425  196610   96M 83 Linux
/dev/sdd4       1425427 3522580 2097154    1G 83 Linux
/dev/sdd5         65540  114693   49154   24M 83 Linux
/dev/sdd6        114695  638984  524290  256M 83 Linux
/dev/sdd7        638986  688139   49154   24M 83 Linux
/dev/sdd8        688141 1212430  524290  256M 83 Linux
/dev/sdd9       1212432 1228814   16383    8M 83 Linux

Partition table entries are not in disk order.
```
When i boot the odroid this is the serial output:
(i tested with multiple odroid-n2's)

```
HassOS> G12B:BL:6e7c85:7898ac;FEAT:E0F83180:402000;POC:F;RCY:0;EMMC:0;READ:0;0.
bl2_stage_init 0x01
bl2_stage_init 0x81
hw id: 0x0000 - pwm id 0x01
bl2_stage_init 0xc1
bl2_stage_init 0x02

L0:00000000
L1:00000703
L2:0000c067
L3:14000020
B2:00402000
B1:e0f83180

TE: 191095

BL2 Built : 06:17:13, Jun 28 2019. g12b gf0505d7-dirty - qi.duan@droid13

Board ID = 4
Set A53 clk to 24M
Set A73 clk to 24M
Set clk81 to 24M
A53 clk: 1200 MHz
A73 clk: 1200 MHz
CLK81: 166.6M
smccc: 000332ab
eMMC boot @ 0
[... 100 lines cut ...]
unable to select a mode
switch to partitions #0, OK
mmc1(part 0) is current device
** No partition table - mmc 1 **
MMC Device 2 not found
no mmc device at slot 2
starting USB...
Bus usb@ff500000: Register 3000140 NbrPorts 3
Starting the controller
USB XHCI 1.10
scanning bus usb@ff500000 for devices... 1 USB Device(s) found
usb - USB sub-system

Usage:
usb start - start (scan) USB controller
usb reset - reset (rescan) USB controller
usb stop [f] - stop USB [f]=force stop
usb tree - show USB device tree
usb info [dev] - show available USB devices
usb test [dev] [port] [mode] - set USB 2.0 test mode
    (specify port 0 to indicate the device's upstream port)
    Available modes: J, K, S[E0_NAK], P[acket], F[orce_Enable]

ethernet@ff3f0000 Waiting for PHY auto negotiation to complete..... done
```

agners, 2020-07-23:
This is really weird. Given that you PC definitly sees a valid/proper partition table, I assume the device can somehow not read the eMMC properly. Can you try to execute some commands manually? E.g press Ctrl+C continously to get into the U-Boot prompt. There, type commands such as:

```
HassOS> mmc rescan
HassOS> mmc list
sd@ffe05000: 0
mmc@ffe07000: 1 (eMMC)
HassOS> mmc dev 1 0
switch to partitions #0, OK
mmc1(part 0) is current device
HassOS> mmc info
Device: mmc@ffe07000
Manufacturer ID: 15
OEM: 100
Name: AJTD4 
Bus Speed: 52000000
Mode: MMC High Speed (52MHz)
Rd Block Len: 512
MMC version 5.1
High Capacity: Yes
Capacity: 14.6 GiB
Bus Width: 8-bit
Erase Group Size: 512 KiB
HC WP Group Size: 8 MiB
User Capacity: 14.6 GiB WRREL
Boot Capacity: 4 MiB ENH
RPMB Capacity: 4 MiB ENH
HassOS> mmc part

Partition Map for MMC device 1  --   Partition Type: DOS

Part    Start Sector    Num Sectors     UUID            Type
  1     16384           49154           48617373-01     0c Boot
  2     65539           1163276         48617373-02     05 Extd
  3     1228816         196610          48617373-03     83
  4     1425427         29110253        48617373-04     83
  5     65540           49154           48617373-05     83
  6     114695          524290          48617373-06     83
  7     638986          49154           48617373-07     83
  8     688141          524290          48617373-08     83
  9     1212432         16383           48617373-09     83
```

fbaccretio, 2020-08-03:
@agners 

No i could not interrupt the boot process.
I bought another set of emmc's and the new ones are working.
So i am pretty sure that it is related to faulty emmc's. It's very strange beacuse if i flash a regular Odroid Ubuntu the emmc's  work perfect. 

So i don't know exactly whats causing the issues with the bad emmcs.
Thank you very much for your help so far ;-)

Best regards.

agners, 2020-09-03:
Since the new eMMC seem to work fine assuming this was a hardware problem.

closed 2020-09-03 by agners

tjengbudi, 2020-12-30:
@fbaccretio sorry for open it once again. could you inform me what emmc hardware revision?

thanks