home-assistant/operating-system #773: Conbee 2 usb stick not recognised

BlackICE31, 2020-07-16:
**HassOS release with the issue:**

4.11

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

[60679.247329] usb 1-1.1: new full-speed USB device number 127 using xhci_hcd
[60679.347604] usb 1-1.1: device descriptor read/64, error -32
[60679.567599] usb 1-1.1: device descriptor read/64, error -32
[60679.787355] usb 1-1.1: new full-speed USB device number 3 using xhci_hcd
[60679.887599] usb 1-1.1: device descriptor read/64, error -32
[60680.107623] usb 1-1.1: device descriptor read/64, error -32
[60680.227598] usb 1-1-port1: attempt power cycle
[60680.887353] usb 1-1.1: new full-speed USB device number 4 using xhci_hcd
[60680.887580] usb 1-1.1: Device not responding to setup address.
[60681.107509] usb 1-1.1: Device not responding to setup address.
[60681.327337] usb 1-1.1: device not accepting address 4, error -71
[60681.427345] usb 1-1.1: new full-speed USB device number 5 using xhci_hcd
[60681.427559] usb 1-1.1: Device not responding to setup address.
[60681.647500] usb 1-1.1: Device not responding to setup address.
[60681.877354] usb 1-1.1: device not accepting address 5, error -71
[60681.877643] usb 1-1-port1: unable to enumerate USB device
[60683.347374] usb 1-1.1: new full-speed USB device number 6 using xhci_hcd
[60683.447631] usb 1-1.1: device descriptor read/64, error -32
[60683.667639] usb 1-1.1: device descriptor read/64, error -32
[60683.699125] Bluetooth: hci0: Frame reassembly failed (-84)
[60683.702979] Bluetooth: hci0: Frame reassembly failed (-84)
[60683.741642] Bluetooth: hci0: Frame reassembly failed (-84)
[60683.825309] Bluetooth: hci0: Frame reassembly failed (-84)
[60683.855342] Bluetooth: hci0: Frame reassembly failed (-84)
[60683.887378] usb 1-1.1: new full-speed USB device number 7 using xhci_hcd
[60683.987655] usb 1-1.1: device descriptor read/64, error -32
[60684.207636] usb 1-1.1: device descriptor read/64, error -32
[60684.327565] usb 1-1-port1: attempt power cycle
[60684.987395] usb 1-1.1: new full-speed USB device number 8 using xhci_hcd
[60684.987713] usb 1-1.1: Device not responding to setup address.
[60685.207544] usb 1-1.1: Device not responding to setup address.
[60685.427392] usb 1-1.1: device not accepting address 8, error -71
[60685.527403] usb 1-1.1: new full-speed USB device number 9 using xhci_hcd
[60685.527631] usb 1-1.1: Device not responding to setup address.
[60685.546077] Bluetooth: hci0: Frame reassembly failed (-84)
[60685.559664] Bluetooth: hci0: Frame reassembly failed (-84)
[60685.606476] Bluetooth: hci0: Frame reassembly failed (-84)
[60685.651570] Bluetooth: hci0: Frame reassembly failed (-84)
[60685.657898] Bluetooth: hci0: Frame reassembly failed (-84)
[60685.699826] Bluetooth: hci0: Frame reassembly failed (-84)
[60685.709794] Bluetooth: hci0: Frame reassembly failed (-84)
[60685.737617] Bluetooth: hci0: Frame reassembly failed (-84)
[60685.747561] usb 1-1.1: Device not responding to setup address.
[60685.767299] Bluetooth: hci0: Frame reassembly failed (-84)
[60685.840406] Bluetooth: hci0: Frame reassembly failed (-84)
[60685.863470] Bluetooth: hci0: Frame reassembly failed (-84)
[60685.880236] Bluetooth: hci0: Frame reassembly failed (-84)
[60685.967401] usb 1-1.1: device not accepting address 9, error -71
[60685.967740] usb 1-1-port1: unable to enumerate USB device
[60698.967530] usb 1-1.1: new full-speed USB device number 10 using xhci_hcd
[60699.067777] usb 1-1.1: device descriptor read/64, error -32
[60699.211311] Bluetooth: hci0: Frame reassembly failed (-84)
[60699.287797] usb 1-1.1: device descriptor read/64, error -32
[60699.463072] Bluetooth: hci0: Frame reassembly failed (-84)
[60699.507562] usb 1-1.1: new full-speed USB device number 11 using xhci_hcd
[60699.556446] Bluetooth: hci0: Frame reassembly failed (-84)
[60699.560165] Bluetooth: hci0: Frame reassembly failed (-84)
[60699.577889] Bluetooth: hci0: Frame reassembly failed (-84)
[60699.607764] usb 1-1.1: device descriptor read/64, error -32
[60699.726983] Bluetooth: hci0: Frame reassembly failed (-84)
[... 1458 lines cut ...]
[60784.118713] Bluetooth: hci0: Frame reassembly failed (-84)
[60784.128414] usb 1-1.1: new full-speed USB device number 49 using xhci_hcd
[60784.128642] usb 1-1.1: Device not responding to setup address.
[60784.154310] Bluetooth: hci0: Frame reassembly failed (-84)
[60784.168101] Bluetooth: hci0: Frame reassembly failed (-84)
[60784.272272] Bluetooth: hci0: Frame reassembly failed (-84)
[60784.291792] Bluetooth: hci0: Frame reassembly failed (-84)
[60784.348578] usb 1-1.1: Device not responding to setup address.
[60784.363173] Bluetooth: hci0: Frame reassembly failed (-84)
[60784.561801] Bluetooth: hci0: Frame reassembly failed (-84)
[60784.568407] usb 1-1.1: device not accepting address 49, error -71
[60784.568627] usb 1-1-port1: unable to enumerate USB device

**Description of problem:**
Conbee 2 usb stick is not recognised on any of the USB ports.
Several usb error about device not accepting address and inability to enumerate.
No success with rebooting or restarting HA.
Unplugging conbee2 stick and rebooting: /dev/ttyAMA0 still there. But no other usb device is attached
Notice also the bluetooth error 84. Plenty of them, dunno if related (I don't think so)

Hardware infos:
[image]

System info:
[image]

agners, 2020-07-17:
Maybe related to power, do you use a powerful enough power supply?

BlackICE31, 2020-07-18:
5.1v 3A. Probably not enough?

BlackICE31, 2020-07-18:
Just tried with my MacBook Pro's power supply but same results. So power supply shouldn't be the cause.
The cone 2 stick arrive just a couple days ago. Do I have a way to check if the stick works fine? I have a MacBook and a Synology.
No windows and no linux laptops.

agners, 2020-07-18:
3A should be enough typically but depends on what is connected. But since you tried a second power supply it is unlikely power related, I agree.

Do you use a USB extension cord or similar? Do you have any other (USB) devices plugged into the device? Did this work before? You might want to try 4.10 or 4.9 (which comes with the 5.4 upstream kernel. That release had its own set of probems, but maybe it helps for this case...).

BlackICE31, 2020-07-18:
I did try both with and without extension cord but no success. No other usb device connected.
Dunno if it worked before as the stick just arrived me. I may try rolling back to 4.9.

I’ll keep you posted. Thx for the help

BlackICE31, 2020-08-04:
@agners tried downgrading to 4.9 but I am still having the same usb issue. Also, still getting those bluetooth frame reassembly failures.
Worth trying a downgrade to 4.8? Do you have anybody else with same issues?

BlackICE31, 2020-08-04:
Additional note: downgrading to 4.7 makes the bluetooth frame reassembly error go away.
USB conbee 2 dongle still not recognised and still same logs.
Reattaching logs for completeness.
[4.7-HostLogs.txt](https://github.com/home-assistant/operating-system/files/5021950/4.7-HostLogs.txt)

en2te, 2020-08-08:
Same here! No Conbee II Stick accessible.  No lights, no sensors, nothing. Stick works if i use the Web-ui but the connection to HA is broken.

agners, 2020-08-10:
Can you try with 4.12? Also can you provide kernel logs from startup?

 I did test a ConBee 2 USB stick on 4.11 and 4.12 here, and in both cases it seems two work for me.

```
# ha info
arch: armv7
channel: stable
docker: 19.03.11
hassos: "4.12"
homeassistant: 0.113.3
hostname: homeassistant
logging: info
machine: raspberrypi4
supervisor: "229"
supported_arch:
- armv7
- armhf
timezone: UTC
```

Installing 4.12 from scratch helps.

BlackICE31, 2020-08-24:
Sorry for the late reply. Was finally in vacation :)
I did update to 4.12 but no success.
I attach full logs of host since startup.

[HassOS-logs.pdf](https://github.com/home-assistant/operating-system/files/5118272/HassOS-logs.pdf)

closed 2021-05-12 by stale[bot]