home-assistant/operating-system #774: Need Intel nuc NUC10i7FNH bluetooth driver

Wampukas, 2020-07-17:
**HassOS release with the issue:**HassOS 4.11 and earlier

Home Assistant 0.112.4
**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

[620820.901572] bluetooth hci0: Direct firmware load for intel/ibt-19-0-4.sfi failed with error -2
[620820.902055] Bluetooth: hci0: Failed to load Intel firmware file (-2)
**Description of problem:**
Missing intel nuc NUC10i7FNH bluetooth driver in OVA file.

Wampukas, 2020-07-24:
Anyone?

basnijholt, 2020-09-07:
For me bluetooth used to work! However, recently (with an update) it broke.

I have this problem on a Intel NUC Kit NUC8i3BEH and this is in the logs:
```
[134060.888355] Bluetooth: hci0: Bootloader revision 0.1 build 42 week 52 2015
[134060.890411] Bluetooth: hci0: Device revision is 2
[134060.891090] Bluetooth: hci0: Secure boot is enabled
[134060.891820] Bluetooth: hci0: OTP lock is enabled
[134060.892475] Bluetooth: hci0: API lock is enabled
[134060.893105] Bluetooth: hci0: Debug lock is disabled
[134060.893821] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
[134060.895404] bluetooth hci0: Direct firmware load for intel/ibt-17-16-1.sfi failed with error -2
[134060.896955] Bluetooth: hci0: Failed to load Intel firmware file (-2)
```

moeffju, 2020-09-10:
Is there any way to get eyes on this? I wouldn’t know where to start debugging or fixing this unfortunately :(

CuBiC3D, 2020-09-13:
Same for me stopped working with `Direct firmware load for intel/ibt-18-16-1.sfi`

mpepping, 2021-01-23:
Same here with the NUC 10.

```
[617278.542862] Bluetooth: hci0: Bootloader revision 0.4 build 0 week 30 2018
[617278.545938] Bluetooth: hci0: Device revision is 2
[617278.545941] Bluetooth: hci0: Secure boot is enabled
[617278.545941] Bluetooth: hci0: OTP lock is enabled
[617278.545942] Bluetooth: hci0: API lock is enabled
[617278.545943] Bluetooth: hci0: Debug lock is disabled
[617278.545944] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
[617278.545990] bluetooth hci0: Direct firmware load for intel/ibt-19-0-4.sfi failed with error -2
[617278.545992] Bluetooth: hci0: Failed to load Intel firmware file (-2)
```

regoras, 2021-01-25:
On a related note, running from a VM, but also HassOS, trying to pass in the bluetooth from an AX200: 

```
[    1.702780] Bluetooth: hci0: Bootloader revision 0.3 build 0 week 24 2017
[    1.709408] Bluetooth: hci0: Device revision is 1
[    1.710875] Bluetooth: hci0: Secure boot is enabled
[    1.711952] Bluetooth: hci0: OTP lock is enabled
[    1.712940] Bluetooth: hci0: API lock is enabled
[    1.713901] Bluetooth: hci0: Debug lock is disabled
[    1.714894] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
[    1.716018] bluetooth hci0: Direct firmware load for intel/ibt-20-1-3.sfi failed with error -2
[    1.717412] Bluetooth: hci0: Failed to load Intel firmware file (-2)
```

Any way we could get the more recent Intel firmwares?

dalanmiller, 2021-02-14:
fyi this is what I see on Ubuntu 20.04.2 host:

```
$ dmesg | grep -i blue
[    5.086614] Bluetooth: Core ver 2.22
[    5.086629] Bluetooth: HCI device and connection manager initialized
[    5.086633] Bluetooth: HCI socket layer initialized
[    5.086635] Bluetooth: L2CAP socket layer initialized
[    5.086638] Bluetooth: SCO socket layer initialized
[    5.244890] Bluetooth: hci0: Bootloader revision 0.1 build 42 week 52 2015
[    5.245909] Bluetooth: hci0: Device revision is 2
[    5.245910] Bluetooth: hci0: Secure boot is enabled
[    5.245911] Bluetooth: hci0: OTP lock is enabled
[    5.245911] Bluetooth: hci0: API lock is enabled
[    5.245912] Bluetooth: hci0: Debug lock is disabled
[    5.245913] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
[    5.246939] Bluetooth: hci0: Found device firmware: intel/ibt-17-16-1.sfi
[    6.676194] Bluetooth: hci0: Waiting for firmware download to complete
[    6.676843] Bluetooth: hci0: Firmware loaded in 1402806 usecs
[    6.676899] Bluetooth: hci0: Waiting for device to boot
[    6.690084] Bluetooth: hci0: Device booted in 12910 usecs
[    6.690333] Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-17-16-1.ddc
[    6.692978] Bluetooth: hci0: Applying Intel DDC parameters completed
[   16.565976] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   16.565977] Bluetooth: BNEP filters: protocol multicast
[   16.565980] Bluetooth: BNEP socket layer initialized
[   18.581210] Bluetooth: RFCOMM TTY layer initialized
[   18.581215] Bluetooth: RFCOMM socket layer initialized
[   18.581218] Bluetooth: RFCOMM ver 1.11
```

And yet, I still get an error: 

```
$ sudo hcitool lescan
Set scan parameters failed: Input/output error
```

basnijholt, 2021-05-12:
This is still a problem, although it might be fixes (see https://github.com/home-assistant/operating-system/issues/820#issuecomment-817739369).

mpepping, 2021-06-13:
Works since Home Assistant OS 6.0!

agners, 2021-06-17:
@mpepping cool, thanks for the update!

closed 2021-06-17 by agners