home-assistant/operating-system #778: RPi4 Kernel 4.19.127-v7l with annoing issue why?

dianlight, 2020-07-19:
- If you are using Raspberry Pi overlays in config.txt and your system doesn't work anymore please report this to the Raspberry Pi kernel project.

**_There is no official kernel 4.19.127-v7l so no support from Rasperry Pi kernel project :(_**

**HassOS release with the issue:**
4.10 and 4.11

**Hardware:**
Raspberry Pi 4 Model B Rev 1.1
BOOTLOADER: up-to-date
 CURRENT: Thu Apr 16 17:11:26 UTC 2020 (1587057086)
 LATEST: Thu Apr 16 17:11:26 UTC 2020 (1587057086)
VL805: up-to-date
 CURRENT: 000137ad
 LATEST: 000137ad

**Description of problem:**

I found several very boring malfunctions with last two version of operating.system. On my RPi4 the first is the Wifi can't connect to a Mash Wifi with fast routing enabled (#733) the second is the wrong mapping of /dev/serial1 on ttyAMA1 if 
uart2 is enable in config. ( no issue because a easy manual boot script fix,  /dev/serial1 need to be map to /dev/AMA0 for btuart ).

Because Raspbian has kernel 4.19.118-v7l+ and no issue on the same hw I try to switch the kernel on the hassio and all returns to work perfectly. I go to Raspberry Pi Kernel repository to open the issue and found that the last kernel is 4.19.118-v7l :( Stable kernel switch from 4.19.118-v7l to 5.4.35 on 1 May so no 4.19.127-v7l 

My question is: 
- what are the advantages of using the .127 kernel?
- where can I find the sources of the kernel .127 used?

L.

agners, 2020-07-19:
- what are the advantages of using the .127 kernel?

As with any stable kernel release: mainly bug and potentially security fixes...

- where can I find the sources of the kernel .127 used?

The kernel is referenced from the buildroot RPi4 configuration: https://github.com/home-assistant/operating-system/blob/rel-4/buildroot-external/configs/rpi4_defconfig

You can find the hash (a03605b08ae73107100c72dc92c92adf0dde3e42) in the official 4.19.y kernel branch of the RPi kernel tree here: https://github.com/raspberrypi/linux/commits/rpi-4.19.y

dianlight, 2020-07-20:
Thank you. 
I am doing my homework by scrolling through the various commits and I have found the cause of the serial link switch (I am preparing the PR for the fix) but I have not yet found the cause of the problem with the wifi.

@agners I agree that the .127 version may contain improvements and security fixes but usually in distros, to avoid problems, you do not simply change the kernel but backport what interests you. So does raspbian especially because the .127 version is released in the branch rpi-4.19.y but has not been released as stable firmware  https://github.com/raspberrypi/firmware/commits/stable

L

agners, 2020-07-20:
> I am doing my homework by scrolling through the various commits and I have found the cause of the serial link switch (I am preparing the PR for the fix) [..]

Cool!

> [...] but I have not yet found the cause of the problem with the wifi.

Just realized that according to #733 that actually only worked with 4.9 correct? Back then Home Assistant OS used the newer 5.4 kernel afaict, so maybe this actually never really worked with the 4.19 kernel...?

> I agree that the .127 version may contain improvements and security fixes but usually in distros, to avoid problems, you do not simply change the kernel but backport what interests you.

Yeah but that is more work... I assume that Home Assistant does not have a kernel team like most distros have.

Besides, also distros usually merge sub-releases these days, probably makes it to a bit bigger QA afterwards. But that is upstream patch releases....  I guess in this case it was anyway something which got changed by the RPi project.

dianlight, 2020-07-21:
> Just realized that according to #733 that actually only worked with 4.9 correct? Back then Home Assistant OS used the newer 5.4 kernel afaict, so maybe this actually never really worked with the 4.19 kernel...?

No works with 3.x hassos version and Raspbian that are also on 4.19. Also a "quick and dirty" backport of the Raspbian 4.19.118 kernel on top of hassos 4.11 work. 

> Yeah but that is more work... I assume that Home Assistant does not have a kernel team like most distros have.

I understand, in fact I would surpass the opposite given the purpose of the project. Precisely for this reason I would have expected that a kernel from another distribution would be used without starting from the sources.

corvy, 2020-08-09:
There is a new kernel in HassOS 4.12, is this also affected?

dianlight, 2020-08-09:
I don’t know, I don't have physical access to the RPi until the end of August, I prefer to do kernel tests when I can intervene in case of problems.

L

corvy, 2020-08-10:
I will do a test on mesh wifi problem and report back.

corvy, 2020-08-10:
The new OS works with mesh network and fast user switching. :)

https://www.dropbox.com/s/on50n1klh19igqa/Screenshot%202020-08-10%20at%2015.59.07.png?dl=0

mattclar, 2020-08-29:
It seems like the 5.4 kernel is now stable from raspberry pi foundation. Maybe we can add this to the next beta?

agners, 2020-09-09:
@mattclar rel-4 will stay on 4.19. For the upcoming rel-5 the 5.4 kernel seems sensible. I can tackle it at one point, but currently I do have other priorities.

@dianlight is there anything open from your side on this issue or can it be closed?

dianlight, 2020-09-09:
From my side:

- The Wifi problem is fixed with PR #830
- The wrong mapping of /dev/serial1 on ttyAMA1 is fixable with an update of the 99-rpi.rules ( operating-system/buildroot-external/board/raspberrypi/rootfs-overlay/usr/lib/udev/rules.d ) with the last version from https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/master/etc.armhf/udev/rules.d/99-com.rules. I'm unable to propose a PR (working disclosure reason) if anyone can  I would be happy

L.

agners, 2020-09-10:
@dianlight we seem to have such rules already: https://github.com/home-assistant/operating-system/blob/dev/buildroot-external/board/raspberrypi/rootfs-overlay/usr/lib/udev/rules.d/99-rpi.rules.

At least on 4.13 it seems to work with UART2 enabled:

```
# ls -la /dev/ttyAMA*
crw-rw----    1 root     dialout   204,  64 Sep 10 22:04 /dev/ttyAMA0
crw-rw----    1 root     dialout   204,  65 Sep 10 22:04 /dev/ttyAMA1
# ls -la /dev/serial*
lrwxrwxrwx    1 root     root             5 Sep 10 22:04 /dev/serial0 -> ttyS0
lrwxrwxrwx    1 root     root             7 Sep 10 22:04 /dev/serial1 -> ttyAMA
```
Maybe this breakage was related to the 5.4 kernel change in releases from 4.6 to 4.9.

dianlight, 2020-09-11:
Is not the same. 
In my config rpi4 with uart2 overlay and not mini bt. 

I have not ttyS0 so for the rule 
Serial1 point to ttyAMA1.
btuart uses always serial1 so it need to point to Bluetooth that is ttyAMA0

The updated rules with version 4.10 works. I have not tested with other because is much faster connect to ssh, change the link and restart the Bluetooth than rebuild all os.

I don't know how to read rule in right way but in the second ( ttyAMA1 ) you can se that if AMA0 exist AMA1 point always to serial1 that is wrong ( for me :( ) but do the right stuff and after boot the serial1 is a link to ttyAMA0.

L.

agners, 2020-10-12:
@dianlight the remaining issue should now be fixed in 4.14 as well. Thanks for the detailed reports and help to get them solved. Ideally though, multiple issues should be reported in separate tickets :wink:

closed 2020-10-12 by agners