home-assistant/operating-system #785: Make it possible to set Wifi regulatory domain

gyger, 2020-07-24:
I am using the Raspberry Pi 3 in a country that allows to switch to channel 13, which is allowed on most of the world, but unfortunately not in the US, so by default the wifi chip does not connect to this channel.

It is possible in principle in Linux to set the regulatory domain, but I found no way to do it in HassOS.
[Balena](https://www.balena.io/docs/reference/OS/network/2.x/) e.g. allows to set a JSON file in the boot partition to set this regulatory domain.

pvizeli, 2020-07-29:
I'm open for an RFC

hendricksond, 2020-08-11:
In Raspberry Pi OS, this is done using wpa_supplicant.conf, by setting `country=<Insert 2 letter ISO 3166-1 country code here>`. See https://www.raspberrypi.org/documentation/configuration/wireless/headless.md

I haven't been able to figure out a way of doing this using the HassOS `my-network` file. If it's possible, it's not documented. Is it possible to add?

Vlad-Star, 2020-08-13:
+1. Same story. RPi4 doesn't work with 5.0GHz band without a country code, which determines proper radio frequencies to use.
https://en.m.wikipedia.org/wiki/ISO_3166-1_alpha-2

hendricksond, 2020-10-13:
This is still an issue.

martin-klima, 2021-01-25:
I have the same problem.

boesing, 2021-03-31:
Can confirm that Raspberry Pi 4 connects to 5GHz as `na` but the speed is capped to 150MBit/s on a unifi network.

rslindee, 2021-04-22:
+1. I am having the same issue, as have other users, it seems (https://community.home-assistant.io/t/change-country-code-wifi-hassos/111063 and https://community.home-assistant.io/t/wifi-country-setting/105852/4)

Vlad-Star, 2021-04-22:
I played with it a bit today. There are at least 3 methods to set up regulatory domains.

0) Connected to SSH port 22222 of the Hassio:
```
Using username "root".
Authenticating with public key "imported-openssh-key"

Welcome to the Home Assistant command line.

System information
[...]
  OS Version:               Home Assistant OS 5.13
  Home Assistant Core:      2021.4.6

  Home Assistant URL:       http://homeassistant.local:8123
  Observer URL:             http://homeassistant.local:4357

ha > login
#
```

1) via /etc/wpa_supplicant.conf. Obviously, can't be done due to readonly filesystem.
 ```
 # echo "country=US" >/etc/wpa_supplicant.conf
/bin/ash: can't create /etc/wpa_supplicant.conf: Read-only file system
#
```
2) via CRDA. Doesn't work either.
https://linux.die.net/man/8/crda
```
# crda
COUNTRY environment variable not set.
# export COUNTRY=US
# crda
Failed to set regulatory domain: -7
#
```
3) via module options in /etc/modprobe.d. It's somewhat working, but doesn't survive the reboot (I believe, because kernel modules are loaded prior to the /etc/modprobe.d mounted off the card)
```
# mount | grep /etc/modprobe.d
/dev/mmcblk0p7 on /etc/modprobe.d type ext4 (rw,relatime)
# nmcli d wifi list
IN-USE  BSSID              SSID         MODE   CHAN  RATE        SIGNAL  BARS  SECURITY
        3A:2C:4A:95:62:EA  XXXXXXXX24   Infra  10    405 Mbit/s  100     ****  WPA1 WPA2
        38:2C:4A:95:62:E8  XXXX24       Infra  10    405 Mbit/s  100     ****  WPA1 WPA2
# lsmod | grep cfg80211
cfg80211              815104  1 brcmfmac
# echo 'options cfg80211 ieee80211_regdom="US"' >/etc/modprobe.d/cfg80211.conf
# ls -l /etc/modprobe.d/cfg80211.conf
-rw-r--r--    1 root     root            39 Apr 22 16:59 /etc/modprobe.d/cfg80211.conf
(reloading the modules)
# rmmod brcmfmac
# rmmod cfg80211
# modprobe cfg80211
# modprobe brcmfmac
(5.0 SSID now showed up in the list)
# nmcli d wifi list
IN-USE  BSSID              SSID         MODE   CHAN  RATE        SIGNAL  BARS  SECURITY
        3A:2C:4A:95:62:EA  XXXXXXXX24   Infra  10    405 Mbit/s  100     ****  WPA1 WPA2
        3A:2C:4A:95:62:EE  XXXXXXXX50   Infra  149   195 Mbit/s  100     ****  WPA1 WPA2
        38:2C:4A:9B:4F:88  XXXX24       Infra  10    405 Mbit/s  94      ****  WPA1 WPA2
        38:2C:4A:9B:4F:8C  XXXX50       Infra  149   195 Mbit/s  80      ***   WPA1 WPA2
#
# nmcli d wifi connect <SSID> password <PASS>
Device 'wlan0' successfully activated with '41063037-1fb3-4fdd-87ee-b93bdf165f41'.
(interface wlan0 is certainly up now)
# ip addr
[...]
33: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel qlen 1000
    link/ether dc:a6:32:14:63:64 brd ff:ff:ff:ff:ff:ff
    inet 192.168.X.X/24 brd 192.168.X.255 scope global dynamic wlan0
       valid_lft 86271sec preferred_lft 86271sec
    inet6 fe80::8e71:8045:3506:ad18/64 scope link
       valid_lft forever preferred_lft forever
#
# reboot
(and I had to repeat the same steps above to get 5.0Ghz working, though /etc/modprobe.d/cfg80211.conf was there)
```

agners, 2021-10-13:
Whoops, did not realize that we have this old issue, should have checked first :innocent: 

However, closing this one in favor of #1530 since it describes a work around.

> Shouldn't this be marked as a bug (not an enhancement), since it prevents proper use of the 5ghz wifi?

I guess this is a bit philosophical. It won't matter much from a prioritization perspective, especially since the "bug" isn't a regression.

closed 2021-10-13 by agners