home-assistant/operating-system #788: New Beta V5.0 boots OS with SSD, but failed to start Home assistant

jodur, 2020-07-25:
**HassOS release with the issue:**

V5.0 Beta, RPI4 64 bit image

**Supervisor logs:**

[32m20-07-25 17:27:37 INFO (MainThread) [supervisor.api] Start API on 172.30.32.2[0m
[32m20-07-25 17:27:37 INFO (MainThread) [supervisor.utils.gdbus] Call de.pengutronix.rauc.Installer.Mark on /[0m
[32m20-07-25 17:27:37 INFO (MainThread) [supervisor.hassos] Rauc: A - marked slot kernel.0 as good[0m
[32m20-07-25 17:27:37 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.INITIALIZE' start 0 add-ons[0m
[32m20-07-25 17:27:37 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SYSTEM' start 0 add-ons[0m
[32m20-07-25 17:27:37 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SERVICES' start 0 add-ons[0m
[32m20-07-25 17:27:37 INFO (SyncWorker_1) [supervisor.docker.interface] Start homeassistant/raspberrypi4-64-homeassistant[0m
[31m20-07-25 17:33:53 ERROR (MainThread) [supervisor.homeassistant] Home Assistant has crashed![0m
[32m20-07-25 17:33:53 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.APPLICATION' start 0 add-ons[0m
[32m20-07-25 17:33:53 INFO (MainThread) [supervisor.misc.tasks] All core tasks are scheduled[0m
[32m20-07-25 17:33:53 INFO (MainThread) [supervisor.misc.hwmon] Started Supervisor hardware monitor[0m
[32m20-07-25 17:33:53 INFO (MainThread) [supervisor.core] Supervisor is up and running[0m
[32m20-07-25 17:33:53 INFO (MainThread) [supervisor.host.info] Update local host information[0m
[32m20-07-25 17:33:53 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/hostname1[0m
[32m20-07-25 17:33:53 INFO (MainThread) [supervisor.updater] Fetch update data from https://version.home-assistant.io/stable.json[0m
[32m20-07-25 17:33:53 INFO (MainThread) [supervisor.host.services] Update service information[0m
[32m20-07-25 17:33:53 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.systemd1.Manager.ListUnits on /org/freedesktop/systemd1[0m
[32m20-07-25 17:33:53 INFO (MainThread) [supervisor.host.network] Update local network DNS information[0m
[32m20-07-25 17:33:53 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/NetworkManager/DnsManager[0m
[32m20-07-25 17:33:53 INFO (MainThread) [supervisor.host.sound] Update PulseAudio information[0m
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing... 
[17:36:06] INFO: [32mUpdate udev informations[0m
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[32m20-07-25 17:36:08 INFO (MainThread) [__main__] Initialize Supervisor setup[0m
[32m20-07-25 17:36:09 INFO (MainThread) [supervisor.bootstrap] Setup coresys for machine: raspberrypi4-64[0m
[32m20-07-25 17:36:09 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attach to Supervisor homeassistant/aarch64-hassio-supervisor with version 229[0m
[32m20-07-25 17:36:09 INFO (MainThread) [__main__] Setup Supervisor[0m
[32m20-07-25 17:36:09 INFO (MainThread) [supervisor.utils.gdbus] Connect to dbus: org.freedesktop.systemd1 - /org/freedesktop/systemd1[0m
[32m20-07-25 17:36:09 INFO (MainThread) [supervisor.utils.gdbus] Connect to dbus: org.freedesktop.hostname1 - /org/freedesktop/hostname1[0m
[32m20-07-25 17:36:09 INFO (MainThread) [supervisor.utils.gdbus] Connect to dbus: de.pengutronix.rauc - /[0m
[32m20-07-25 17:36:09 INFO (MainThread) [supervisor.utils.gdbus] Connect to dbus: org.freedesktop.NetworkManager - /org/freedesktop/NetworkManager/DnsManager[0m
[32m20-07-25 17:36:09 INFO (MainThread) [supervisor.host.info] Update local host information[0m
[32m20-07-25 17:36:09 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/hostname1[0m
[32m20-07-25 17:36:09 INFO (MainThread) [supervisor.host.services] Update service information[0m
[32m20-07-25 17:36:09 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.systemd1.Manager.ListUnits on /org/freedesktop/systemd1[0m
[32m20-07-25 17:36:09 INFO (MainThread) [supervisor.host.network] Update local network DNS information[0m
[32m20-07-25 17:36:09 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/NetworkManager/DnsManager[0m
[32m20-07-25 17:36:09 INFO (MainThread) [supervisor.host.sound] Update PulseAudio information[0m
[32m20-07-25 17:36:09 INFO (MainThread) [supervisor.host.apparmor] Load AppArmor Profiles: {'hassio-supervisor'}[0m
[32m20-07-25 17:36:09 INFO (MainThread) [supervisor.host.services] Reload local service hassos-apparmor.service[0m
[32m20-07-25 17:36:09 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.systemd1.Manager.ReloadOrRestartUnit on /org/freedesktop/systemd1[0m
[32m20-07-25 17:36:09 INFO (SyncWorker_0) [supervisor.docker.interface] Attach to homeassistant/aarch64-hassio-dns with version 9[0m
[32m20-07-25 17:36:09 INFO (MainThread) [supervisor.plugins.dns] Start CoreDNS plugin[0m
[32m20-07-25 17:36:09 INFO (MainThread) [supervisor.misc.forwarder] Start DNS port forwarding to 172.30.32.3[0m
[32m20-07-25 17:36:09 INFO (SyncWorker_0) [supervisor.docker.interface] Clean hassio_dns application[0m
[32m20-07-25 17:36:10 INFO (SyncWorker_0) [supervisor.docker.dns] Start DNS homeassistant/aarch64-hassio-dns with version 9 - 172.30.32.3[0m
[32m20-07-25 17:36:10 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf[0m
[32m20-07-25 17:36:10 INFO (SyncWorker_0) [supervisor.docker.interface] Attach to homeassistant/aarch64-hassio-audio with version 16[0m
[32m20-07-25 17:36:10 INFO (MainThread) [supervisor.plugins.audio] Start Audio plugin[0m
[32m20-07-25 17:36:10 INFO (SyncWorker_0) [supervisor.docker.interface] Clean hassio_audio application[0m
[32m20-07-25 17:36:11 INFO (SyncWorker_0) [supervisor.docker.audio] Start Audio homeassistant/aarch64-hassio-audio with version 16 - 172.30.32.4[0m
[32m20-07-25 17:36:11 INFO (SyncWorker_0) [supervisor.docker.interface] Attach to homeassistant/aarch64-hassio-cli with version 25[0m
[32m20-07-25 17:36:11 INFO (MainThread) [supervisor.plugins.cli] Start cli plugin[0m
[32m20-07-25 17:36:11 INFO (SyncWorker_0) [supervisor.docker.interface] Clean hassio_cli application[0m
[32m20-07-25 17:36:12 INFO (SyncWorker_0) [supervisor.docker.cli] Start CLI homeassistant/aarch64-hassio-cli with version 25 - 172.30.32.5[0m
[32m20-07-25 17:36:12 INFO (SyncWorker_0) [supervisor.docker.interface] Attach to homeassistant/aarch64-hassio-multicast with version 2[0m
[... 713 lines cut ...]
[  396.801556] mmc0: sdhci: Cmd:       0x0000341a | Max curr: 0x00080008
[  396.803148] mmc0: sdhci: Resp[0]:   0x00000000 | Resp[1]:  0x00000000
[  396.804544] mmc0: sdhci: Resp[2]:   0x00000000 | Resp[3]:  0x00000000
[  396.805918] mmc0: sdhci: Host ctl2: 0x00000000
[  396.807473] mmc0: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0x00000000
[  396.808776] mmc0: sdhci: ============================================

**Description of problem:**

I burned the  Beta V5.0 RPI4 64 image and the OS is booting up from SSD.

Docker is running, but the `docker ps` command only gives me the following running containers:

hassiio_multicast
hassio_cli
hassio_dns
hassio_supervisor

No running instances of homeassistant core?

`docker images` shows that te image (raspberrypi4-64-homeassistant) is present but not started?

results in starting eudev-3.2.9, but exits with error

Log Show a fatal error during startup of home-assistant core

pepeEL, 2020-07-25:
Hi
Can you write me step by step how burn and run HASSOS on USB SSD on RPI4 ?
I download image v5 and then burn by Raspberry Imager v1.2 on my SSD mSATA with adapter USB. Then connect it to RPI4 and power on. But Rpi4 no boot and Home Assistant not working. Please help...

jodur, 2020-07-26:
reimaged the SSD drive again and added sd_poll_once=on to dtparam in config.txt and to my suprise it is now running. Maybe a bad flash Earlier or the change in the config made It working. Have restored my config and now testing for stability

closed 2020-07-26 by jodur

dweston, 2020-09-10:
> reimaged the SSD drive again and added sd_poll_once=on to dtparam in config.txt and to my suprise it is now running. Maybe a bad flash Earlier or the change in the config made It working. Have restored my config and now testing for stability

@jodur Out of interest how do you access `config.txt` on the imaged SSD drive? I'm on macos and the imaged SSD card does not mount.

dweston, 2020-09-10:
> Hi
> Can you write me step by step how burn and run HASSOS on USB SSD on RPI4 ?
> I download image v5 and then burn by Raspberry Imager v1.2 on my SSD mSATA with adapter USB. Then connect it to RPI4 and power on. But Rpi4 no boot and Home Assistant not working. Please help...
> 
> I have updated my RPI4 EEPROM
> pi@raspberrypi:~ $ sudo rpi-eeprom-update -a
> BCM2711 detected
> Dedicated VL805 EEPROM detected
> BOOTFS /boot
> BOOTLOADER: up-to-date
> CURRENT: Thu 16 Jul 15:15:46 UTC 2020 (1594912546)
> LATEST: Thu 16 Jul 15:15:46 UTC 2020 (1594912546)
> FW DIR: /lib/firmware/raspberrypi/bootloader/stable
> VL805: up-to-date
> CURRENT: 000138a1
> LATEST: 000138a1

@pepeEL did you resolve this?

jodur, 2020-09-10:
The first time i connected the SSD to my windows computer and edit the config.txt. 
I now make changes on my running system trough SSH on port 22222  and use the linux VI editor to edit the config file in `\mnt\boot`  To bad that HassOS has not installed the linux nano-editor, becaus that is easier to edit text files

dweston, 2020-09-10:
> The first time i connected the SSD to my windows computer and edit the config.txt.
> I now make changes on my running system trough SSH on port 22222 and use the linux VI editor to edit the config [...]

I tried gaining SSH access but must not have executed the set-up correctly. Anyhow the Windows edit path worked.

I have the Rpi4 booting from SSD with hassos_rpi4-64-5.2.img.gz. The speed is blazing comparative to the Rpi3.