home-assistant/operating-system #789: HassOS OVA wont import into Ovirt 4.3.10 because of Invalid VM Name

derekatkins, 2020-07-25:
I was trying to import HassOS OVA 4.11 (OVA version) into my Ovirt 4.3.10 infrastructure.
The first time I attempted this it complained about the VM Name:

Error while executing action:
Home Assistant:
  * Can not import VM. The given name contains special characters. Only lower-case and upper-case letters, numbers, '_', '-', '.' are allowed.
  * Attribute: vm.vmStatic.name

By my reading, the issue is that the VM is named "Home Assistant" (with the space).  Could you consider renaming the VM in the OVA without the space?

pvizeli, 2020-07-27:
I think that is a bug in ovirt. I see nothing in the OVF specs with would disallow this. However, if we doing it wrong and the OVF specs did not allow it, we can change it, otherwise ovirt need to fix it because it would be not our bug.

derekatkins, 2020-07-27:
The OVF specification is behind a paywall so I have no way to verify the specification.  All the links I've found at VMware are expired.  I did find on google cache entry which says "A VM name cannot contain any special characters" but then links off to a dead knowledge base entry:  https://webcache.googleusercontent.com/search?q=cache:jBmcHvb438wJ:https://pubs.vmware.com/vcd-820/topic/com.vmware.vcloud.admin.doc/GUID-023D4FBC-3016-4757-B52B-4A97984ABD7F.html+&cd=10&hl=en&ct=clnk&gl=us&client=firefox-b-1-d

In the meantime, there are other issues with trying to import the OVA into ovirt, like virt-v2v doesn't like the OS so can't process the disk.  I'm going to try the QCOW2 route and load that into virt-manager and have virt-v2v pull from there.  A lot of work to get this into ovirt!

PR #794 (2020-07-29): Change VM naming handling
This should be compatible with most systems now.
Fix: #789
closed 2020-07-29 by pvizeli