home-assistant/operating-system #790: HASSOS V5 boot from USB SSD on RPI4

pepeEL, 2020-07-25:
Hi
I upgrade my RPI4 eeprom and firmware to latest:
pi@raspberrypi:~ $ sudo rpi-eeprom-update -a
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTFS /boot
BOOTLOADER: up-to-date
CURRENT: Thu 16 Jul 15:15:46 UTC 2020 (1594912546)
 LATEST: Thu 16 Jul 15:15:46 UTC 2020 (1594912546)
 FW DIR: /lib/firmware/raspberrypi/bootloader/stable
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1

Then i download image HASSOS v5 and burn it by Raspberry Imager v1.2 on my SSD mSATA with adapter USB. Then connect it to RPI and power ON but RPI no boot. Why ? Is anybody who can write me step by step how do it ? Please....

jodur, 2020-07-26:
Did you try boot with original Pi OS?, in that way you now it is capable of booting from SSD. 
This firmware is Beta and you can expect issues. Also did you use the V5 RPI 64bit image?
This is the only one that supports usb boot loading

erik73, 2020-07-26:
Have you checked the output of vcgencmd bootloader_config ?
What is reported as BOOT_ORDER

pepeEL, 2020-07-26:
I try booting raspbian buster and also not working.
Hassos i try only version 32bit.
In boot_order i have 0xf41 and firmware is stable not beta. Version 16.07.2020 is version stable and this version i have.

erik73, 2020-07-26:
Ok, the boot order and eeprom firmware looks OK. 
The 32-bit version of HassOS odes not have USB boot for RPI4. 
Try the 64-bit version.

jodur, 2020-07-26:
If you cannot run it with the original PiOs i doubt if you will have succes with HassOS. Check youre hardeware config if it is compatible with the pi

pepeEL, 2020-07-26:
Yes you are right version 64bit boo ok from SSD. Thanks :)

closed 2020-07-26 by pepeEL

BerniApple, 2020-08-17:
> Yes you are right version 64bit boo ok from SSD. Thanks :)

May you help me? Can you please share a step by step guide?
Many thanks

Appyx, 2020-08-19:
@BerniApple 
I used this guide until step 7 to upgrade the bootloader: https://www.tomshardware.com/how-to/boot-raspberry-pi-4-usb

I tried to boot off the RPI OS but that didn't work. 
So I just tried HA OS and downloaded the 5.1 release for 64 bit.
Flashed it onto the SSD with balenaEtcher and plugged it in.
Done.

Works like magic :)