home-assistant/operating-system #791: Make it possible to switch between Raspberry Pis

Nowaker, 2020-07-26:
- **HassOS release with the issue:** all
- **Supervisor logs:** irrelevant
- **Journal logs:** irrelevant
- **Kernel logs:** irrelevant

**Description of problem:**

[People really want to be able to migrate between Raspberry Pis](https://community.home-assistant.io/t/migrating-from-raspberry-pi-3-to-pi-4/159768), especially upgrading from 3 to 4. It's currently not possible. Please make it possible.

frenck, 2020-07-26:
The only way this is possible (and this is already possible), is by taking a snapshot of your old system, and restore it on a new, fresh, system.

As the Home Assistant Operating System is an embedded operating system, it means that the OS is specifically built for each type of board, and thus not inter-exchangeable.

closed 2020-07-26 by frenck

Nowaker, 2020-07-30:
> The only way this is possible (and this is already possible), is by taking a snapshot of your old system, and restore it on a new, fresh, system.

I've done this instead so I don't have to do snapshots at all.

```
losetup -Pf hassos_rpi4-64-5.0.img
for NUM in 1 2 3 4 5 6 7; do pv < /dev/loop0p${NUM} > /dev/sdh${NUM}
```

All works! More in https://github.com/home-assistant/operating-system/issues/740#issuecomment-666229866.

agners, 2020-07-31:
@Nowaker hm, I see, so you omit partition 8 (data partition)  as well as partition 7 (OS overlay partition). Kinda clever! However, it does make assumptions of Home Assistant OS's partition layout, so if that changes in the future it won't work.

To be on the safe side, the snapshot method should be the method of choice for most users.

Nowaker, 2020-07-31:
@agners Of course. Technically it wasn't an assumption as I investigated partition tables and file systems of both your image and my SD card before proceeding. :) Specifically, I made sure that all partitions are the same size, and that each respective partition holds similar data.

That said, I think we could develop a migration tool that checks if a requested image is compatible with the partition layout on the SD card, and then do dd/pv for you after making a local backup. The tool would be external, not part of HA Core or HassOS. Seeing how many people struggle with Pi upgrades because snapshots don't always do the work for them (plus it's unneeded extra work compared to my solution), I think there's a need for an officially maintained tool like this. While I can start and implement it, I'm only willing to do it if there's a willingness for HA higherups to accept this is an official migration tool and have it covered in our documentation. @frenck @agners Can you guys discuss it internally with the team?

frenck, 2020-08-01:
Well, this would not cover all migration cases, which snapshot does cover. (e.g., migrating from a supervised to a OS based system, or the other way around).

Developing another method, adds a new level of maintenance, which IMHO isn't needed as tools are already in place.

agners, 2020-08-01:
@Nowaker I am with frenck, adding yet another method adds maintenance burden and can lead to confusion. If possible, extending the current solutions (snapshots) and address the shortcomings should be preferred.

I also think that the file-based backup approach (tarballs) has also technological advantages: It allows to resize partitions, it can support new partition layouts, new file systems, and it also helps in case the file system is corrupted.

If there are bugs in snapshots, those should be addressed, and create feature requests for missing features. One thing which annoys me personally with snapshot is that they are quite slow when large database is used. Maybe we could move away from gzip and/or use something which supports multi-threaded compression. Direct upload capability on the web interface would certainly also be nice.