home-assistant/operating-system #798: No sound card detected under vmware

SilverWx, 2020-08-02:
I'm using vhdx image under VmWare Workstation, I've tested to connect virtual sound card but Hassos doesnnot recognize it :

**HassOS release with the issue:**
4.11 

**Supervisor logs:**

**Journal logs:**

**Description of problem:**
Hardware description -> No sound card

serial:
/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2118359-if00
/dev/ttyACM0
/dev/ttyS2
/dev/ttyS3
/dev/ttyS1
/dev/ttyS0
input:
AT Translated Set 2 keyboard
VMware VMware Virtual USB Mouse
ImPS/2 Generic Wheel Mouse
Power Button
disk:
gpio:
audio:

SilverWx, 2020-08-02:
Commit  #770  seems to resolve it, wait for hassos update

pvizeli, 2020-08-18:
Fix with 5.1

closed 2020-08-18 by pvizeli