home-assistant/operating-system #799: BusyBox/ed crashes with illegal instruction on Ovirt using OVA QCOW2 4.11

derekatkins, 2020-08-02:
Running 'ed configuration.yaml' and I get an Illegal Instruction (core dumped).

**HassOS release with the issue:**

HassOS 4.11

**Supervisor logs:**

There is nothing in the log from the timeframe of the crash.

**Journal logs:**

config $ journalctl
-bash: journalctl: command not found

**Kernel logs:**

```
[446091.798254] traps: ed[15810] trap invalid opcode ip:55ba78f707d4 sp:7ffeaa206580 error:0 in busybox[55ba78eed000+9c000]
[446091.802066] audit: type=1701 audit(1596393071.567:119): auid=4294967295 uid=0 gid=0 ses=4294967295 subj==docker-default (enforce) pid=15810 comm="ed" exe="/bin/busybox" sig=4 res=1
```

**Description of problem:**
I am running HassOS 4.11 within my OVirt VM infrastructure.  I installed the OVA QCOW2 image and built the VM around that using the OVA configuration information (2G RAM, 2 CPUs).  The system has been up and running for a while.

I was trying to use 'ed' to modify the configuration.yaml (I hate vi, and you don't have emacs installed).  But all of a sudden it seems to crash:

```
config $ ed configuration.yaml 
"configuration.yaml", 47 lines, 1071 chars
: 2p
# Configure a default setup of Home Assistant (frontend, api, etc)
: s/#
ed: missing 2nd delimiter for substitute
: s/#//
Illegal instruction (core dumped)
```

It was not doing this when I first started using HassOS.  I do not see any issues in the VM Host logs.

derekatkins, 2020-10-05:
HI.  I upgraded to HassOS 4.13 and still see the problem.

derekatkins, 2021-03-11:
Still seeing this periodically in 5.10.  Will update shortly, but the issue is still there.
Wish I had a good way to reproduce reliably, or a way to get a crash dump to push into gdb when it does crash.

derekatkins, 2021-05-12:
I am still seeing this.  I did find a way to reproduce this behavior, however I am currently away from my main computer so cannot recreate it now; I will update next week when I get back to my HA instance.  IIRC, the issue shows up when trying to remove characters from a line.

derekatkins, 2022-01-04:
Yep, still an issue with HassOS 6.6 and the "Terminal & SSH" Add-On:
```
[core-ssh config]$ ed sensors.yaml
"sensors.yaml", 155 lines, 5495 chars
: $p
      unit_of_measurement: 'days'
: s/_of_//p
Illegal instruction (core dumped)
[core-ssh config]$
```

agners, 2022-01-04:
There is no `ed` on OS level. It seems you are using the Terminal Add-on. The Terminal runs in its own container which is Alpine Linux based.

But I can indeed reproduce the core dump here:
```
[core-ssh ~]$ echo "# test" > test
[core-ssh ~]$ ed test
"test", 1 lines, 7 chars
: s/#//
Trace/breakpoint trap (core dumped)
```

Moving to Add-on issue tracker.