home-assistant/operating-system #817: Dev Release 5.1 didn't generate an image for Raspi4

mattclar, 2020-08-08:
He @pvizeli I assume there is some error in the generation pipeline, but the latest 5.1dev release didnt include an image for raspi4 I was trying to manually update but it wasnt working obviously because it couldnt find the file!
Hopefully there is an easy fix.

mattclar, 2020-08-08:
It seems like 
"Fix RPi4 with buildroot v2020.02.4 (#815)"
Didn't make it into the release as it says this is the 1 commit to master after the 5.1 release, which is weird because the timing seems to indicate the release was generated after PR#815 was merged.... weird

liamstears, 2020-08-10:
Will 5.1 be generated for pi4? Will the 32 bit support usb boot on pi4?

pvizeli, 2020-08-10:
uploaded. 32bit support comes in future. I think until end of the year

closed 2020-08-10 by pvizeli

mattclar, 2020-08-11:
thanks @pvizeli , ledgend