home-assistant/operating-system #819: HassOS won't boot on Intel NUC when using large SSDs

stefangries, 2020-08-11:
**HassOS release with the issue:**
I tried 4.11 and 4.12, both not working

**Supervisor logs:**
Not booting, so no supervisor logs

**Journal logs:**
Not booting, so no journal logs

**Kernel logs:**
Not booting, so no kernel logs

**Description of problem:**

After flashing the NUC image to my new 2TB SSD I get this error when trying to boot:

[image]

After the error is displayed, I only get a black screen. If you then press and hold "Enter" you will see the following (slow motion):

https://www.dropbox.com/s/j98z78zt7jocs4b/hassos.mp4?dl=0

I already postet this issue in the Home Assistant Forum, there is another user with this problem. Seems to be related to large drives.

https://community.home-assistant.io/t/hassos-image-not-working-on-nuc-2tb-ssd/217563/2

I’m using a SanDisk SSD Plus 2 TB inside a BOXNUC7CJYH2. It’s a SATA SSD.

agners, 2020-08-18:
Hm, just tried here with Qemu and a 3TB disk, it seems to generally work. Also we do use GPT, which does support partitions/disks larger than 2TB, so this should really work.

I do get the "environment load ...: No such file or directory" error as well, so this seems not to be the root cause.

I wonder if the ubuntu boot entry still present in UEFI could cause the issue. Can you get rid of that entry from the UEFI (BIOS) firmware menu maybe? Is there another drive connected alongside?

agners, 2020-08-18:
I take this back. The system actually did boot the first time, but the second time the graphical output gets stuck like in your case and on the serial console I get this output:

```
BdsDxe: failed to load Boot0001 "UEFI QEMU DVD-ROM QM00003 " from PciRoot(0x0)/Pci(0x1,0x1)/Ata(Secondary,Master,0x0): Not Found
BdsDxe: loading Boot0002 "UEFI QEMU HARDDISK QM00001 " from PciRoot(0x0)/Pci(0x1,0x1)/Ata(Primary,Master,0x0)
BdsDxe: starting Boot0002 "UEFI QEMU HARDDISK QM00001 " from PciRoot(0x0)/Pci(0x1,0x1)/Ata(Primary,Master,0x0)

barebox 2020.04.0 #1 Thu Aug 6 21:53:10 UTC 2020

Board: barebox EFI payload
EFI v2.70 by EDK II v65536
EFI Event timer too slow freq = 100 Hz
acpi2: Found XSDT (OEM: BOCHS BX) with 5 entries
efi-fs handle-000000003ee28698: mounted on /boot
efi-block-io handle-000000003ee2a898: Cannot read MBR/partition table
malloc space: 0x2bf36000 -> 0x3bf35fff (size 256 MiB)
state: state failed to parse path to backend: No such device
initcall 0x0000003e4effea failed: No such device
description: UEFI QEMU DVD-ROM QM00003 
path: pci_root(0)/Pci(0x1,0x1)/Ata(Secondary,Master)
description: UEFI QEMU HARDDISK QM00001 
path: pci_root(0)/Pci(0x1,0x1)/Ata(Primary,Master)
description: UEFI PXEv4 (MAC:525400123456)
path: pci_root(0)/Pci(0x3,0x0)/Mac(525400123456,1)/IPv4(0.0.0.0,0,DHCP)
environment load /efivars/barebox-env-5b91f69c-8b88-4a2b-9269-5f1d802b5175: No such file or directory
Maybe you have to create the partition.
running /env/bin/init...
- Hit m for menu or wait for autoboot -
bootchooser: Cannot get state 'state'
Nothing bootable found on 'bootchooser'
Nothing bootable found

 HassOS boot Menu:
       1: Autoboot
       2: Boot System 0
       3: Boot System 1
       4: Shell
```

I resized the virtual disk as follows:

```
gunzip hassos_intel-nuc-4.12.img.gz
truncate --size 3TB hassos_intel-nuc-4.12.img
```

Then started using qemu:
```
qemu-system-x86_64 -m 1G -pflash /usr/share/edk2-ovmf/x64/OVMF.fd -drive file=hassos_intel-nuc-4.12.img.huge,format=raw,index=0,media=disk
```

agners, 2020-08-18:
It seems barebox can no longer load the kernel from system0/1:

```
Booting entry 'system0'
ext4 ext40: probe failed: Invalid argument
mount: Invalid argument
could not open /mnt/system/bzImage: No such file or directory
Booting entry 'system0' failed
Nothing bootable found
```

stefangries, 2020-08-19:
> I wonder if the ubuntu boot entry still present in UEFI could cause the issue. Can you get rid of that entry from the UEFI (BIOS) firmware menu maybe? Is there another drive connected alongside?

Thanks for your reply!

No other drive was connected at the same time. I flashed HassOS to the brand-new SSD before trying ubuntu, it had exactly the same error. The NUC had previously seen nothing but HassOS installations. After that I started experimenting with ubuntu.

Kjeldgaard, 2020-09-06:
I have the exact same problem. I also started writing HassOs images to the harddrive. After several failed attemps, I installed Ubuntu, which worked like a charm.

HW: Intel NUC NUC8CCHKR
Harddrive: eMMC 64GB
HassOS images tried: 2.12 and 4.12

bobhavlin, 2020-10-20:
Same problem here with vers 4.12 on virtualbox.

supratachophobia, 2020-10-24:
Im having this issue on normal 250GB Samsung SSD's. I've tried 2 of them.

4.13 and 4.14 both have this issue. 4.13 can sometimes be "fixed" by power button cycle, but not if you pull power from unit to reboot. It's DOA at that point and you need to reimage.

4.14 won't boot at all from NVME in SSD enclosure OR SATA SSD. Currently on 4.12 and scared to reboot.

supratachophobia, 2020-10-24:
Oh, also tried turning off UEFI but then the NUC won't "see" the HA image on the SSD and claims no boot device found. Secureboot is off, as well as wifi. Fan set to run all the time (thought heat might be a culprit but too specific/regular on the where it freezes).

rafkra, 2020-10-31:
> I have the exact same problem. I also started writing HassOs images to the harddrive. After several failed attemps, I installed Ubuntu, which worked like a charm.
> 
> HW: Intel NUC NUC8CCHKR
> Harddrive: eMMC 64GB
> HassOS images tried: 2.12 and 4.12

Same system and problem here. The Linux kernel of HassOS doesn't find the root partition. It doesn't seem to support eMMC. I installed a cheap M.2 SSD, which works fine.

supratachophobia, 2020-11-03:
> Same system and problem here. The Linux kernel of HassOS doesn't find the root partition. It doesn't seem to support eMMC. I installed a cheap M.2 SSD, which works fine.

Just to confirm now, it was a bad piece of RAM / wrong speed of RAM that was causing my issue described in the original post and your post as well. I did not try a M.2 SSD, only a NVMe via USB and 2 x Samsung 850's. None of which works until I swapped the RAM.

admered1, 2020-11-06:
I'm using the qcow2 formatted appliance, which has been working fine for a few months now. However, I rebooted it last night and now I get exactly the same as the headline image, word for word.

IN case it helps, I have attached the qcow2 image to another VM and mounted the first partition, the EFI one, the file it is looking for (/efivars/barebox-env-5b91f69c-8b88-4a2b-9269-5f1d802b5175) is not there.

tmm1, 2020-11-06:
Would be useful to run Barebox in debug mode so messages like this can be seen: https://github.com/saschahauer/barebox/blob/422069498f814b3fd02e2dd7ec231258865b0ca0/drivers/usb/storage/usb.c#L304

admered1, 2020-11-16:
As nobody seems keen on supporting the appliance, I have scrubbed the machine and rebuilt from scratch using a bog standard CentOS 8 install. I could get nothing back from the morass of different partitions in the qcow2 image, so could not even migrate to the CentOS machine, had to do the whole setup all over again and lost all the historical data.

Appliances are great for non-technical users ... until they blow up, then there needs to be someone who can support them and get people running again.

BigDoom, 2020-12-19:
Had the exact same issue; this post helped me: https://community.home-assistant.io/t/my-steps-to-success-for-hass-io-on-intel-nuc/43915/356

What I ended up doing was to take out the SSD from my NUC, connect it to my laptop, directly load the HassOS image for NUC onto the SSD, but it back into my NUC, booted and everything started like it should. 

Perhaps it works for you as well.

cveenman, 2021-01-28:
Still have the exact same problem. loading hassos 5.10 either from another pc or via Ubuntu on the NUC on either an SATA SSD 1 TB or M2 SSD  120 GB ends up in the same screen as posted in the first post. Any progress/ideas?

supratachophobia, 2021-02-02:
Mine was bad ram. Have you tested that thoroughly?

cveenman, 2021-02-02:
Looks like it was the Uefi boot. I took out all other instances in the bios menu and it worked.

GaryOkie, 2021-02-23:
I've been happily running Home Assistant (Intel NUC image) on a laptop with a Samsung EVO **512GB** SSD (UEFI) for several weeks.   Today I updated to Hass OS 5.12.   Then a few hours later, I inserted a new USB Zwave stick and the system hung.   Now it won't boot.   I'm seeing the same errors as the OP.  (Also had to take a video to capture the errors since the screen immediately goes to black).

I've run full diagnostics on the laptop.  Memory, etc., is fine.  512GB drive isn't all that large.    I have BIOS set up so that UEFI boot is the only option.   

 Is there a special keystroke to trigger **Barebox debug** mode (which was suggested)?

wroadd, 2021-03-08:
Same problem here. Interesting, that I can not see the "menu", but if I press 1 and enter, then the boot start. Then it fails to start the docker service and shut down itself.

agners, 2021-05-26:
This most likely is resolved by this change: https://github.com/saschahauer/barebox/commit/130fbc7f6c67357a58823c8f17ecef7d276386a9

The change is in Barebox 2021.05.0, so with the update to latest Barebox this issue should be resolved:
https://github.com/home-assistant/operating-system/pull/1380

Please try using HA OS 6.0.rc3.

closed 2021-05-26 by agners

arsenicks, 2021-06-15:
> This most likely is resolved by this change: [saschahauer/barebox@130fbc7](https://github.com/saschahauer/barebox/commit/130fbc7f6c67357a58823c8f17ecef7d276386a9)
> 
> The change is in Barebox 2021.05.0, so with the update to latest Barebox this issue should be resolved:
> #1380
> 
> Please try using HA OS 6.0.rc3.

Any way to update this when we're not able to boot ? I'm stuck with the appliance, exact same as @GaryOkie I plugged a zwave usb stick and now it's dead. My appliance is running on my ovirt cluster so I'm not sure how to fix it.. tried several things but can't get it to boot. At the best it boot once, crashed and never boot again.

agners, 2021-06-17:
Probably reinstalling from scratch is the easiest way to go.

It depends why your system is not booting exaclty, but if its related to the boot loader you can write a new boot partition/boot loader using this instructions:
https://github.com/home-assistant/operating-system/issues/1125#issuecomment-750457231

a3f, 2021-06-24:
> Any way to update this when we're not able to boot ?

barebox can update itself with the barebox_update command. For example if you insert a USB flash drive `usbdisk0` with the `barebox.efi` in the first partition:

```
/mnt/usbdisk0.0/barebox.efi
```

To start the new barebox executable, which should support your larger disk and there:

```
barebox_update /mnt/usbdisk0.0/barebox.efi
```

I have not (yet) used barebox with Home Assistant, so there might be some differences. HTH.