home-assistant/operating-system #820: Intel NUC Bluetooth Firmware missing? (ibt-17-16-1.sfi)

moeffju, 2020-08-11:
**HassOS release with the issue:** HassOS 4.12

**Kernel logs:**
```
bluetooth hci0: Direct firmware load for intel/ibt-17-16-1.sfi failed with error -2
bluetooth: hci0: Failed to load Intel firmware file (-2)
```

**Description of problem:**

I cannot use Bluetooth on my intel NUC because HassOS does not contain the needed firmware  for either the builtin bluetooth chip or an external bluetooth dongle (RealTek 0bda:8771 USB stick – rtl8761b or rtl8761a), and I have no idea where to “correctly” put the firmware.

basnijholt, 2020-09-07:
I just posted this in https://github.com/home-assistant/operating-system/issues/774 too.

I have the same problem and I am 100% sure that Bluetooth used to work. @pvizeli, do you know if something changed recently?

moeffju, 2020-09-09:
Thanks, I must’ve missed that when searching for bluetooth firmware. Unfortunately I’m not familiar enough with the build pipeline for the hassio images to debug this further or try to fix it. I saw that bluetooth support has been an issue a few times before (https://github.com/home-assistant/operating-system/pull/563) for driver and/or firmware reasons.

I also tried adding an external dongle, but hassos doesn’t support that either - I don’t know if that’s something that can be installed via package or whether it needs a full update of the OS image?

If there is a simple way to use a Raspberry Pi Zero W as a bluetooth “head” I would appreciate any pointers to guides for that, what I found seemed very confusing with the duplicate setup and mqtt...

basnijholt, 2020-09-10:
@moeffju, are you able to execute:
```
ha os update --version 4.11
```
For me this raises an error, but maybe it works for you. Then we could figure out whether the bug is introduced in 4.12 or even earlier.

moeffju, 2020-09-10:
@basnijholt No luck:

```
➜  ~ ha os update --version 4.11
Processing... Done.

Error:
➜  ~
```

Re-running with loglevel debug shows that a `status="400 Bad Request" statuscode=400` is returned with `body="{\"result\": \"error\", \"message\": \"\"}"`.

moeffju, 2020-09-10:
Just for completeness’ sake, `bluetoothctl` says it can not find any controllers – neither the built-in Intel bluetooth chipset nor the USB-connected BT5 dongle.

basnijholt, 2020-09-11:
On Discord, where I asked for suggestions @pvizeli said:
> we don't remove firmware. Maybe intel have update his binary blob with compatibility issues :person_shrugging:  it's all magic around https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git and linux-firmware buildroot packages

Now I got a bit further. I ran
```
docker exec -it homeassistant /bin/bash  # to go into host OS
apk add linux-firmware
```
and then it installed many packages:
```
bash-5.0# apk add linux-firmware
fetch http://dl-cdn.alpinelinux.org/alpine/v3.12/main/x86_64/APKINDEX.tar.gz
fetch http://dl-cdn.alpinelinux.org/alpine/v3.12/community/x86_64/APKINDEX.tar.gz
(1/86) Installing linux-firmware-yamaha (20200519-r0)
(2/86) Installing linux-firmware-yam (20200519-r0)
(3/86) Installing linux-firmware-vxge (20200519-r0)
(4/86) Installing linux-firmware-vicam (20200519-r0)
(5/86) Installing linux-firmware-ueagle-atm (20200519-r0)
(6/86) Installing linux-firmware-ttusb-budget (20200519-r0)
(7/86) Installing linux-firmware-tigon (20200519-r0)
(8/86) Installing linux-firmware-ti-keystone (20200519-r0)
(9/86) Installing linux-firmware-ti-connectivity (20200519-r0)
(10/86) Installing linux-firmware-tehuti (20200519-r0)
(11/86) Installing linux-firmware-sxg (20200519-r0)
(12/86) Installing linux-firmware-sun (20200519-r0)
(13/86) Installing linux-firmware-slicoss (20200519-r0)
(14/86) Installing linux-firmware-sb16 (20200519-r0)
(15/86) Installing linux-firmware-rtw88 (20200519-r0)
(16/86) Installing linux-firmware-rtlwifi (20200519-r0)
(17/86) Installing linux-firmware-rtl_nic (20200519-r0)
(18/86) Installing linux-firmware-rtl_bt (20200519-r0)
(19/86) Installing linux-firmware-rtl8192e (20200519-r0)
(20/86) Installing linux-firmware-rsi (20200519-r0)
(21/86) Installing linux-firmware-rockchip (20200519-r0)
(22/86) Installing linux-firmware-radeon (20200519-r0)
(23/86) Installing linux-firmware-r128 (20200519-r0)
(24/86) Installing linux-firmware-qlogic (20200519-r0)
(25/86) Installing linux-firmware-qed (20200519-r0)
(26/86) Installing linux-firmware-qcom (20200519-r0)
(27/86) Installing linux-firmware-qca (20200519-r0)
(28/86) Installing linux-firmware-ositech (20200519-r0)
(29/86) Installing linux-firmware-nvidia (20200519-r0)
(30/86) Installing linux-firmware-netronome (20200519-r0)
(31/86) Installing linux-firmware-myricom (20200519-r0)
(32/86) Installing linux-firmware-mwlwifi (20200519-r0)
(33/86) Installing linux-firmware-mwl8k (20200519-r0)
(34/86) Installing linux-firmware-mrvl (20200519-r0)
(35/86) Installing linux-firmware-moxa (20200519-r0)
(36/86) Installing linux-firmware-microchip (20200519-r0)
(37/86) Installing linux-firmware-meson (20200519-r0)
(38/86) Installing linux-firmware-mellanox (20200519-r0)
(39/86) Installing linux-firmware-mediatek (20200519-r0)
(40/86) Installing linux-firmware-matrox (20200519-r0)
(41/86) Installing linux-firmware-liquidio (20200519-r0)
(42/86) Installing linux-firmware-libertas (20200519-r0)
(43/86) Installing linux-firmware-korg (20200519-r0)
(44/86) Installing linux-firmware-keyspan_pda (20200519-r0)
(45/86) Installing linux-firmware-keyspan (20200519-r0)
(46/86) Installing linux-firmware-kaweth (20200519-r0)
(47/86) Installing linux-firmware-isci (20200519-r0)
(48/86) Installing linux-firmware-intel (20200519-r0)
(49/86) Installing linux-firmware-inside-secure (20200519-r0)
(50/86) Installing linux-firmware-imx (20200519-r0)
(51/86) Installing linux-firmware-i915 (20200519-r0)
(52/86) Installing linux-firmware-go7007 (20200519-r0)
(53/86) Installing linux-firmware-ess (20200519-r0)
(54/86) Installing linux-firmware-ene-ub6250 (20200519-r0)
(55/86) Installing linux-firmware-emi62 (20200519-r0)
(56/86) Installing linux-firmware-emi26 (20200519-r0)
(57/86) Installing linux-firmware-edgeport (20200519-r0)
(58/86) Installing linux-firmware-e100 (20200519-r0)
(59/86) Installing linux-firmware-dsp56k (20200519-r0)
(60/86) Installing linux-firmware-dpaa2 (20200519-r0)
(61/86) Installing linux-firmware-dabusb (20200519-r0)
(62/86) Installing linux-firmware-cxgb4 (20200519-r0)
(63/86) Installing linux-firmware-cxgb3 (20200519-r0)
(64/86) Installing linux-firmware-cpia2 (20200519-r0)
(65/86) Installing linux-firmware-cis (20200519-r0)
(66/86) Installing linux-firmware-cavium (20200519-r0)
(67/86) Installing linux-firmware-cadence (20200519-r0)
(68/86) Installing linux-firmware-brcm (20200519-r0)
(69/86) Installing linux-firmware-bnx2x (20200519-r0)
(70/86) Installing linux-firmware-bnx2 (20200519-r0)
(71/86) Installing linux-firmware-av7110 (20200519-r0)
(72/86) Installing linux-firmware-atusb (20200519-r0)
(73/86) Installing linux-firmware-atmel (20200519-r0)
(74/86) Installing linux-firmware-ath9k_htc (20200519-r0)
(75/86) Installing linux-firmware-ath6k (20200519-r0)
(76/86) Installing linux-firmware-ath10k (20200519-r0)
(77/86) Installing linux-firmware-ar3k (20200519-r0)
(78/86) Installing linux-firmware-amdgpu (20200519-r0)
(79/86) Installing linux-firmware-amd-ucode (20200519-r0)
(80/86) Installing linux-firmware-amd (20200519-r0)
(81/86) Installing linux-firmware-advansys (20200519-r0)
(82/86) Installing linux-firmware-adaptec (20200519-r0)
(83/86) Installing linux-firmware-acenic (20200519-r0)
(84/86) Installing linux-firmware-3com (20200519-r0)
(85/86) Installing linux-firmware-other (20200519-r0)
(86/86) Installing linux-firmware (20200519-r0)
OK: 720 MiB in 257 packages
```

Is this expected? Or should this already be installed?

CuBiC3D, 2020-09-13:
Same here, worked before but suddenly stopped without touching HassOS version after restart with the missing firmware. `/lib/firmware` is empty on the host

CuBiC3D, 2020-09-13:
@basnijholt You installed the package inside the homeassistant docker container and there are no volume mounts for libs on the host. `linux-firmware` needs to be on the host running buildroot.

CuBiC3D, 2020-09-13:
Okay after some monologue in discord I found the source for my problem: I extracted some hassOS images and found that NUC images do contain some firmware, but the one I use OVA does not and I took a look in some versions - none did.

Since my BT had worked before yesterdays restart without, there was something wrong: I migrated some months ago from Arch Linux as host OS to hassOS and I did not powered off the host since, so the intel BT module had still the firmware loaded from Arch Linux. After yesterdays powerdown the module's RAM flushed the firmware and there was none available to load after reboot from hassOS.

So this leads to: I'm missing `linux-firmware` in OVA images.

moeffju, 2020-09-17:
Is there a way to install this firmware on the side? I tried running the `apk` line from above on the image and rebooting but it didn’t seem to make a difference.

moeffju, 2020-09-28:
Just for kicks, I tried to get this to work with Debian 10 & Home Assistant Supervised. Also no dice, because somehow the Debian Linux 5.8.0 kernel does not want to load the `rtl8761b_fw.bin` firmware file, even though https://github.com/torvalds/linux/blob/v5.8/drivers/bluetooth/btrtl.c suggests that the correct detection would be in there.

moeffju, 2020-09-28:
Update on the update: Got it to work with the built-in BT.

- Install Debian 10 in Proxmox Qemu
- Pass through the intel USB BT device to the guest VM (will show as Unknown, probably)
- Install the linux-firmware packages on the guest
- Run `bluetoothctl scan on` and it should work
- Install Home Assistant Supervised following the instructions

So really the only thing missing to get it to work on HassOS is putting the firmware files in and making sure the kernel config is right.

basnijholt, 2020-09-28:
Great work! Will you stay on Debian? Or will you try to get it to work on Home Assistant OS?

moeffju, 2020-09-28:
I’ll stay on Debian for now, but kept the other VM around. Unfortunately I have no clue yet how to start tackling the issue on HassOS and don’t have the time right now to understand how the images are built. If anyone has pointers I’m happy to take a look in a week or so.

endrift, 2020-10-06:
If the HassOS OVA image had a way to install packages like `linux-firmware` this seems like it would be an easy fix. Unfortunately the image seems to be pretty static apart from the Docker containers, and you can't really install it inside of Docker and expect the host to be able to load the file from there. Short of copying the files out and making an overlay in `/lib/firmware` (which I haven't figured out how to do yet; I know a number of overlays are used but I don't know where they're described), I don't think this can be fixed without intervention from the Home Assistant developers.

SPCulhane, 2020-10-18:
I just updated to HassOS 4.14 hoping it would be working again, nope :-( I feel like it was only a few months ago when it started to work in the first place? I'm running esxi on a NUC with the OVA image.

moeffju, 2020-10-18:
If one of the devs could point to some resources about building the images, that would be great. I looked through https://github.com/home-assistant/operating-system/blob/dev/Documentation/bluetooth.md and https://github.com/home-assistant/operating-system/blob/dev/Documentation/development.md but there’s not a lot of detail to go on.

vBrolin, 2020-10-20:
I'm having the same issue on a NUC image

SPCulhane, 2020-12-27:
Its been a couple of months and still no luck with the latest HassOS 5.9. I've been debating or not to do a fresh install using the latest OVA file but my gut is telling me I will be in the boat.

mpepping, 2021-02-06:
HassOS 5.10 as well.

moeffju, 2021-02-06:
It would be really good to hear something, anything from the team on this - should we expect any changes here, has it even been seen?

brianewman, 2021-02-24:
+1.  I would like to use bluetooth on Intel NUC with ESXi.  Need firmware in the vmdk image.

streilu, 2021-04-12:
Looks like it should be included by default in the upcoming release.

See commit https://github.com/home-assistant/operating-system/commit/23fb6b8dd32588841b8b931dfbffeda482a5f322

agners, 2021-06-05:
It seems like the native Intel NUC image (now Generic x86-64) had the firmware file in question (`ibt-17-16-1.sfi`) since quite a while. But the virtual image (which is what the original poster @moeffju is using from what I understand) did not contain that firmware.

With enabling `BR2_PACKAGE_LINUX_FIRMWARE_IBT` for OVA this is fixed with PR #1283, so 6.0.rc3 should already have everything needed. Hence this issue can be closed.

As for `rtl8761b_fw.bin`, this is a different firmware and should be tracked in a separate issue. It came up in another thread as well and is now tracked with #1392.

LuckyPants703, 2021-09-28:
The issue still remains. It has been working with some versions, but the problem has returned.

My system is NUC7PJYH (latest firmware 0065) with Home Assistant OS 6.4 and core-2021.9.7. The message is:
[    1.058941] bluetooth hci0: Direct firmware load for intel/ibt-17-16-1.sfi failed with error -2

brianewman, 2021-09-28:
I have been running the Intel NUC image on ESXi since right after my last comment (February) and have had no issues.

agners, 2021-09-29:
@LuckyPants703 is Bluetooth actually not working or just that message your seeing? The message has been printed very early, presumably because the Bluetooth driver is built-in, but at that time the root file system is not available yet, leading to error -2 (No such file or directory). The OS should load the firmware later.

The firmware is definitely deployed in latest 6.4 HAOS:

```
# find /usr -name ibt-17-16-1.sfi
/usr/lib/firmware/intel/ibt-17-16-1.sfi
# ha os info
board: generic-x86-64
boot: A
disk_data: .
update_available: false
version: "6.4"
version_latest: "6.4"
```

LuckyPants703, 2021-09-29:
@agners It is not working. When I connect to the machine via ssh and type in "bluetoothctl" it says:
No default controller available

Doesn't show anything with the scan either...

It's good news that firmware is there so it must be possible to resolve it somehow.

agners, 2021-09-29:
Ok. Change #1558 will make sure the driver is a module. Firmware loading should then succeed on first try, let's see.

basnijholt, 2021-09-29:
I can also confirm that Bluetooth has been (still) broken on my NUC for more than a year now.

agners, 2021-10-05:
This is fixed with HAOS 6.5 (currently on beta channel).

closed 2021-10-05 by agners