home-assistant/operating-system #821: Bluetooth Drivers Realtek rtl 8761a missing

SilverWx, 2020-08-12:
HassOS 4.12

**Description of problem:**
Drivers for bluetooth 5 adapter based on Realtek rtl 8761a is missing while is currently supported by linux 

[image]

Drivers can be downloaded here : https://mpow.s3-us-west-1.amazonaws.com/mpow_MPBH456AB_driver+for+Linux.tgz

pvizeli, 2020-08-18:
- This is for bugs only. Feature and enhancement requests should go in our community forum: https://community.home-assistant.io/c/feature-requests

closed 2020-08-18 by pvizeli

SilverWx, 2020-08-19:
It's not really a feature since bluetooth dongle are supported, the problem is all  bluetooth 5 dongle are not supported because of drivers missing