home-assistant/operating-system #822: Add xen guest tools

rb2k, 2020-08-17:
We already seem to ship VMWare's open-vm-tools with the operating system, it would be great if we could also add xen guest utilities. This would allow for a clean shutdown and better metrics.

The discussion came up here already:
https://github.com/home-assistant/operating-system/pull/265#issuecomment-671254522

The XCP-NG project's wiki on the topic:
https://xcp-ng.org/docs/guests.html#linux

The official repo is:
https://github.com/xenserver/xe-guest-utilities

I guess a fix would look similar to https://github.com/home-assistant/operating-system/pull/505/files

resistr, 2020-08-18:
Not adding large value to this thread; but echoing the need for this. Thanks!

pvizeli, 2020-08-18:
- This is for bugs only. Feature and enhancement requests should go in our community forum: https://community.home-assistant.io/c/feature-requests

closed 2020-08-18 by pvizeli

rb2k, 2020-08-18:
Moved it to https://community.home-assistant.io/t/add-xen-guest-tools-package-to-hass-os/219573