home-assistant/operating-system #823: HASSOS V5.1 can't boot from USB SSD on RPI4 4gb, 64bit

BerniApple, 2020-08-17:
Hello everybody,
i have just flashed the last version of HassOS (v5.1) on an SSD using balena etcher, i have connected it to my raspi pi 4 via usb but nothing happen, it doesn't boot up.
Where I'm wrong?

thanks in advance

McGiverGim, 2020-08-17:
You need to update the eeprom of your raspberry pi too. Have you done that?

BerniApple, 2020-08-17:
> You need to update the eeprom of your raspberry pi too. Have you done that?

nope, how can i do that?

BerniApple, 2020-08-17:
> You need to update the eeprom of your raspberry pi too. Have you done that?

pi@raspberrypi:~ $ vcgencmd bootloader_version
Apr 16 2020 18:11:26
version a5e1b95f320810c69441557c5f5f0a7f2460dfb8 (release)
timestamp 1587057086

McGiverGim, 2020-08-17:
Look at this: https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md you need to install the latest "stable" version (by default is configured for critical).

BerniApple, 2020-08-17:
> Look at this: https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md you need to install the latest "stable" version (by default is configured for critical).

pi@raspberrypi:~ $ sudo rpi-eeprom-update -a
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTFS /boot
BOOTLOADER: up-to-date
CURRENT: Fri 31 Jul 13:43:39 UTC 2020 (1596203019)
 LATEST: Fri 31 Jul 13:43:39 UTC 2020 (1596203019)
 FW DIR: /lib/firmware/raspberrypi/bootloader/stable
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1

should it works?

closed 2020-08-17 by BerniApple