home-assistant/operating-system #825: Enable I2C on the Home Assistant Operating System via import from usb does not work

ArdescoConsulting, 2020-08-19:
**Home Assistant release with the issue:**
Home Assistant 0.114.2
HassOS 4.12

**Operating environment (HassOS):**
arch | armv7l
chassis | embedded
dev | false
docker | true
docker_version | 19.03.11
hassio | true
host_os | HassOS 4.12
installation_type | Home Assistant OS
os_name | Linux
os_version | 4.19.127-v7l
python_version | 3.8.3
supervisor | 234
timezone | Europe/Brussels
version | 0.114.2
virtualenv | false

Raspberry PI 4 B

**Supervisor logs:**
20-08-19 07:22:58 INFO (SyncWorker_2) [supervisor.docker.interface] Clean hassio_multicast application
20-08-19 07:22:58 INFO (MainThread) [supervisor.plugins.dns] Stop CoreDNS plugin
20-08-19 07:22:58 INFO (SyncWorker_0) [supervisor.docker.interface] Stop hassio_dns application
20-08-19 07:23:03 INFO (SyncWorker_0) [supervisor.docker.interface] Clean hassio_dns application
20-08-19 07:23:03 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.systemd1.Manager.Reboot on /org/freedesktop/systemd1
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing... 
[07:23:29] INFO: Update udev information
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
20-08-19 07:23:32 INFO (MainThread) [__main__] Initialize Supervisor setup
20-08-19 07:23:33 INFO (MainThread) [supervisor.bootstrap] Initialize Supervisor Sentry
20-08-19 07:23:33 INFO (MainThread) [supervisor.bootstrap] Setup coresys for machine: raspberrypi4
20-08-19 07:23:33 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attach to Supervisor homeassistant/armv7-hassio-supervisor with version 234
20-08-19 07:23:33 INFO (MainThread) [__main__] Setup Supervisor
20-08-19 07:23:33 INFO (MainThread) [supervisor.utils.gdbus] Connect to dbus: org.freedesktop.systemd1 - /org/freedesktop/systemd1
20-08-19 07:23:33 INFO (MainThread) [supervisor.utils.gdbus] Connect to dbus: org.freedesktop.hostname1 - /org/freedesktop/hostname1
20-08-19 07:23:33 INFO (MainThread) [supervisor.utils.gdbus] Connect to dbus: de.pengutronix.rauc - /
20-08-19 07:23:33 INFO (MainThread) [supervisor.utils.gdbus] Connect to dbus: org.freedesktop.NetworkManager - /org/freedesktop/NetworkManager/DnsManager
20-08-19 07:23:33 INFO (MainThread) [supervisor.host.info] Update local host information
20-08-19 07:23:33 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/hostname1
20-08-19 07:23:33 INFO (MainThread) [supervisor.host.services] Update service information
20-08-19 07:23:33 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.systemd1.Manager.ListUnits on /org/freedesktop/systemd1
20-08-19 07:23:33 INFO (MainThread) [supervisor.host.network] Update local network DNS information
20-08-19 07:23:33 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/NetworkManager/DnsManager
20-08-19 07:23:33 INFO (MainThread) [supervisor.host.sound] Update PulseAudio information
20-08-19 07:23:33 INFO (MainThread) [supervisor.host.apparmor] Load AppArmor Profiles: {'hassio-supervisor'}
20-08-19 07:23:33 INFO (MainThread) [supervisor.host.services] Reload local service hassos-apparmor.service
20-08-19 07:23:33 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.systemd1.Manager.ReloadOrRestartUnit on /org/freedesktop/systemd1
20-08-19 07:23:33 INFO (SyncWorker_0) [supervisor.docker.interface] Attach to homeassistant/armv7-hassio-dns with version 9
20-08-19 07:23:33 INFO (MainThread) [supervisor.misc.forwarder] Start DNS port forwarding to 172.30.32.3
20-08-19 07:23:33 INFO (MainThread) [supervisor.plugins.dns] Start CoreDNS plugin
20-08-19 07:25:31 INFO (SyncWorker_0) [supervisor.docker.dns] Start DNS homeassistant/armv7-hassio-dns with version 9 - 172.30.32.3
20-08-19 07:25:31 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
20-08-19 07:25:31 INFO (SyncWorker_0) [supervisor.docker.interface] Attach to homeassistant/armv7-hassio-audio with version 17
20-08-19 07:25:31 INFO (MainThread) [supervisor.plugins.audio] Start Audio plugin
20-08-19 07:25:36 INFO (SyncWorker_0) [supervisor.docker.audio] Start Audio homeassistant/armv7-hassio-audio with version 17 - 172.30.32.4
20-08-19 07:25:36 INFO (SyncWorker_0) [supervisor.docker.interface] Attach to homeassistant/armv7-hassio-cli with version 25
20-08-19 07:25:36 INFO (MainThread) [supervisor.plugins.cli] Start cli plugin
20-08-19 07:25:42 INFO (SyncWorker_0) [supervisor.docker.cli] Start CLI homeassistant/armv7-hassio-cli with version 25 - 172.30.32.5
[... 37 lines cut ...]
20-08-19 07:26:49 INFO (MainThread) [supervisor.host.info] Update local host information
20-08-19 07:26:49 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/hostname1
20-08-19 07:26:49 INFO (MainThread) [supervisor.updater] Fetch update data from https://version.home-assistant.io/stable.json
20-08-19 07:26:49 INFO (MainThread) [supervisor.host.services] Update service information
20-08-19 07:26:49 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.systemd1.Manager.ListUnits on /org/freedesktop/systemd1
20-08-19 07:26:50 INFO (MainThread) [supervisor.host.network] Update local network DNS information
20-08-19 07:26:50 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/NetworkManager/DnsManager
20-08-19 07:26:50 INFO (MainThread) [supervisor.host.sound] Update PulseAudio information
20-08-19 07:29:03 INFO (MainThread) [supervisor.api.security] /hardware/info access from core_ssh
20-08-19 07:29:37 INFO (MainThread) [supervisor.api.security] /hardware/info access from core_ssh
20-08-19 07:29:58 INFO (MainThread) [supervisor.api.security] /hardware/info access from core_ssh
20-08-19 07:30:15 INFO (MainThread) [supervisor.api.security] /hardware/info access from core_ssh
20-08-19 07:33:14 INFO (MainThread) [supervisor.api.security] /supervisor/logs access from core_ssh
20-08-19 07:33:47 INFO (MainThread) [supervisor.hassos] Syncing configuration from USB with HassOS.
20-08-19 07:33:47 INFO (MainThread) [supervisor.host.services] Restart local service hassos-config.service
20-08-19 07:33:47 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.systemd1.Manager.RestartUnit on /org/freedesktop/systemd1
20-08-19 07:34:01 INFO (MainThread) [supervisor.api.security] /supervisor/logs access from core_ssh

**Description of problem:**
If followed the instruction https://www.home-assistant.io/hassio/enable_i2c/ to the letter.
Tried with multiple sticks in different usb ports.
Clicking the import from USB button just turns it green for 1 second and nothing else happens.
There is a log entry 20-08-19 07:33:47 INFO (MainThread) [supervisor.hassos] Syncing configuration from USB with HassOS
But I2C not available.
The only way to get I2C to work is by doing the updates to the files on the first partition of the SD card via another linux system.

agners, 2020-08-31:
From what I can tell the script responsible to copy the `config.txt` file is actually copying a source file named `boot-env.txt`:
https://github.com/home-assistant/operating-system/blob/dev/buildroot-external/board/raspberrypi/hassos-hook.sh#L12

@pvizeli I guess this comes from the fact that the file is named `boot-env.txt` in buildroot source. It feels to me that `config.txt` would be more user friendly, but not sure if people use `boot-env.txt` today already...

Misiu, 2020-09-02:
@ArdescoConsulting I have the same problem.
Did you follow the instruction carefully? Did you create a folder named `/modules` (the name should contain slash at the beginning)?
Info from here: https://community.home-assistant.io/t/mcp23017-i-o-expander/170486/20?u=misiu

Problem is, that on Windows we can't create folders with a slash in name.
Can this be changed, so both names (with and without a slash) will work?

This was asked in WTH: https://community.home-assistant.io/t/wth-why-enabling-i2c-is-so-hard/219771

closed 2020-09-03 by agners

Misiu, 2020-09-03:
@agners I'll check that at home. Thank you for updating the docs!
There is still one wrong link on that page. More details in [WTH thread](https://community.home-assistant.io/t/wth-why-enabling-i2c-is-so-hard/219771/14?u=misiu).