home-assistant/operating-system #826: Invalid OVF checksum algorithm: SHA256

JohnGauthier, 2020-08-19:
**HassOS release with the issue:**
4.12 OVA, downloaded 8/19/2020

**Description of problem:**
VMware vCenter 7 (vSphere Client version 7.0.0.10600) error on deploying OVF template:

> _**The provided manifest file is invalid: Invalid OVF checksum algorithm: SHA256 .**_

vicfergar, 2020-08-29:
I have followed [this steps](https://medium.com/@rootedshell/failed-to-open-virtual-machine-sha1-digest-of-file-xxxxxxxx779-disk1-vmdk-does-not-match-manifest-7d0d94db4451) and it works for me :wink:

Apocrathia, 2020-12-05:
Deleting the manifest file didn't work. Now getting the following error:

> Issues detected with selected template. Details: - 71:7:VALUE_ILLEGAL: Value ''3'' of Parent element does not refer to a ref of type DiskControllerReference.

This issue is still present in 4.19 OVA.

DennisFaucher, 2020-12-09:
Same issue 

Issues detected with selected template. Details: - 71:7:VALUE_ILLEGAL: Value ''3'' of Parent element does not refer to a ref of type DiskControllerReference.

B3DTech, 2020-12-14:
So the OVA that Home Assistant releases was designed for VMWare Player, or Workstation, or a consumer version.  On ESXi/vSphere, it has no concept of a SATA controller.  Only IDE or SCSI.  Extract the OVF and VMDK.  Change the controller section starting at line 55 to:

      <Item>
        <rasd:Address>0</rasd:Address>
        <rasd:Caption>SCSIController</rasd:Caption>
        <rasd:Description>SCSI Controller</rasd:Description>
        <rasd:ElementName>SCSIController</rasd:ElementName>
        <rasd:InstanceID>3</rasd:InstanceID>
        <rasd:ResourceSubType>lsilogic</rasd:ResourceSubType>
        <rasd:ResourceType>6</rasd:ResourceType>
      </Item>

Then you can either repackage the OVA, or use the OVF/VMDK to create your VM.

DennisFaucher, 2020-12-14:
Thank you. I will try this and report back.

DennisFaucher, 2020-12-14:
Same error (
Selected OVA file is not a valid template.) . Maybe I did not package the OVA properly. I unzipped it, changed that section and re-zipped. Attached changed file. (Added .TXT to the name)
[home-assistant.ovf.txt](https://github.com/home-assistant/operating-system/files/5691894/home-assistant.ovf.txt)

B3DTech, 2020-12-14:
You need to recalculate the sha256 hash of the ovf, edit the mf file with it, then repackage it.

DennisFaucher, 2020-12-15:
Decided to dry a supported Type 2 Hypervisor instead. Failed to import the OVA into VMware Fusion v11. 
[image]
[image]

sohai9, 2021-01-05:
> So the OVA that Home Assistant releases was designed for VMWare Player, or Workstation, or a consumer version. On ESXi/vSphere, it has no concept of a SATA controller. Only IDE or SCSI. Extract the [...]

1. Export **hassos_ova-5.9** using 7.zip.
2. Delete .mf file
3. Follow the steps by @B3DTech mentioned above
4. Go to vCenter (My version is 7.0.1) and Right click on the VM -> Edit Settings -> VM Options -> General Options
5. Change "Guest OS" to **Linux** and "Guest OS Version" to **Other Linux (64-bit)**

DrSeussFreak, 2021-01-18:
I also had to edit line 25, and make it say `<OperatingSystemSection ovf:id="101" vmw:osType="otherLinux64Guest">` BEFORE I could import it into my vSphere 7 farm.  I know you say to edit it after import

sl0n, 2021-01-27:
I repackaged it as per instructions from @B3DTech and @DrSeussFreak, thanks guys!
Here is the OVA file that I was able to add with  vSphere 6.7: 

https://yadi.sk/d/I4wGW9_BeSjFXg
md5sum: 463b5ada97ec6df18ae600339b94b411

DennisFaucher, 2021-01-27:
Nice work. I just installed on ESXi 7.0.1 and it works flawlessly. Thank you.

Gedankenleser, 2021-08-19:
Can we just remove the .mf File? The checksums are correct but it does not work. I don't know why. 
https://github.com/home-assistant/operating-system/blob/5bb42ff43c22b8cac87ac19fc096905ec0587d60/buildroot-external/board/pc/ova/hassos-hook.sh#L36

DrSeussFreak, 2021-08-19:
> Can we just remove the .mf File? 

the manifest file is not truly required, but from a security standpoint, I like seeing it.   That said, I cannot speak to what else would be impacted for the install in terms of other references or checks.  Probably none, but...

You can always install ovftool and generate a .mf via the cli.  It is extremely simple, I've done it a bunch, and a great blogger has a simple article about it --> https://williamlam.com/2012/01/how-to-create-manifest-file-for-ovf.html

countbuggula, 2021-12-13:
This is definitely still a problem with the ova that's provided on the webpage for download/install.  Please either update that ova with the repacked version that works or just remove the download to stop confusing people.

agners, 2021-12-13:
@countbuggula as far as I remember there is no one way to make it work for Virtual Box (which requires sha256) and VMWare (which seems to hate sha256). IMHO, just use software which supports sha256 :)

countbuggula, 2021-12-13:
> @countbuggula as far as I remember there is no one way to make it work for Virtual Box (which requires sha256) and VMWare (which seems to hate sha256). IMHO, just use software which supports sha256 :)

I don't think so - the ova in [this post](https://github.com/home-assistant/operating-system/issues/826#issuecomment-768196657) worked just fine in my esxi server - I just finished deploying it.  It's just HA 5.10 while the current version is 7.0

agners, 2021-12-14:
So reportedly HAOS 7.0 ova image works with ESXi version: 7.0.1. What version are you using?

> I don't think so - the ova in this post worked just fine in my esxi server

I am not worried about ESXi Server, I worry that Virtual Box won't work anymore... That said, as @DrSeussFreak already pointed out, it seems that nowadays Virtual Box don't have problems importing without manifest file.

However, the manifest file has a purpose: It makes sure that the disk file is coherent.

One thing I just realized is that our manifest file has a space after the algorithm, and it seems that the error reports that "SHA256 " (with space) is an invalid checksum algorithm:

```
SHA256 (home-assistant.mf) = e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855
SHA256 (home-assistant.ovf) = c181d57827d975995e28a455127a8e09fb87fb1be37f678b103fe09c5e4e6ada
SHA256 (home-assistant.vmdk) = 61809204f1226bf46617737cc6aefd8fdb24225afd31899d2e8424c31b78d7ca
```

I generate some nightly builds without that space, let's see if that resolves the problem with you ESXi Server installation.

agners, 2021-12-14:
@countbuggula can you try the `ova` file from this development build?

https://os-builds.home-assistant.io/8.0.dev2021121401674/

countbuggula, 2021-12-14:
> @countbuggula can you try the `ova` file from this development build?
> 
> https://os-builds.home-assistant.io/8.0.dev2021121401674/

Same SHA digest error in VMware Workstation for some reason.   In ESXi it got past that and gave a new different error:
71:7:VALUE_ILLEGAL: Value "3" of Parent element does not refer to a ref of type DiskControllerReference.

agners, 2021-12-14:
> Same SHA digest error in VMware Workstation for some reason. 

What VMware Workstation version are you using? Can you post a screenshot maybe or/and the exact message (particularly, I wonder if there is a space after SHA256 and the dot at the end of the message)?

> In ESXi it got past that and gave a new different error:
> 71:7:VALUE_ILLEGAL: Value "3" of Parent element does not refer to a ref of type DiskControllerReference.

Seems that this has been discovered earlier already:

https://github.com/home-assistant/operating-system/issues/826#issuecomment-744734477

It seems that VirtualBox supports SCSI, and it doesn't seem to matter really (from https://www.virtualbox.org/manual/ch05.html):

> Whether you use SATA, SCSI, or SAS does not make any real difference.

countbuggula, 2021-12-14:
VMware Workstation 16 Pro
16.2.1 build-18811642
[image]

countbuggula, 2021-12-14:
> > In ESXi it got past that and gave a new different error:
> > 71:7:VALUE_ILLEGAL: Value "3" of Parent element does not refer to a ref of type DiskControllerReference.
> 
> Seems that this has been discovered earlier already:
> 
> [#826 (comment)](https://github.com/home-assistant/operating-system/issues/826#issuecomment-744734477)
> 
> It seems that VirtualBox supports SCSI, and it doesn't seem to matter really (from https://www.virtualbox.org/manual/ch05.html):
> 
> > Whether you use SATA, SCSI, or SAS does not make any real difference.

Yep, since Workstation and VirtualBox both support SCSI, I support changing the default.  I know esxi isn't a super common use case but I know there are others who like me run a vsphere homelab.  In my case, since those servers are running 24/7 anyways, it makes sense to make things like this as a VM instead of using separate hardware.

agners, 2021-12-14:
@countbuggula this build now uses SCSI LSI Logic as controller type:

https://os-builds.home-assistant.io/8.0.dev2021121401674/haos_ova-8.0.dev2021121401674.ova

countbuggula, 2021-12-14:
> @countbuggula this build now uses SCSI LSI Logic as controller type:
> 
> https://os-builds.home-assistant.io/8.0.dev2021121401674/haos_ova-8.0.dev2021121401674.ova

Awesome, that works perfectly in esxi now.  Still getting the sha error in Workstation, which is weird, but esxi was my primary use case for this so I'm happy to call it good for now.  Thank you so much for helping figure this out!

agners, 2022-01-12:
The fix #1674 got also backported to 7.1.

closed 2022-01-12 by agners

quinnell, 2024-06-29:
This issue seems to be happening again. I experienced it attempting to install HAOS 12.3 and 12.4 with the .ova file on VMWare ESXi 8.0 U2

TiTidom-RC, 2024-07-04:
Same problem here with ESXi 8.0 U2 and HAOS 12.4

nicojmb, 2024-08-13:
Same problem with ESXI 8.0.0, 20513097 and HAOS 12.4 ova

CaptainHindsigh, 2024-10-07:
Also with ESXI 7 and haos_ova-13.1.ova, SHA-256 checksum error. 
Took me days to find out there was nothing wrong with transfer or HDD.

Bazoogle, 2024-11-19:
> and how did you solve it? because adding the vmdk disk works but this way I can't make snapshots.

I was able to fix the issue by unzipping the ova file with 7zip, and then opened the .mf file with a text editor. It had two SHA-256 hashes, and after checking the file hashes, they matched. The hashes were formatted like this:
```
SHA2-256(home-assistant.ovf)= 64504b3ae7dd5bf32d74187874538bcf2ea204e1ce0cfbbee399e00ed46d4982
SHA2-256(home-assistant.vmdk)= 3cbdc7cb6267f308e089920f86324553061bf841faabdb1c45fb8889ac19fe25
```
All I did was delete the 2 in `SHA2-256` so the final result was:
```
SHA-256(home-assistant.ovf)= 64504b3ae7dd5bf32d74187874538bcf2ea204e1ce0cfbbee399e00ed46d4982
SHA-256(home-assistant.vmdk)= 3cbdc7cb6267f308e089920f86324553061bf841faabdb1c45fb8889ac19fe25
```
I then selected all 3 files, and added them to an archive with 7zip. The archive format should be "tar" and you'll want to rename the .tar to .ova again. Upload that file and see if it works for you.

oetiker, 2025-03-14:
I finally found the solution how to create `.ova` files which do not cause complaints about checksums.

Use `create-vmdk` from <https://github.com/vmware/open-vmdk/tree/master/vmdk> instead of `qemu-img`  to build the `vmdk` file.

Now things will just work. Vmware will not complain about checksums anymore when loading the ova.

agners, 2025-04-11:
FWIW, this new breakage is addressed with #3981.