home-assistant/operating-system #832: Automatic deletion of hassio-supervisor image in offline environment

theFork, 2020-08-24:
We are using Home Assistant running on a Raspberry Pi in an offline environment.    
After setting everything up at home, the Pi was transported to the offline environment and everything worked like a charm. :-)    
    
However, after hard-rebooting the device by pulling the plug, which is common practice for many Pi users, Home Assistant broke:     
The device responds to `ping` but that's it. No web-interface, nor SSH (Terminal addon was installed and working before).    
                         
After some debugging we found that the supervisor Docker image was missing.        
We suspect the following block of `/usr/sbin/hassos-supervisor`:    
```    
# Check if previous runtime of the container meets the minimal runtime    
# If not, we might be in trouble. Image or container corruption, bad release?    
# Delete the container, delete the image, pull a fresh one    
if [ -n "${SUPERVISOR_CONTAINER_ID}" ] && [ "${SUPERVISOR_IMAGE_ID}" = "${SUPERVISOR_CONTAINER_ID}" ]; then    
    SUPERVISOR_START=$(docker inspect --format='{{.State.StartedAt}}' hassio_supervisor | sed -re "s/([-0-9]+)T([0-9\:]+).*/\1 \2/g")    
    SUPERVISOR_STOP=$(docker inspect --format='{{.State.FinishedAt}}' hassio_supervisor | sed -re "s/([-0-9]+)T([0-9\:]+).*/\1 \2/g")    
    
    if [ -n "${SUPERVISOR_START}" ] && [ -n "${SUPERVISOR_STOP}" ]; then    
        START=$(date --date="${SUPERVISOR_START}" +%s)    
        STOP=$(date --date="${SUPERVISOR_STOP}" +%s)    
        SUPERVISOR_RUNTIME=$((STOP-START))    
        # Minimial runtime not met, remove container and image    
        if [ "${SUPERVISOR_RUNTIME}" -lt "${SUPERVISOR_MIN_RUNTIME}" ]; then    
            echo "[WARNING] Supervisor container exited too quickly, forcing a fresh image and container..."    
            docker container rm --force hassio_supervisor || true    
            SUPERVISOR_CONTAINER_ID=""    
            docker rmi --force "${SUPERVISOR_IMAGE_ID}" || true    
            SUPERVISOR_IMAGE_ID=""    
        fi    
    fi    
fi    
```    
    
What do you think about verifying that we have a working internet connection before `docker rm`ing stuff? ;-)    
Maybe `ping`ing some well-known server?

frenck, 2020-08-25:
It really doesn't matter, because if it drops into this logic, it was broken to being with.
So RM-ing stuff or not... internet or not, it is broken.

haggl, 2020-09-10:
Not really. The code which decides if the supervisor image is dumped doesn't verify a checksum or anything. It just looks at the latest start and finish time of the container. So I was thinking: "Maybe pulling the plug leads to faulty timestamps."

I just tested my theory:

1. Commented out everything in the innermost `if`-block, except for the log message.
2. Booted with Internet connection.
  => supervisor is downloaded and started as expected.
3. Pulled supply voltage and Ethernet cable and rebooted.
  => Seems to have done no harm. Supervisor starts and I don't see the warning message in the journal.
4. Pulled supply voltage and rebooted again. => BAM!
   ```
   # journalctl -b -u hassos-supervisor | grep WARNING
   Sep 10 12:00:48 homeassistant hassos-supervisor[748]: [WARNING] Supervisor container exited too quickly, forcing a fresh image and container...
  
   # docker inspect --format='{{.State.StartedAt}}' hassio_supervisor
   2020-09-12T12:00:51.298970144Z
   # docker inspect --format='{{.State.FinishedAt}}' hassio_supervisor
   2020-09-12T12:00:37.278974993Z
   ```
All services are started and seem to function as expected. After plugging in the Ethernet cable I can access the web interface, login, ... Everything is fine.

I suggest either checking if we can actually download a fresh image before making things worse or using checksums to determine if the image is corrupt instead of timestamps.

Yes, turning any computer off through a power-cut is a bad idea. But it can happen at any time due to power failures and it's common practice for single-board PC users.

agners, 2020-11-03:
Not having a reliable timestamp available is something the OS should account for.

What we basically do in the script is trying to figure out the runtime of the container to infer if the supervisor successfully started. Probably better would be to actually check if the container runs properly. Docker has a [healthcheck mechanism](https://docs.docker.com/engine/reference/builder/#healthcheck) integrated, maybe that is worth a try?

pvizeli, 2020-11-03:
The supervisor uses s6-overlay to archive this, so we have a working health and watchdog on the supervisor container

agners, 2020-11-26:
For some reason I forgot to tag this issue, but this problem is actually addressed by #952, so should be fine from 5.5 onwards.

closed 2020-11-26 by agners