home-assistant/operating-system #835: Hassos 4.12 not booting on odroid n2+

pawelrohde, 2020-08-26:
**HassOS release with the issue:** 4.12

**Supervisor logs:** none

**Journal logs:** none

**Kernel logs:** none

**Description of problem:**
New odroid n2+ and new SD SanDisk 64MB A2
- image from github written by balentaEther

No boot, no display on hdmi

I take Ubuntu 18.04 lts and write to the same SD card - working without problems - booting and displaying on hdmi login prompt. Ethernet was configured VIA DHCP and i can login VIA ssh.

I have another card - the same - Ubuntu booting, hassos not.

pvizeli, 2020-08-27:
Do you boot with MMC switch or from internal bootloader? Can you post the serial output from UART board interface / maybe you need an connecter: https://www.hardkernel.com/shop/usb-uart-module-kit/

pvizeli, 2020-08-27:
Can you try 4.11 ?

pawelrohde, 2020-08-27:
> Can you try 4.11 ?

Yes, and 4.11 is working 

> Do you boot with MMC switch or from internal bootloader? Can you post the serial output from UART board interface / maybe you need an connecter: https://www.hardkernel.com/shop/usb-uart-module-kit/

No, I don't have this stuff, but maybe arduino can I connect as serial monitor?

pawelrohde, 2020-08-27:
> Can you try 4.11 ?

Is better, but not good ...
See pictures from my phone: https://photos.app.goo.gl/dEL6C8nM5YdUhJuV6
I was logging into system, typed os update and after reboot - black screen, no boot process any more...

closed 2020-08-28 by pvizeli

pawelrohde, 2020-09-02:
Please, tell me what I can do with this commit?
My odroid is still not working with hassos.
Please help me with installation.

agners, 2020-09-02:
@pawelrohde you'll have to wait for the next release. Meanwhile you can use the 4.11 relesae.

scottbob09, 2020-10-09:
There is a trick.  I think this works (trying the new release now)
If you are stuck at the MMC / craig init failed.  unplug power.
switch to external boot switch the extremely breakable little bitty switch towards the display connector.
if you have no external media, or snapped it in half while trying to remove the power...
remove it.  remove anything like a zwave stick that has a battery inside it.
press the boot button.   yes, unplugged.
wait 5 seconds
switch the extremely breakable switch back towards the white spring post (internal boot)
plug in power.
watch it boot.
To be "that guy" I have booted in the wrong position, waited for it to fail... switched the switch, then pressed the boot button.
It seems fine.  I have boot logs from the last time I was going to comment.  But I'll snag the console logs from the upgrade and boot.

scottbob09, 2020-10-10:
The upgrade went about as most have gone.  Bad.  I am up and running on 4.13 and will let it burn in and do a backup.
The files are ordered by the number in which they were captured.
[1-first-upgrade-to-reboot.txt](https://github.com/home-assistant/operating-system/files/5358206/1-first-upgrade-to-reboot.txt)
[2-something good in here.txt](https://github.com/home-assistant/operating-system/files/5358207/2-something.good.in.here.txt)
[3-reboot after 1st good upgrade.txt](https://github.com/home-assistant/operating-system/files/5358208/3-reboot.after.1st.good.upgrade.txt)
[4-bad boot after ordered reboot post upgrade.txt](https://github.com/home-assistant/operating-system/files/5358209/4-bad.boot.after.ordered.reboot.post.upgrade.txt)
[5-bad boot after upgrade-auto-downgrade with good bootfollowing.txt](https://github.com/home-assistant/operating-system/files/5358210/5-bad.boot.after.upgrade-auto-downgrade.with.good.bootfollowing.txt)
[6-success-4.12-4.13.txt](https://github.com/home-assistant/operating-system/files/5358211/6-success-4.12-4.13.txt)

PR #836 (2020-08-28): Fix ODROID-N2(+) SD card boot (#835)
Fix: #835