home-assistant/operating-system #837: System has no internet access after reboot and changed IP

juergen-venne, 2020-08-29:
**HassOS release with the issue:**

Home Assistant 0.114.4

**Supervisor logs:**

20-08-29 09:52:51 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.INITIALIZE' start 0 add-ons
20-08-29 09:52:51 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SYSTEM' start 0 add-ons
20-08-29 09:52:51 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SERVICES' start 2 add-ons
20-08-29 09:52:53 INFO (SyncWorker_3) [supervisor.docker.addon] Start Docker add-on homeassistant/armv7-addon-ssh with version 8.6.0
20-08-29 09:54:39 INFO (SyncWorker_2) [supervisor.docker.addon] Start Docker add-on homeassistant/armv7-addon-samba with version 9.2.0
20-08-29 09:54:44 INFO (SyncWorker_1) [supervisor.docker.interface] Start homeassistant
20-08-29 09:54:45 INFO (MainThread) [supervisor.homeassistant] Wait until Home Assistant is ready
20-08-29 09:55:00 INFO (MainThread) [supervisor.api.security] /host/logs access from core_ssh
20-08-29 09:55:01 INFO (MainThread) [supervisor.homeassistant] Updated Home Assistant API token
20-08-29 09:55:02 INFO (MainThread) [supervisor.api.security] /host/logs access from core_ssh
20-08-29 09:55:03 INFO (MainThread) [supervisor.api.security] /host/logs access from core_ssh
20-08-29 09:55:04 INFO (MainThread) [supervisor.api.security] /host/logs access from core_ssh
20-08-29 09:55:05 INFO (MainThread) [supervisor.api.security] /host/logs access from core_ssh
20-08-29 09:55:05 INFO (MainThread) [supervisor.api.security] /host/logs access from core_ssh
20-08-29 09:55:06 INFO (MainThread) [supervisor.api.security] /host/logs access from core_ssh
20-08-29 09:55:06 INFO (MainThread) [supervisor.api.security] /host/logs access from core_ssh
20-08-29 09:55:06 INFO (MainThread) [supervisor.homeassistant] Detect a running Home Assistant instance
20-08-29 09:55:06 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.APPLICATION' start 2 add-ons
20-08-29 09:55:07 INFO (MainThread) [supervisor.api.security] /host/logs access from core_ssh
20-08-29 09:55:08 INFO (SyncWorker_0) [supervisor.docker.addon] Start Docker add-on homeassistant/armv7-addon-configurator with version 5.0.0
20-08-29 09:55:09 INFO (SyncWorker_2) [supervisor.docker.addon] Start Docker add-on hassioaddons/node-red-armv7 with version 7.1.0
20-08-29 09:55:12 INFO (MainThread) [supervisor.api.security] /host/logs access from core_ssh
20-08-29 09:55:13 INFO (MainThread) [supervisor.api.security] /dns/info access from a0d7b954_nodered
20-08-29 09:55:14 INFO (MainThread) [supervisor.misc.tasks] All core tasks are scheduled
20-08-29 09:55:15 INFO (MainThread) [supervisor.misc.hwmon] Started Supervisor hardware monitor
20-08-29 09:55:15 INFO (MainThread) [supervisor.core] Supervisor is up and running
20-08-29 09:55:15 INFO (MainThread) [supervisor.host.info] Update local host information
20-08-29 09:55:15 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/hostname1
20-08-29 09:55:15 INFO (MainThread) [supervisor.updater] Fetch update data from https://version.home-assistant.io/stable.json
20-08-29 09:55:15 WARNING (MainThread) [supervisor.updater] Can't fetch versions from https://version.home-assistant.io/stable.json: Cannot connect to host version.home-assistant.io:443 ssl:default [Address not available]
20-08-29 09:55:15 INFO (MainThread) [supervisor.host.services] Update service information
20-08-29 09:55:15 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.systemd1.Manager.ListUnits on /org/freedesktop/systemd1
20-08-29 09:55:15 INFO (MainThread) [supervisor.host.network] Update local network DNS information
20-08-29 09:55:15 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/NetworkManager/DnsManager
20-08-29 09:55:15 INFO (MainThread) [supervisor.host.sound] Update PulseAudio information
20-08-29 09:55:15 INFO (MainThread) [supervisor.api.security] /host/logs access from core_ssh
20-08-29 09:55:29 INFO (MainThread) [supervisor.api.security] /host/logs access from core_ssh
20-08-29 09:55:36 INFO (MainThread) [supervisor.api.security] /host/logs access from core_ssh
20-08-29 09:55:38 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-08-29 09:55:38 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-08-29 09:55:38 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
20-08-29 09:56:19 INFO (MainThread) [supervisor.api.security] /core/check access from core_ssh
20-08-29 09:56:19 INFO (SyncWorker_1) [supervisor.docker] Run command 'python3 -m homeassistant -c /config --script check_config' on homeassistant/raspberrypi4-homeassistant
20-08-29 09:56:28 INFO (MainThread) [supervisor.homeassistant] Home Assistant config is valid
20-08-29 09:56:35 INFO (MainThread) [supervisor.api.security] /core/info access from core_ssh
20-08-29 09:56:50 INFO (MainThread) [supervisor.api.security] /core/options access from core_ssh
20-08-29 09:57:01 INFO (MainThread) [supervisor.api.security] /core/restart access from core_ssh
20-08-29 09:57:01 INFO (SyncWorker_0) [supervisor.docker.interface] Restart homeassistant/raspberrypi4-homeassistant
20-08-29 09:57:01 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 8:1000 is not str
20-08-29 09:57:01 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
20-08-29 09:57:08 INFO (MainThread) [supervisor.homeassistant] Wait until Home Assistant is ready
20-08-29 09:57:22 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-08-29 09:57:22 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-08-29 09:57:23 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
20-08-29 09:57:24 INFO (MainThread) [supervisor.homeassistant] Detect a running Home Assistant instance
20-08-29 09:57:36 INFO (MainThread) [supervisor.api.security] /core/rebuild access from core_ssh
20-08-29 09:57:36 INFO (SyncWorker_1) [supervisor.docker.interface] Stop homeassistant application
20-08-29 09:57:36 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 8:1000 is not str
20-08-29 09:57:36 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
20-08-29 09:57:43 INFO (SyncWorker_1) [supervisor.docker.interface] Clean homeassistant application
20-08-29 09:57:43 INFO (MainThread) [supervisor.homeassistant] Update pulse/client.config: /data/tmp/homeassistant_pulse
20-08-29 09:57:43 INFO (SyncWorker_2) [supervisor.docker.homeassistant] Start homeassistant homeassistant/raspberrypi4-homeassistant with version 0.114.4
20-08-29 09:57:43 INFO (MainThread) [supervisor.homeassistant] Wait until Home Assistant is ready
20-08-29 09:57:59 INFO (MainThread) [supervisor.homeassistant] Detect a running Home Assistant instance
[... 24 lines cut ...]
20-08-29 10:04:47 ERROR (SyncWorker_1) [supervisor.docker.interface] Can't install hassioaddons/ssh-armv7:7.5.0 -> 500 Server Error: Internal Server Error ("Get https://registry-1.docker.io/v2/: dial tcp 23.22.155.84:443: connect: connection refused").
20-08-29 10:05:47 INFO (MainThread) [supervisor.addons] Create Home Assistant add-on data folder /data/addons/data/core_ssh
20-08-29 10:05:47 INFO (SyncWorker_1) [supervisor.docker.interface] Pull image homeassistant/armv7-addon-ssh tag 8.6.0.
20-08-29 10:05:47 ERROR (SyncWorker_1) [supervisor.docker.interface] Can't install homeassistant/armv7-addon-ssh:8.6.0 -> 500 Server Error: Internal Server Error ("Get https://registry-1.docker.io/v2/: dial tcp 3.211.199.249:443: connect: connection refused").
20-08-29 10:06:21 INFO (SyncWorker_4) [supervisor.docker.interface] Restart homeassistant/raspberrypi4-homeassistant
20-08-29 10:06:22 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 8:1000 is not str
20-08-29 10:06:22 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
20-08-29 10:06:29 INFO (MainThread) [supervisor.homeassistant] Wait until Home Assistant is ready
20-08-29 10:06:44 INFO (MainThread) [supervisor.homeassistant] Detect a running Home Assistant instance
20-08-29 10:06:47 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
[... cut ...]
20-08-29 10:06:47 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running

**Journal logs:**

Where do I have to execute this command? I have no access to any command line in your system.

**Kernel logs:**

Where do I have to execute this command? I have no access to any command line in your system.

**Description of problem:**
The system has no acces to the internet anymore.
I assigned a static IP in my router and rebooted.
After that, the internet connection was lost. Network works, I can access the UI in my browser.

juergen-venne, 2020-08-30:
What a pitty really nice system does not work. What a pitty. I'll give the docker image a try.

closed 2020-08-30 by juergen-venne

juergen-venne, 2020-08-30:
I found the issue in my local network, but I will stay with the docker-image. :-)

agners, 2020-08-30:
@juergen-venne thanks for the update, appreciated! As long as its Home Assistant ;-)