home-assistant/operating-system #839: "No valid ingress session" spamming supervisor logs

pashdown, 2020-09-02:
**HassOS release with the issue:**
```
arch | x86_64
chassis | vm
dev | false
docker | true
docker_version | 19.03.11
hassio | true
host_os | HassOS 4.12
installation_type | Home Assistant OS
os_name | Linux
os_version | 5.4.56
python_version | 3.8.3
supervisor | 235
```

**Supervisor logs:**
```
20-09-01 23:58:52 WARNING (MainThread) [supervisor.api.ingress] No valid ingress session 057ad19beb8cd74180680fc7a37eccec2c7830922226b99697f444e03ad42404a82b9edb8036db31663e5cd0204aa3c68fe9f22088bad5ce2ff29e43b72853c7
20-09-01 23:58:55 WARNING (MainThread) [supervisor.api.ingress] No valid ingress session 057ad19beb8cd74180680fc7a37eccec2c7830922226b99697f444e03ad42404a82b9edb8036db31663e5cd0204aa3c68fe9f22088bad5ce2ff29e43b72853c7
20-09-01 23:58:58 WARNING (MainThread) [supervisor.api.ingress] No valid ingress session 057ad19beb8cd74180680fc7a37eccec2c7830922226b99697f444e03ad42404a82b9edb8036db31663e5cd0204aa3c68fe9f22088bad5ce2ff29e43b72853c7
20-09-01 23:59:04 WARNING (MainThread) [supervisor.api.ingress] No valid ingress session 057ad19beb8cd74180680fc7a37eccec2c7830922226b99697f444e03ad42404a82b9edb8036db31663e5cd0204aa3c68fe9f22088bad5ce2ff29e43b72853c7
20-09-01 23:59:10 WARNING (MainThread) [supervisor.api.ingress] No valid ingress session 057ad19beb8cd74180680fc7a37eccec2c7830922226b99697f444e03ad42404a82b9edb8036db31663e5cd0204aa3c68fe9f22088bad5ce2ff29e43b72853c7
20-09-01 23:59:10 WARNING (MainThread) [supervisor.api.ingress] No valid ingress session 057ad19beb8cd74180680fc7a37eccec2c7830922226b99697f444e03ad42404a82b9edb8036db31663e5cd0204aa3c68fe9f22088bad5ce2ff29e43b72853c7
20-09-01 23:59:15 WARNING (MainThread) [supervisor.api.ingress] No valid ingress session 057ad19beb8cd74180680fc7a37eccec2c7830922226b99697f444e03ad42404a82b9edb8036db31663e5cd0204aa3c68fe9f22088bad5ce2ff29e43b72853c7
```
**Journal logs:**
```
➜  ~ journalctl
zsh: command not found: journalctl

```

**Kernel logs:**
[dmesg.txt](https://github.com/home-assistant/operating-system/files/5159317/dmesg.txt)

**Description of problem:**

See "No valid ingress session" above.  It is logging this over and over, regardless of browser and browser cache.  My browser does have trouble loading anything in the supervisor, but I've also tried cleaning the cache in Firefox, and also using Safari and Chrome, with clean caches and fresh installs.

This started after my disk holding the qcow2 of hassio filled, and I've been unable to resolve it.  This morning, I did a backup of the configuration and a fresh install (on it's own partition for god's sake) and restored to it.  The error is still happening.  I also ran an "ha su repair" to no avail.

pashdown, 2020-09-04:
Found it.  My issue.

closed 2020-09-04 by pashdown

corvy, 2020-09-30:
What was the issue? I have this after migrating to a new host....

pashdown, 2020-09-30:
A workstation that was on and trying to load with an old cookie.

Fridolin34, 2021-03-18:
How did you find that out? I'm having this from time to time, but until now I have no clue where this is coming from.
Currently I'm trying to raise the log level, but no luck so far.

JamoDevNich, 2021-06-26:
If anyone is having this issue on Firefox, you could try clicking the shield 🛡️ button in the address bar, and switching off enhanced tracking protection. If the ingress addon now loads successfully, go ahead and switch tracking protection back on again.