home-assistant/operating-system #840: ODROID N2+ boot from USB- any chance ?

pepeEL, 2020-09-02:
Hi I know that on Odroid N2 is no available boot from USB (ex SSD). But is any chance to add it for ODROID N2+ like now is added for RPI4.... We download image operating system and only burn on SSD and then only connect to USB and boot. This will be very good solution also for ODROID N2 and also for ODROID N2+

Is any chance ?

agners, 2020-09-02:
Hm, from what I can see from [this](https://forum.odroid.com/viewtopic.php?t=34361) ODROID N2 thread, the boot ROM does not support direct boot from USB (just like RPi 4 essentially). Since there is now a SPI with petitboot, it should be possible to boot via pettitboot.

pepeEL, 2020-09-02:
On petiboot it is not working. On RPI4 boot from usb works perfect but only on version image x64.

agners, 2020-09-02:
@pepeEL yeah sorry I was a bit unclear, I am aware that USB boot does not work on N2 currently. All I am saying that in theory it is possible to boot from USB via petitboot, but that of course needs integration work in Home Assistant Operating System...

Also, since HA OS is using A/B partitioning for updates managed via U-Boot, we really need to be able to load U-Boot from petitboot, which might not be possible...

In any case, this is a feature request and should go to our community forum: https://community.home-assistant.io/c/feature-requests

closed 2020-09-02 by agners