home-assistant/operating-system #843: HassOS v5.1 unable to boot - Kernel Time Synchronised hung

drthanwho, 2020-09-02:
**HassOS release with the issue:** v5.1

**Kernel logs:**

Since I wasn't able to boot I could only take a picture of the screen to show the log https://imgur.com/a/c4SxXMr

**Description of problem:**

I flashed my SSD with the hassos_rpi4-64-5.1.img.gz image and plugged it in on my Rpi4 (4GB). The picture is where it has been stuck at. 

I don't know if I should provide anything else

tech-freak, 2020-09-03:
are you connected to internet ?
it will not proceed further if you are not connected to internet as it requires time (NTP) from google
use raspian with RTC  module or build local NTP server with PC/ESP32/8266 with RTC module and modify timesyncd.conf

[here ](https://github.com/home-assistant/supervisor/issues/504#issuecomment-397451918)they mentioned it will be stored in sd card in future update but still its not solved anything related NTP issue :/

agners, 2020-09-06:
It could be that the timesync issue is a prolem caused by not being able to mount the data partition. I think I have seen the same issue on N2+ here:

[image]

drthanwho, 2020-09-06:
There was definitely internet connected. But I think it might have been related to unsuccessful mounting since I also had issues trying to boot Raspberry OS from SSD before attempting HassOS stand-alone. 

Perhaps a incompatible sata to usb adapter. It would work maybe once after moving SD card to SSD but after a reboot it would never mount partition again.

pvizeli, 2020-09-06:
It hangs on waiting for the hassos-data partition on your SSD. Some adapter need quirks and not all are working

drthanwho, 2020-09-06:
That is what I tried adding to the cmdline.txt file for the Raspberry OS SSD boot. Although didn't see any improvement. Is it possible to add quirks for HassOS too?

agners, 2020-09-09:
@drthanwho can you test if this is still a problem for you with 5.2?

drthanwho, 2020-09-09:
@agners Swapped to an external SSD and 5.1 was able to boot just fine. Only comments I have is that it seems to take it's time (given it's an SSD) during reboot of the host. But installing addons I can see it's way faster to my SD card system. I don't know if the booting time is supposed to faster than what I'm getting or not. 

I will try 5.2 on the previous SSD that had the pronlem and update you

dweston, 2020-09-10:
@agners I've just booted a Rpi4 from SSD with hassos_rpi4-64-5.2.img. All OK. The speed is blazing comparative to the Rpi3.

carras, 2020-09-21:
I have a ORICO M.2 SATA case with a JMS580 chip, and I see it working with hassos_rpi4-64-5.2.img. Is there a ETA for stable HassOS 5? the U-Boot v2020.10 will help with this problems, won't it?

mattfro, 2020-09-29:
I don't have any logs or a screen to connect to my pi 4. But for me hassos_rpi4-64-5.2.img works and hassos_rpi4-5.2.img does not startup. Wanted to use 32 bit version because of the support for gpio. But I guess I can live without it.

drthanwho, 2020-09-29:
@mattfro

mattfro, 2020-09-29:
> @mattfro that is normal. Only the 64-bit version of HassOS currently supports USB boot

Ok Thanks for the explanation

agners, 2020-10-12:
@drthanwho I assume this is fixed as I haven't been able to reproduce on my end. If you still see it, it seems that #896 describes a similar issue and has already quite some information. In that case the system does reach a shell, but still has has the wait for time synchronization service stuck.

closed 2020-10-12 by agners