home-assistant/operating-system #844: WMIIR error at boot-up

giffordj, 2020-09-03:
**HassOS release with the issue:**

arch | x86_64
-- | --
chassis | vm
dev | false
docker | true
docker_version | 19.03.11
hassio | true
host_os | HassOS 4.12
installation_type | Home Assistant OS
os_name | Linux
os_version | 5.4.56
python_version | 3.8.3
supervisor | 235
timezone | America/Los_Angeles
version | 0.114.4
virtualenv | false

**Supervisor logs:**

[segfault.txt](https://github.com/home-assistant/operating-system/files/5165675/segfault.txt)

**Journal logs:**

**Kernel logs:**

[log.txt](https://github.com/home-assistant/operating-system/files/5165676/log.txt)

**Description of problem:**
Noticed segfault at boot up.

agners, 2020-09-09:
There are actually two segfaults, `ozwadmin` and `wmiir`. `ozwadmin` is OpenZWave related, and `wmiir` 9P file system related. Are those segfaults reproducible (e.g. on every boot)?

giffordj, 2020-09-11:
The OpenZwave is only once in a while, the WMIIR is all the time

closed 2021-05-12 by stale[bot]