home-assistant/operating-system #853: Homeassistant unavailable within 15 minutes when no lan cable is connected

FaserF, 2020-09-09:
**HassOS release with the issue:**

 Arch | armv7l
chassis | embedded
dev | false
docker | true
docker_version | 19.03.11
hassio | true
host_os | HassOS 4.12
installation_type | Home Assistant OS
os_name | Linux
os_version | 5.4.56
python_version | 3.8.3
supervisor | 242
timezone | Europe/Berlin
version | 0.114.4
virtualenv | false

**Supervisor logs:**

No errors available (as the system just shuts down)

**Journal logs:**

Command not found in the ssh addon? 

**Kernel logs:**

Will come - copy from ssh addon isn't pissible

**Description of problem:**
As my lan ports got limited I wanted to switch my Homeassistant OS installation from lan to wifi.
I could successfully enable a wifi connection. 
My issue is that it looks like the installation hangs up/shuts down automatically when there is no lan connection available? As I could perfectly use my wifi installation for around 15 minutes, then the whole Homeassistant installation wasn't available anymore. Tried it 4 times and everytime was the same, the os just got unavailable and I had to force reboot it.
Now with connecting a lan cable back it's working fine again.

I also tried fully disabling the lan port -> same

agners, 2020-09-09:
Which board are you using? It might be resolved by #848, can you try the 4.13 release?

FaserF, 2020-09-09:
I am using a Asus Tinkerboard. 
I have already installed the new 4.13 release. I will give it a try tomorrow

FaserF, 2020-09-10:
Sadly this did not fix it. Still the same issue.

FaserF, 2020-10-26:
I have sold my ASUS Tinkerboard - so the issue is no problem anymore for me.

closed 2020-10-26 by FaserF