home-assistant/operating-system #855: After update to 4.13 pi4 eth won't come up

olbjan, 2020-09-09:
I remotely updated the os on my pi4 4gb from 4.12 to 4.13 and unfortunately it hasn't come back up.
I've never had that happen before and unfortunately can't physically access the system for another week.

I tried (multiple times) to restart my pi4 remotely (it's powered via a poe splitter) but that hasn't helped. Port won't come up but it's drawing power.

Since I can't troubleshoot and wasn't able to find someone to contradict/repro I'm creating this issue in the hope it's a false alert and something else broke on my setup.

agners, 2020-09-09:
Just tried an update from 4.12 to 4.13 on a RPi4 2GB because I suspected it was related to https://github.com/home-assistant/operating-system/commit/a2a4cf8668c3c333b4641da264d5eed18a786179. However, all good here, update worked smoothly.

What I did saw some of the boards is that U-Boot sometimes was hanging. I sometimes had to reboot a couple of times until it works. Make sure to wait long enough (~1 minute) in between tries.

JayNewstrom, 2020-09-10:
I have a pi 4 4gb that I updated from 4.12 to 4.13 as well. It took one pull of the power plug for it to come back up.

olbjan, 2020-09-16:
Turns out this was bad timing with my poe switch apparently deciding it no longer wants to send enough power to the pi.

Closing the issue thus.

Thx to anyone who replied!

closed 2020-09-16 by olbjan