home-assistant/operating-system #856: Won't take

mcfarlde, 2020-09-10:
**HassOS release with the issue:**
Latest

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

20-09-10 02:32:30 ERROR (MainThread) [supervisor.hassos] HassOS update failed with: Installation error: Failed updating slot rootfs.0: Failed splicing data: Error reading from file: Input/output error
20-09-10 02:32:30 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-2837' coro=<HassOS.update() done, defined at /usr/src/supervisor/supervisor/hassos.py:125> exception=HassOSUpdateError()>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/hassos.py", line 159, in update
    raise HassOSUpdateError()
supervisor.exceptions.HassOSUpdateError

agners, 2020-09-11:
How did you update? Sounds like a broken SD card...

mcfarlde, 2020-09-12:
Power cycle and multiple resets and its all good now.

closed 2020-09-12 by mcfarlde