home-assistant/operating-system #857: Please add nano to HassOS

jodur, 2020-09-10:
**HassOS release with the issue:**

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

HassOS has currently only installed the vi-editor for editing files. In some case (for example) rapsberry pi we have to add additional lines to config.txt or cmdline.txt (usb-quirck for example) and the nano editor is more userfriendly for most people than the current vi editor.

carras, 2020-09-10:
It's just a question, but, can you install it on your own? (I mean using ssh) And if you do, in the future OS updates does it stay or you need to reinstall every time? Sorry, for taking you a bit offtopic. Thanks.

pvizeli, 2020-09-12:
Future requests please into forum

closed 2020-09-12 by pvizeli