home-assistant/operating-system #858: RPi4 WiFi won't connect when fast roaming enabled on Ubiquiti Wireless Network (even with 4.13)

shortbloke, 2020-09-10:
**HassOS release with the issue:**
Operating System: 4.13
Supervisor: 242
Home Assistant: 0.114.4

**Description of problem:**
Given I have a WiFi network using Ubiquiti Nano HD access points
When I enable Fast Roaming (802.11r) for the WLAN my RPi4
Then when the APs provisioning is complete, the RPi is no longer connected to the network

Restarting the RPi and the device still fails to connect to the wireless network. Nothing obvious is logged in the kernel log except a few entries related to power save being enabled:
```
[    9.623554] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
```

Disabling fast roaming and restarting the Pi and it connects to the WLAN without issue.
I've tried a clean install on a different Pi4 and it suffered the same problem

I had assumed the problem was caused by #733 however this is fixed in 4.13. I'm not seeing `invalid cipher group` in the log.

agners, 2020-09-11:
Hm, this is weird, I was able to reproduce with 802.11r enabled on OpenWrt, and it disappeared with 4.13. Can you maybe reflash/reconnect the WiFi to make sure everything else is fine?

Otherwise I guess there is something wrong with the 802.11r implementation fo Ubiquiti <=> RPi4... :-(

shortbloke, 2020-09-12:
Hi @agners I tried a fresh clean setup on another RPi4 doesn't connect to Ubiquiti with fast roaming enabled. 
I have downloaded and installed the latest RaspberyPi OS (Raspbian) 2020-08-20 kernel version 5.4, plus latest updates. Connected to my WiFi network switched on fast roaming and it continues to work. Boot the same Pi again from HassOS and it fails to connect until I switch off fast roaming.

Any tips on what logs to try and collect to aid debugging welcome.

brentpabst, 2021-01-07:
Hey there, still seeing similar issues here.  Know there was a fix for 802.11r but still seeing the issue with the following versions:

Version: 2020.12.7
OS: 5.9

Can I provide additional logs or info to troubleshoot?

closed 2021-05-12 by stale[bot]