home-assistant/operating-system #859: Please add TRIM on an external SSD for the RPi

k2v1n58, 2020-09-11:
**HassOS release with the issue:**
all

**Description of problem:**
It's not an issue more a functionality to add to HASS OS. TRIM offer better performance when we use a SSD because the SSD knows SSD which areas contain data that is no longer in use. It has been recently discovered how to activate it on RPi (https://www.jeffgeerling.com/blog/2020/enabling-trim-on-external-ssd-on-raspberry-pi).
Is it possible to activate in on HASS OS for RPi?

agners, 2020-09-11:
Note that sometimes trim also slows down operations since it takes a while on some SSDs (there are various bug reports in Fedora/Ubuntu, or also this LWN article https://lwn.net/Articles/787272/). I would also not expect massive performance gains.

That being said, it is good for the long term health and performance of a SSDs. Definitely a valuable addition for HAOS.

Feature requests should go to the Community category [Feature Requests](https://community.home-assistant.io/c/feature-requests/13). Closing here.

closed 2020-09-11 by agners

energywave, 2021-01-14:
I wanted only to add that TRIM activation is totally possible on Home Assistant OS. I wrote a step by step guide on my blog site here: https://henriksozzi.it/2021/01/ssd-con-trim-raspberry-pi4-procedura-ufficiale/
It's in Italian but I believe it's easily translatable to English with Google Translate or similar tools.
That would be a GREAT addition if HA integrated SSD data moving (datactl move) and TRIM activation in the web UI when it detect an external disk...