home-assistant/operating-system #860: Error after login: rlwrap: error: My terminal reports width=0 (is it emacs?)

snozzlebert, 2020-09-11:
**HassOS release with the issue:**
HassOS 4.13 (also HassOS 4.12) it is related to hassio_cli:26

**Supervisor logs:**
**Journal logs:**
**Kernel logs:**

**Description of problem:**

I'm running Home Assistant Operating in a VM (qemu/libvirt). My /boot/cmdline.txt  contains ```console=ttyS0``` to be able to access the console of the OS using ```sudo virsh console hassos```
This did work until an upgrade to 4.13 and a reboot yesterday. 
After a reboot I'm able to enter the console and I see:
```
Welcome to Home Assistant
homeassistant login:
```

After loging with  the user ```root``` the system gets stuck in a loop with the output of the ```ha``` help text and the following line:

```
rlwrap: error: My terminal reports width=0 (is it emacs?)  I can't handle this, sorry!
```

Sending an interrupt with ctrl+c, stopped the loop. But this results in termination of Serial Getty on ttyS0 and the console returns to the login prompt.
I think it is related to the change in https://github.com/home-assistant/plugin-cli/pull/15

Using ssh I was able to downgrade hass_cli to version 25 using the command:

```
ha cli update --version=25
```

After the downgrade I was able to login to the console again using root.
The plugin-cli project on Github doesn't have a issue tracker, that is why I post the issue here.

TonyApuzzo, 2020-09-16:
@snozzlebert thank you for posting this!  I had coincidentally switched to a new serial console hardware simultaneously with this error being introduced and I was going insane trying to "fix" my end without realizing a new version of the CLI add-on had been introduced.

snozzlebert, 2020-10-11:
I can confirm it's fixed in version 4.14

closed 2020-10-11 by snozzlebert

agners, 2020-10-12:
For the rel-5 series, 5.3 should fix this as well.