home-assistant/operating-system #861: Cant update to 4.13

camui-alexandru, 2020-09-11:
I cant update to 4.13. I get this în logs:

20-09-11 15:49:24 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-110818' coro=<HassOS.update() done, defined at /usr/src/supervisor/supervisor/hassos.py:125> exception=FileNotFoundError(2, 'No such file or directory')>

brotwi, 2020-09-15:
Tengo el mismo problema al intentar actualizar
20-09-15 12:46:46 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
20-09-15 12:53:24 WARNING (MainThread) [supervisor.hassos] Can't fetch versions from https://github.com/home-assistant/operating-system/releases/download/4.13/hassos_rpi3-4.13.raucb: 
20-09-15 12:53:24 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-127926' coro=<HassOS.update() done, defined at /usr/src/supervisor/supervisor/hassos.py:125> exception=HassOSUpdateError()>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/hassos.py", line 136, in update
    int_ota = await self._download_raucb(version)
  File "/usr/src/supervisor/supervisor/hassos.py", line 88, in _download_raucb
    raise HassOSUpdateError()
supervisor.exceptions.HassOSUpdateError

Muchas Gracias

FaserF, 2020-10-26:
Give it another try with the latest 4.15 release.

terencevs, 2020-10-26:
Same issue here, trying to update to 4.15 from 4.13:

`20-10-26 18:29:19 WARNING (MainThread) [supervisor.hassos] Can't fetch versions from https://github.com/home-assistant/operating-system/releases/download/4.15/hassos_rpi4-64-4.15.raucb: 
20-10-26 18:29:19 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-595' coro=<HassOS.update() done, defined at /usr/src/supervisor/supervisor/hassos.py:127> exception=HassOSUpdateError()>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/hassos.py", line 138, in update
    int_ota = await self._download_raucb(version)
  File "/usr/src/supervisor/supervisor/hassos.py", line 88, in _download_raucb
    raise HassOSUpdateError()
supervisor.exceptions.HassOSUpdateError`

dcole64, 2020-10-27:
Same issue upgrading from 4.13 to 4.15:

20-10-27 00:36:08 WARNING (MainThread) [supervisor.hassos] Can't fetch versions from https://github.com/home-assistant/operating-system/releases/download/4.15/hassos_rpi4-4.15.raucb: 
20-10-27 00:36:08 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-526545' coro=<HassOS.update() done, defined at /usr/src/supervisor/supervisor/hassos.py:127> exception=HassOSUpdateError()>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/hassos.py", line 138, in update
    int_ota = await self._download_raucb(version)
  File "/usr/src/supervisor/supervisor/hassos.py", line 88, in _download_raucb
    raise HassOSUpdateError()
supervisor.exceptions.HassOSUpdateError

closed 2021-05-12 by stale[bot]