home-assistant/operating-system #868: No IP (no network device) after update to 4.13

coldpenguin, 2020-09-15:
Hopefully I am in the correct repo. After updating the OS using the dashboard, the system didn't come back online. 
Investigation from the terminal was hampered by lack of familiar commands :(
I believe that the kernel module for the onboard gigabit ethernet is missing. 
I updated the OS before Core (which I have not done and am loath to do until knowing that it is the cause of the issue/this is fixed). 

$ha os info
board: odroid-xu4
boot: A
version: "4.13"
version_latest: "4.13"

~ $ ha info
arch: armv7
channel: stable
docker: 19.03.11
features:
- reboot
- shutdown
- services
- hostname
- hassos
hassos: "4.13"
homeassistant: 0.113.3
hostname: homeassistant
logging: info
machine: odroid-xu
operating_system: HassOS 4.13
supervisor: "243"
supported: true
supported_arch:
- armv7
- armhf
timezone: Europe/London

**Supervisor logs:**
Nothing pertinent to cause I think, completely full of loop: 
[[33m20-09-15 13:45:15 WARNING (MainThread) [supervisor.misc.tasks] Watchdog found a problem with Multicast plugin![[0m
[[32m20-09-15 13:45:15 INFO (MainThread) [supervisor.plugins.multicast] Start Multicast plugin[[0m
[[32m20-09-15 13:45:15 INFO (SyncWorker_7) [supervisor.docker.interface] Clean hassio_multicast application[[0m
[[32m20-09-15 13:45:17 INFO (SyncWorker_7) [supervisor.docker.multicast] Start Multicast homeassistant/armv7-hassio-mult
icast with version 3 - Host[[0m

**Journal logs:**
not journald :(

**Kernel logs:**
[dmesg.txt](https://github.com/home-assistant/operating-system/files/5225526/dmesg.txt)

**Description of problem:**

Using the UI, selected supervisor, and saw there was an update for Home Assistant OS (I believe). The version I was on I think was 4.12. I noted that there was a Home Assistant Core 0.114.4 available (I am still on 0.113.3 which would be the same vintage as the OS). 
Selected update.
Expected outcome: System reboots and new version is running.
After 20 minutes, still no ping, attached a monitor to the system. There are various pieces of information on USB etc. 
Running ip addr showed no expected devices for gigabit ethernet. 
The only semi-interesting line from dmesg appeared to be: [   36.969338] eth0: renamed from veth5f36723
Rebooted - no change. 
lspci not present (probably cos there isn't any)
lsusb shows: 
~ $ lsusb
Bus 005 Device 001: ID 1d6b:0002
Bus 003 Device 001: ID 1d6b:0002
Bus 002 Device 002: ID 10c4:ea60
Bus 001 Device 001: ID 1d6b:0002
Bus 003 Device 004: ID 0bda:8176
Bus 006 Device 001: ID 1d6b:0003
Bus 004 Device 001: ID 1d6b:0003
Bus 004 Device 002: ID 05e3:0616
Bus 002 Device 001: ID 1d6b:0001
Bus 003 Device 003: ID 046d:c31c
Bus 003 Device 002: ID 05e3:0610

lsmod:
~ $ lsmod
Module                  Size  Used by    Tainted: G
aes_arm_bs             24576  3
crypto_simd            16384  1 aes_arm_bs
cryptd                 24576  1 crypto_simd
bnep                   28672  2
xfrm_user              40960  1
rtl8192cu              90112  0
rtl_usb                20480  1 rtl8192cu
rtl8192c_common        61440  1 rtl8192cu
rtlwifi               122880  3 rtl8192cu,rtl_usb,rtl8192c_common
exynos_gsc             40960  0
s5p_jpeg               49152  0
v4l2_mem2mem           32768  2 exynos_gsc,s5p_jpeg
cp210x                 32768  0
s5p_mfc               151552  0
videobuf2_dma_contig    20480  3 exynos_gsc,s5p_jpeg,s5p_mfc
videobuf2_memops       16384  1 videobuf2_dma_contig
videobuf2_v4l2         24576  4 exynos_gsc,s5p_jpeg,v4l2_mem2mem,s5p_mfc
videobuf2_common       53248  5 exynos_gsc,s5p_jpeg,v4l2_mem2mem,s5p_mfc,videobuf2_v4l2
s5p_cec                16384  0

Any others having the issue: temporary workaround was to use a USB wifi adapter, after configuration using nmcli to the old IP (manual) it is working.

coldpenguin, 2020-09-15:
A symptom rather than a cause, integration with alexa through emulated hue appears to be non-functional. I seem to remember that it was a multicast/discovery protocol so it might not be playing nicely with the wireless, so probably nothing to do really with the issue above.
Noted here in case it does point to something silly.

agners, 2020-09-24:
@coldpenguin hm, can you confirm it works on 4.12? You can downgrade using `ha os update --version 4.12`

coldpenguin, 2020-09-25:
Yes that works. (The downgrade didn't 'complete' as such, I had to power the device off and on again. But that is a different problem). 
(I did of course try this step earlier. Couldn't get it to work. In the rush of things, forgot that it needed a network connection!)
LSUSB/LSMOD from 4.12, dmesg will be attached. I have highlighted a device, it looks up as the RTL8153 . There will be a missing USB wireless device, removed to confirm it was definitely ethernet based. 
https://devicehunt.com/view/type/usb/vendor/0BDA/device/8153

`~ $ lsusb
Bus 005 Device 001: ID 1d6b:0002
Bus 003 Device 001: ID 1d6b:0002
Bus 002 Device 003: ID 10c4:ea60
Bus 001 Device 001: ID 1d6b:0002
Bus 006 Device 001: ID 1d6b:0003
Bus 004 Device 001: ID 1d6b:0003
Bus 004 Device 002: ID 05e3:0616
Bus 002 Device 001: ID 1d6b:0001
Bus 003 Device 002: ID 05e3:0610
**Bus 006 Device 002: ID 0bda:8153** 
~ $ lsmod
Module                  Size  Used by    Not tainted
bnep                   28672  2
xfrm_user              40960  1
exynos_gsc             40960  0
s5p_jpeg               49152  0
v4l2_mem2mem           32768  2 exynos_gsc,s5p_jpeg
cp210x                 32768  0
s5p_mfc               151552  0
videobuf2_dma_contig    20480  3 exynos_gsc,s5p_jpeg,s5p_mfc
videobuf2_memops       16384  1 videobuf2_dma_contig
videobuf2_v4l2         24576  4 exynos_gsc,s5p_jpeg,v4l2_mem2mem,s5p_mfc
videobuf2_common       53248  5 exynos_gsc,s5p_jpeg,v4l2_mem2mem,s5p_mfc,videobuf2_v4l2
s5p_cec                16384  0
~ $
[dmesg2.txt](https://github.com/home-assistant/operating-system/files/5281680/dmesg2.txt)
`

insajd, 2021-01-05:
I am having same issue on Odroid XU4.
Previously when this happened, I wrote a fresh Odroid XU4 image of HA to card, booted it, and restored from backup. Then it's working for some time, a few days at most, then at one of the reboots, it loses Ethernet connectivity and rebooting doesn't help.
Did this 3 times already.

insajd, 2021-01-05:
When plugged in XU4 in the evening - everything booted up fine, connected to the network.

closed 2021-05-12 by stale[bot]