home-assistant/operating-system #869: HASSOS 4.13 Not booting from SSD / USB on RPI3

nathanmay, 2020-09-16:
**HassOS release with the issue:**
4.13

**Supervisor logs:**
Unable to gain access to GUI (web or console) - also not logs stored on SSD/USB

**Journal logs:**
Unable to gain access to GUI (web or console) - also not logs stored on SSD/USB

**Kernel logs:**
Unable to gain access to GUI (web or console) - also not logs stored on SSD/USB

I have been testing this for the last few days.
**My Setup:** 

- Raspberry Pi 3
- External 120gb Kingston SSD (only drive - no SSD)
- Previously running HASSOS 4.12

I have been running HA for many months and has upgraded the HASSOS last time one was released (4.11 to 4.12?) and this time attempted upgrade the OS to 4.13 as per supervisor notification. The RPI3 never came back up from the update (left for 5 hours).

I thought my SSD had failed however I have tested it and it is perfectly fine.

I decided to to try and install a fresh copy of HA (HASSOS 4.13) on a USB - however doesn't boot I simply get a black screen and no access to web GUI. 
Tried a fresh copy of the latest HA (HASSOS 4.13) on the SSD, again a black screen.
I tried install many fresh copies and leaving for different amounts of time including 2-3 hours and still not display and cant access web GUI. 

I then have tried installing latest version of Raspberry Pi OS  (previously called Raspbian) also PI-Hole on the SSD and boots up perfectly with no issues this lead me to believe it was HASSOS that was failing.

I then tried a fresh install of of a previous version of HASSOS (4.12) installed on the SSD and this boots up perfectly with no issues.

This leads me to believe something with HASSOS 4.13 is causing an issues and wont boot. 

However HASSOS 4.13 does works fine on a SD card - only an issue with SSD/USB boot.

Today I have left it for an hour and removed the SSD from the RPI and plugged into my computer hoping for some logs but nothing. 

Unfortunately I can't really offer anything else due to having no logs. I have been using SSD for months due to the performance benefit. 

Currently I have had to downgrade to HASSOS 4.12 and it is currently working perfect. I just cant upgrade to 4.13 successfully.

agners, 2020-09-20:
Just tried to reproduce this using a Raspberry Pi 3 Model B+: I installed 4.12 on a Samsung USB flash drive and booted successfully from USB what was interesting after reboot Linux was no longer able to detect the USB flash drive.

However, I then connected the flash drive behind a powered USB Hub and booting worked fine again. I then update to 4.13 and rebooted. It seems to work fine here, 4.13 booted and the web frontend was accessible again.

Can you try with a powered USB Hub?

Otherwise, best way to further debug this is by using a serial console to see what is going on after the 4.13 update. There are some high level information in our documentation (https://github.com/home-assistant/operating-system/tree/dev/Documentation/boards/raspberrypi), but any RPi related how-to should work.

nathanmay, 2020-09-26:
I don't have a USB powered hub to test unfortunately. 

Since this issue I have now moved my HA to a thin client / docker setup - Everything is now working perfect since moving away from RPI3b.

closed 2020-11-03 by agners