home-assistant/operating-system #870: hassos crashes after deconz starts with conbee 2 stick

accelle17, 2020-09-17:
**HassOS release with the issue:**4.13

arch | x86_64
-- | --
chassis | vm
dev | false
docker | true
docker_version | 19.03.11
hassio | true
host_os | HassOS 4.13
installation_type | Home Assistant OS
os_name | Linux
os_version | 5.4.56
python_version | 3.8.3
supervisor | 244
timezone | Asia/Manila
version | 0.114.4
virtualenv | false
hardware| NUCi38BEH (m.2 disk and 16 gb ram)
others|proxmox  6.2-11 with conbee 2 and huzbzb-1 passthrough
deconz addon| version 6.2.3
conbee 2 firmware| 26580700

**Supervisor logs:**

can't get it since VM is not responding even on pve console, not sure if we can configure crash dump on hassos; cant find old logs
**Journal logs:**

before the last instance:
Sep 17 01:43:16 homeassistant 9c75c8473a4d[316]:  Home Assistant Supervisor: 244
Sep 17 01:43:16 homeassistant 9c75c8473a4d[316]: -----------------------------------------------------------
Sep 17 01:43:16 homeassistant 9c75c8473a4d[316]:  Please, share the above information when looking for help
Sep 17 01:43:16 homeassistant 9c75c8473a4d[316]:  or support in, e.g., GitHub, forums or the Discord chat.
Sep 17 01:43:16 homeassistant 9c75c8473a4d[316]: -----------------------------------------------------------
Sep 17 01:43:16 homeassistant 9c75c8473a4d[316]: [cont-init.d] 00-banner.sh: exited 0.
Sep 17 01:43:16 homeassistant 9c75c8473a4d[316]: [cont-init.d] 01-log-level.sh: executing... 
Sep 17 01:43:16 homeassistant kernel: hassio: port 15(veth2314832) entered disabled state
Sep 17 01:43:16 homeassistant NetworkManager[276]: <info>  [1600306996.2296] manager: (vetha254a15): new Veth device (/org/freedesktop/NetworkManager/Devices/38)
Sep 17 01:43:16 homeassistant 5ac5755b8b8b[316]: [cont-init.d] users.sh: exited 0.
Sep 17 01:43:16 homeassistant 5ac5755b8b8b[316]: [cont-init.d] vsftpd.sh: executing... 
Sep 17 01:43:16 homeassistant systemd-udevd[8125]: ethtool: autonegotiation is unset or enabled, the speed and duplex are not writable.
Sep 17 01:43:16 homeassistant kernel: hassio: port 15(veth2314832) entered disabled state
Sep 17 01:43:16 homeassistant avahi-daemon[315]: Interface veth2314832.IPv6 no longer relevant for mDNS.
Sep 17 01:43:16 homeassistant avahi-daemon[315]: Leaving mDNS multicast group on interface veth2314832.IPv6 with address fe80::ac73:3aff:fe18:dec3.
Sep 17 01:43:16 homeassistant kernel: device veth2314832 left promiscuous mode
Sep 17 01:43:16 homeassistant kernel: hassio: port 15(veth2314832) entered disabled state
Sep 17 01:43:16 homeassistant audit: ANOM_PROMISCUOUS dev=veth2314832 prom=0 old_prom=256 auid=4294967295 uid=0 gid=0 ses=4294967295
Sep 17 01:43:16 homeassistant avahi-daemon[315]: Withdrawing address record for fe80::ac73:3aff:fe18:dec3 on veth2314832.
Sep 17 01:43:16 homeassistant avahi-daemon[315]: Withdrawing workstation service for vetha254a15.
Sep 17 01:43:16 homeassistant avahi-daemon[315]: Withdrawing workstation service for veth2314832.
Sep 17 01:43:16 homeassistant NetworkManager[276]: <info>  [1600306996.2588] device (veth2314832): released from master device hassio
Sep 17 01:43:16 homeassistant systemd[1]: run-docker-netns-42aec3082649.mount: Succeeded.
Sep 17 01:43:16 homeassistant systemd[1]: var-lib-docker-overlay2-a2685988543e533fd907b08e4fe390160981ec9dec75c93d435aebb17c063b2d-merged.mount: Succeeded.
Sep 17 01:43:16 homeassistant systemd[1]: mnt-data-docker-overlay2-a2685988543e533fd907b08e4fe390160981ec9dec75c93d435aebb17c063b2d-merged.mount: Succeeded.
Sep 17 01:43:16 homeassistant 9c75c8473a4d[316]: [cont-init.d] 01-log-level.sh: exited 0.
Sep 17 01:43:16 homeassistant 9c75c8473a4d[316]: [cont-init.d] zerotier.sh: executing... 
-- Reboot --
Sep 17 01:46:48 homeassistant kernel: Linux version 5.4.63 (builder@9ca1091d2acb) (gcc version 9.3.0 (Buildroot 2020.02.4-g5fdcf8b2)) #1 SMP Tue Sep 8 13:28:08 UTC 2020
Sep 17 01:46:48 homeassistant kernel: Command line: apparmor=1 security=apparmor console=tty1 console=ttyS0 bootchooser.active=B  root=PARTUUID=a3ec664e-32ce-4665-95ea-7ae90ce9aa20 rootfstype=squashfs ro rootwait systemd.machine_id=0b629d1501334db9b748e7e658303d3c zram.enabled=1 zram.num_devices=3

last instance:
Sep 17 01:47:34 homeassistant avahi-daemon[316]: Withdrawing address record for fe80::cc8:43ff:fe90:63d2 on vethf40d1e8.
Sep 17 01:47:34 homeassistant avahi-daemon[316]: Withdrawing workstation service for vethc58d957.
Sep 17 01:47:34 homeassistant avahi-daemon[316]: Withdrawing workstation service for vethf40d1e8.
Sep 17 01:47:34 homeassistant NetworkManager[277]: <info>  [1600307254.1805] device (vethf40d1e8): released from master device hassio
Sep 17 01:47:34 homeassistant systemd[1]: run-docker-netns-c644fd340c51.mount: Succeeded.
Sep 17 01:47:34 homeassistant 9309a5e28bb1[317]: [services.d] done.
Sep 17 01:47:34 homeassistant systemd[1]: var-lib-docker-overlay2-a30007fabf4860acbe1ce46f3efb39b658c2d20911950dc652ab7848f743ff55-merged.mount: Succeeded.
Sep 17 01:47:34 homeassistant systemd[1]: mnt-data-docker-overlay2-a30007fabf4860acbe1ce46f3efb39b658c2d20911950dc652ab7848f743ff55-merged.mount: Succeeded.
Sep 17 01:47:34 homeassistant 303df09e8739[317]: [cont-init.d] nginx.sh: exited 0.
Sep 17 01:47:34 homeassistant 303df09e8739[317]: [cont-init.d] done.
Sep 17 01:47:34 homeassistant 303df09e8739[317]: [services.d] starting services
-- Reboot --
Sep 17 01:49:26 homeassistant kernel: Linux version 5.4.63 (builder@9ca1091d2acb) (gcc version 9.3.0 (Buildroot 2020.02.4-g5fdcf8b2)) #1 SMP Tue Sep 8 13:28:08 UTC 2020
Sep 17 01:49:26 homeassistant kernel: Command line: apparmor=1 security=apparmor console=tty1 console=ttyS0 bootchooser.active=B  root=PARTUUID=a3ec664e-32ce-4665-95ea-7ae90ce9aa20 rootfstype=squashfs ro rootwait systemd.machine_id=0b629d1501334db9b748e7e658303d3c zram.enabled=1 zram.num_devices=3
**Kernel logs:**

can't get it since VM is not responding even on pve console, not sure if we can configure crash dump on hassos; no old dmesg
**Description of problem:**

On hassos 4.12, I am able to start deconz using /tty/ACM0, however as soon as use /dev/serial/by-id address, the OS/console doesn't respond, even after waiting for a long time. I had to do stop/start on proxmox. I have the hassio running for 17 days and decided to test 4.13.
After upgrading, the system is not responding again after boot. I am able to login on the console but after some time it crashes. I was able to avoid the crashing after disabling deconz on startup. Using /tty/ACM0 or by-id address crashes the system afterwards.

OrbitingOcelot, 2020-09-17:
Same problem, hass.io VM completely freezes after running for a few minutes.
I am running 4.13 on QEMU 5.0.0.

accelle17, 2020-09-23:
so, I have updated to deconz hassio addon 6.3.0 and conbee 2 firmware 26580700/26650700 and it is the same even staying on 4.12. This is probably a deconz addon issue instead of hassos.

I was able to fix it by removing other USB devices and using port instead of usb vendor to mount it.
1. usb microphone - prevents booting, console can't even reach hassio cli.
2. huzbzb-1: used for z-wave, not using zigbee on it; I'm also using open z-wave (yaml not the mqtt dependent zwave)

accelle17, 2020-09-23:
Issue can be reproduced with:
1. hass image 0.115.2 by:
2. deconz addon 6.3.0
3. conbee 2 stick passthrough

steps:
1. Start hassos VM
2. start deconz
3. stop deconz
4. remove conbee 2 stick
5. plugin conbee 2 stick
6. start deconz
7. crash

accelle17, 2020-09-24:
This looks like a long pending proxmox 6 issue (usb removal crashes vm and frequent usb disconnects), downgrading it to 6.1 didnt help. Plus after some time, I can see the old conbee 2 stick disconnect and reconnect on proxmox syslog which might trigger the crashes. My new conbee 2 stick doesn't show this behavior.
Here are my references:
1. https://forum.proxmox.com/threads/usb-drive-crashes-vm.70214/
2. https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3001
3. https://forum.proxmox.com/threads/proxmox-6-0-usb-devices-disconnect-and-reconnect-randomly.59507/
4. https://forum.proxmox.com/threads/vm-freezes-when-disconnecting-usb-keyboard-and-mouse.70287/

I tried all firmwares from 0x264 to 0x266 on the old conbee 2 but still the same. Unfortunately, it's past midnight and I dont want to risk downgrading proxmox to 5.4. Ended up swapping the new conbee 2 stick and re-pairing all 100+ devices lol.

closed 2020-09-24 by accelle17