home-assistant/operating-system #871: Issue with OpenZwave addon after updateing HassOS to 4.13/245

Dinth, 2020-09-18:
**HassOS release with the issue:** 4.13

**Supervisor logs:**245

**Kernel logs:**

```
[   20.648160] hassio: port 8(vethed2fb6f) entered disabled state
[   20.962713] eth0: renamed from vetha084a7f
[   20.972576] IPv6: ADDRCONF(NETDEV_CHANGE): vethed2fb6f: link becomes ready
[   20.972700] hassio: port 8(vethed2fb6f) entered blocking state
[   20.972750] hassio: port 8(vethed2fb6f) entered forwarding state
[   21.502884] hassio: port 9(veth18e8c59) entered blocking state
[   21.502917] hassio: port 9(veth18e8c59) entered disabled state
[   21.503109] device veth18e8c59 entered promiscuous mode
[   21.503917] hassio: port 9(veth18e8c59) entered blocking state
[   21.503972] hassio: port 9(veth18e8c59) entered forwarding state
[   21.657142] hassio: port 9(veth18e8c59) entered disabled state
[   21.761833] eth0: renamed from vethdbd13f5
[   21.771347] IPv6: ADDRCONF(NETDEV_CHANGE): veth18e8c59: link becomes ready
[   21.772416] hassio: port 9(veth18e8c59) entered blocking state
[   21.773109] hassio: port 9(veth18e8c59) entered forwarding state
[   21.922521] hassio: port 10(vethb6ea2cb) entered blocking state
[   21.923933] hassio: port 10(vethb6ea2cb) entered disabled state
[   21.926548] device vethb6ea2cb entered promiscuous mode
[   21.927667] hassio: port 10(vethb6ea2cb) entered blocking state
[   21.928548] hassio: port 10(vethb6ea2cb) entered forwarding state
[   22.343536] eth0: renamed from vethd61d69d
[   22.349938] IPv6: ADDRCONF(NETDEV_CHANGE): vethb6ea2cb: link becomes ready
[   22.752101] hassio: port 11(veth98dcdf8) entered blocking state
[   22.753330] hassio: port 11(veth98dcdf8) entered disabled state
[   22.754690] device veth98dcdf8 entered promiscuous mode
[   23.336568] eth0: renamed from vethb744a54
[   23.349107] IPv6: ADDRCONF(NETDEV_CHANGE): veth98dcdf8: link becomes ready
[   23.350535] hassio: port 11(veth98dcdf8) entered blocking state
[   23.351710] hassio: port 11(veth98dcdf8) entered forwarding state
[   28.937421] udevd[190]: starting version 3.2.9
[   28.951208] udevd[191]: starting eudev-3.2.9
[   30.868685] show_signal_msg: 56 callbacks suppressed
[   30.868689] ozwadmin[8168]: segfault at 38 ip 00007f47afe54f35 sp 00007fff7345fcc0 error 4 in libqvnc.so[7f47afe50000+15000]
[   30.870654] Code: 8b 70 04 48 89 ef 31 c9 5d e9 61 fe ff ff 5d c3 90 53 48 89 fb e8 ab ff ff ff 48 8b 3b 5b e9 46 df ff ff 41 54 45 31 e4 55 53 <48> 8b 0f 83 79 04 00 0f 8e 88 00 00 00 4c 8b 41 10 48 89 fb 48 89
[   30.874661] audit: type=1701 audit(1600458695.150:201): auid=4294967295 uid=0 gid=0 ses=4294967295 subj==unconfined pid=8168 comm="ozwadmin" exe="/usr/local/bin/ozwadmin" sig=11 res=1

```

**Description of problem:**

Im running HassOS inside of ESXi on Dell R720 server. Today i have updated Supervisor and HassOS and since then my OZW stopped working.
Im getting the following errors in OZW log. According to this bug report it may be kernel related and/or is a common bug on RPi: https://github.com/OpenZWave/Zwave2Mqtt/issues/141. This being said, i've been using different Aeotec sticks in various Dell servers (both barematal and ESXi) for past three years and this is the first time im seeing this error - therefore it must be related to one of the updates last night. 
 
```
[20200918 21:28:04.079 BST] [ozw.library] [critical]: Error - Node: 0 ERROR: Not enough space in stream buffer 
[20200918 21:28:04.079 BST] [ozw.logging] [debug]: popping Log Mesages 
[20200918 21:28:04.080 BST] [ozw.library] [critical]: Error - Node: 0 ERROR: Not enough space in stream buffer 
[20200918 21:28:04.080 BST] [ozw.logging] [debug]: popping Log Mesages 
[20200918 21:28:04.796 BST] [ozw.values] [debug]: delValue: Removing Value  QVariant(QString, "Air Temperature") QVariant(qulonglong, 281475283697682) 0 
[20200918 21:28:04.797 BST] [ozw.notifications] [debug]: Notification pvt_valueRemoved:  844425237118994 Thread:  0x7f2885f77d48 
[20200918 21:28:05.078 BST] [ozw.library] [critical]: Error - Node: 0 ERROR: Not enough space in stream buffer 
[20200918 21:28:05.078 BST] [ozw.logging] [debug]: popping Log Mesages 
[20200918 21:28:05.080 BST] [ozw.library] [critical]: Error - Node: 0 ERROR: Not enough space in stream buffer 
[20200918 21:28:05.080 BST] [ozw.logging] [debug]: popping Log Mesages 
[20200918 21:28:05.080 BST] [ozw.library] [critical]: Error - Node: 0 ERROR: Not enough space in stream buffer 
[20200918 21:28:05.080 BST] [ozw.logging] [debug]: popping Log Mesages 
[20200918 21:28:07.070 BST] [ozw.values] [debug]: delValue: Removing Value  QVariant(QString, "Illuminance") QVariant(qulonglong, 844425237118994) 0 
[20200918 21:28:07.071 BST] [ozw.notifications] [debug]: Notification pvt_valueRemoved:  1407375190540306 Thread:  0x7f2885f77d48 
[20200918 21:28:09.067 BST] [ozw.values] [debug]: delValue: Removing Value  QVariant(QString, "Humidity") QVariant(qulonglong, 1407375190540306) 0 
[20200918 21:28:09.068 BST] [ozw.notifications] [debug]: Notification pvt_valueRemoved:  2533275097382930 Thread:  0x7f2885f77d48 
[20200918 21:28:10.719 BST] [ozw.values] [debug]: delValue: Removing Value  QVariant(QString, "Barometic Pressure") QVariant(qulonglong, 2533275097382930) 0 
[20200918 21:28:10.720 BST] [ozw.notifications] [debug]: Notification pvt_valueRemoved:  36310272302661650 Thread:  0x7f2885f77d48 
[20200918 21:28:12.483 BST] [ozw.values] [debug]: delValue: Removing Value  QVariant(QString, "Unknown") QVariant(qulonglong, 36310272302661650) 0 
[... 57 lines cut ...]
[20200918 21:28:50.059 BST] [ozw.logging] [debug]: popping Log Mesages 
[20200918 21:28:50.061 BST] [ozw.library] [critical]: Error - Node: 0 ERROR: Not enough space in stream buffer 
[20200918 21:28:50.061 BST] [ozw.logging] [debug]: popping Log Mesages 
[20200918 21:28:50.063 BST] [ozw.library] [critical]: Error - Node: 0 ERROR: Not enough space in stream buffer 
[20200918 21:28:50.063 BST] [ozw.logging] [debug]: popping Log Mesages 
[20200918 21:28:50.064 BST] [ozw.library] [critical]: Error - Node: 0 ERROR: Not enough space in stream buffer 
[20200918 21:28:50.065 BST] [ozw.logging] [debug]: popping Log Mesages 
[20200918 21:28:50.066 BST] [ozw.library] [critical]: Error - Node: 0 ERROR: Not enough space in stream buffer 
[20200918 21:28:50.067 BST] [ozw.logging] [debug]: popping Log Mesages 
[20200918 21:28:50.078 BST] [ozw.values] [debug]: delValue: Removing Value  QVariant(QString, "Acked Frames") QVariant(qulonglong, 2533275106852886) 0 
[20200918 21:28:50.079 BST] [ozw.notifications] [debug]: Notification pvt_valueRemoved:  308297748 Thread:  0x7f2885f77d48 
[20200918 21:28:51.058 BST] [ozw.library] [critical]: Error - Node: 0 ERROR: Not enough space in stream buffer 
[20200918 21:28:51.058 BST] [ozw.logging] [debug]: popping Log Mesages 
[20200918 21:28:51.059 BST] [ozw.library] [critical]: Error - Node: 0 ERROR: Not enough space in stream buffer 
[20200918 21:28:51.059 BST] [ozw.logging] [debug]: popping Log Mesages 
[20200918 21:28:52.313 BST] [ozw.values] [debug]: delValue: Removing Value  QVariant(QString, "Day") QVariant(qulonglong, 308297748) 0 
[20200918 21:28:52.314 BST] [ozw.notifications] [debug]: Notification pvt_valueRemoved:  281475285008401 Thread:  0x7f2885f77d48 
[20200918 21:28:53.852 BST] [ozw.values] [debug]: delValue: Removing Value  QVariant(QString, "Hour") QVariant(qulonglong, 281475285008401) 0 
[20200918 21:28:53.852 BST] [ozw.notifications] [debug]: Notification pvt_valueRemoved:  562950261719057 Thread:  0x7f2885f77d48 
[20200918 21:28:54.620 BST] [ozw.mqtt.publisher] [debug]: Removing CommandClass Topic for  18 1 129 
[20200918 21:28:55.394 BST] [ozw.values] [debug]: delValue: Removing Value  QVariant(QString, "Minute") QVariant(qulonglong, 562950261719057) 0 
[20200918 21:28:55.395 BST] [ozw.notifications] [debug]: Notification pvt_valueRemoved:  316768279 Thread:  0x7f2885f77d48 
[20200918 21:28:56.931 BST] [ozw.values] [debug]: delValue: Removing Value  QVariant(QString, "Library Version") QVariant(qulonglong, 316768279) 0 
[20200918 21:28:56.932 BST] [ozw.notifications] [debug]: Notification pvt_valueRemoved:  281475293478935 Thread:  0x7f2885f77d48 
```

Dinth, 2020-09-19:
I have tried powercycling the server and unplugging the stick for 15 minutes, but doesnt seem to help. Seems there's no communication with the stick at all:
```
[20200919 19:29:34.737 BST] [ozw.mqtt.publisher] [warning]: MQTT Client Error "Transport Invalid" 
[20200919 19:29:34.738 BST] [ozw.library] [info]: Info - Node: 0 mgr,     Driver for controller /dev/serial/by-id/usb-0658_0200-if00 pending removal 
[20200919 19:29:34.738 BST] [ozw.library] [debug]: Detail - Node: 0 Notification: DriverRemoved 
[20200919 19:29:34.738 BST] [ozw.library] [debug]: Detail - Node: 48 CAN received...triggering resend 
[20200919 19:29:34.738 BST] [ozw.notifications] [debug]: Notification pvt_driverRemoved  3918204832 Thread:  0x7f6636270d48 
[20200919 19:29:34.738 BST] [ozw.notifications] [debug]: Notification pvt_nodeRemoved  1 Thread:  0x7f6636270d48 
[20200919 19:29:34.738 BST] [ozw.notifications] [debug]: Notification pvt_valueRemoved:  17301521 Thread:  0x7f6636270d48 
[20200919 19:29:34.739 BST] [ozw.library] [debug]: Detail - Node: 48 CAN received...triggering resend 
[20200919 19:29:41.937 BST] [ozw.values] [debug]: delValue: Removing Value  QVariant(QString, "Basic") QVariant(qulonglong, 17301521) 0 
[20200919 19:29:41.938 BST] [ozw.notifications] [debug]: Notification pvt_valueRemoved:  281474994012177 Thread:  0x7f6636270d48 
[20200919 19:29:44.721 BST] [ozw.library] [info]: Info - Node: 48 Sending (NoOp) message (Attempt 2, Callback ID=0x49, Expected Reply=0x13) - NoOperation_Set (Node=48): 0x01, 0x09, 0x00, 0x13, 0x30, 0x02, 0x00, 0x00, 0x25, 0x49, 0xbb 
[20200919 19:29:44.722 BST] [ozw.library] [info]: Info - Node: 48 Encrypted Flag is 0 
[20200919 19:29:45.722 BST] [ozw.library] [info]: Info - Node: 48 Sending (NoOp) message (Attempt 3, Callback ID=0x4a, Expected Reply=0x13) - NoOperation_Set (Node=48): 0x01, 0x09, 0x00, 0x13, 0x30, 0x02, 0x00, 0x00, 0x25, 0x4a, 0xb8 
[20200919 19:29:45.723 BST] [ozw.library] [info]: Info - Node: 48 Encrypted Flag is 0 
[20200919 19:29:46.724 BST] [ozw.library] [info]: Info - Node: 48 Sending (NoOp) message (Attempt 4, Callback ID=0x4b, Expected Reply=0x13) - NoOperation_Set (Node=48): 0x01, 0x09, 0x00, 0x13, 0x30, 0x02, 0x00, 0x00, 0x25, 0x4b, 0xb9 
[20200919 19:29:46.725 BST] [ozw.library] [info]: Info - Node: 48 Encrypted Flag is 0 
[20200919 19:29:47.725 BST] [ozw.library] [info]: Info - Node: 48 Sending (NoOp) message (Attempt 5, Callback ID=0x4c, Expected Reply=0x13) - NoOperation_Set (Node=48): 0x01, 0x09, 0x00, 0x13, 0x30, 0x02, 0x00, 0x00, 0x25, 0x4c, 0xbe 
[20200919 19:29:47.726 BST] [ozw.library] [info]: Info - Node: 48 Encrypted Flag is 0 
[20200919 19:29:48.048 BST] [ozw.values] [debug]: delValue: Removing Value  QVariant(QString, "Basic Target") QVariant(qulonglong, 281474994012177) 0 
[20200919 19:29:48.049 BST] [ozw.notifications] [debug]: Notification pvt_valueRemoved:  562949970722835 Thread:  0x7f6636270d48 
[20200919 19:29:48.726 BST] [ozw.library] [info]: Info - Node: 48 Sending (NoOp) message (Attempt 6, Callback ID=0x4d, Expected Reply=0x13) - NoOperation_Set (Node=48): 0x01, 0x09, 0x00, 0x13, 0x30, 0x02, 0x00, 0x00, 0x25, 0x4d, 0xbf 
[20200919 19:29:48.726 BST] [ozw.library] [info]: Info - Node: 48 Encrypted Flag is 0 
[20200919 19:29:49.727 BST] [ozw.library] [critical]: Error - Node: 48 ERROR: Dropping command, expected response not received after 6 attempt(s). Command: "NoOperation_Set (Node=48): 0x01, 0x09, 0x00, 0x13, 0x30, 0x02, 0x00, 0x00, 0x25, 0x4d, 0xbf" 
[20200919 19:29:49.727 BST] [ozw.library] [debug]: Detail - Node: 48 Removing current message 
[20200919 19:29:49.727 BST] [ozw.library] [info]: Info - Node: 49 Sending (NoOp) message (Callback ID=0x37, Expected Reply=0x13) - NoOperation_Set (Node=49): 0x01, 0x09, 0x00, 0x13, 0x31, 0x02, 0x00, 0x00, 0x25, 0x37, 0xc4 
[20200919 19:29:49.727 BST] [ozw.library] [info]: Info - Node: 49 Encrypted Flag is 0 
[20200919 19:29:50.727 BST] [ozw.library] [critical]: Error - Node: 49 ERROR: Dropping command, expected response not received after 1 attempt(s). Command: "NoOperation_Set (Node=49): 0x01, 0x09, 0x00, 0x13, 0x31, 0x02, 0x00, 0x00, 0x25, 0x37, 0xc4" 
[20200919 19:29:50.728 BST] [ozw.library] [debug]: Detail - Node: 49 Removing current message 
[20200919 19:29:50.728 BST] [ozw.library] [debug]: Detail - Node: 49 Notification: Notification - TimeOut 
[20200919 19:29:51.302 BST] [ozw.mqtt.publisher] [debug]: Removing CommandClass Topic for  1 1 32 
[20200919 19:29:54.245 BST] [ozw.values] [debug]: delValue: Removing Value  QVariant(QString, "Basic Duration") QVariant(qulonglong, 562949970722835) 0 
[20200919 19:29:54.246 BST] [ozw.notifications] [debug]: Notification pvt_valueRemoved:  22799473140563988 Thread:  0x7f6636270d48 
[20200919 19:29:57.394 BST] [ozw.values] [debug]: delValue: Removing Value  QVariant(QString, "LED indicator configuration") QVariant(qulonglong, 22799473140563988) 0 
[20200919 19:29:57.395 BST] [ozw.notifications] [debug]: Notification pvt_valueRemoved:  61924494903345172 Thread:  0x7f6636270d48 
[20200919 19:30:00.937 BST] [ozw.values] [debug]: delValue: Removing Value  QVariant(QString, "Configuration of the RF power level") QVariant(qulonglong, 61924494903345172) 0 
[20200919 19:30:00.938 BST] [ozw.notifications] [debug]: Notification pvt_valueRemoved:  68116944390979604 Thread:  0x7f6636270d48 
[20200919 19:30:04.164 BST] [ozw.values] [debug]: delValue: Removing Value  QVariant(QString, "Security network enabled") QVariant(qulonglong, 68116944390979604) 0 
[... cut ...]
[20200919 19:32:13.285 BST] [ozw.values] [debug]: delValue: Removing Value  QVariant(QString, "Low Battery") QVariant(qulonglong, 10977524135493649) 0 
[20200919 19:32:13.286 BST] [ozw.notifications] [debug]: Notification pvt_valueRemoved:  11258999112204308 Thread:  0x7f6636270d48 
[20200919 19:32:18.952 BST] [ozw.values] [debug]: delValue: Removing Value  QVariant(QString, "Report Only On Thresholds") QVariant(qulonglong, 11258999112204308) 0 
[20200919 19:32:18.954 BST] [ozw.notifications] [debug]: Notification pvt_valueRemoved:  11540474088914963 Thread:  0x7f6636270d48 
```

agners, 2020-11-03:
Is this problem still happening?

Reading the comments in https://github.com/OpenZWave/open-zwave/issues/673, it sound like this can be a OS problem in case the system is overloaded. But since you are using a server system I hardly think that is the case. Maybe too many messages in the Z-Wave network? Did you add new devices around that time?

Otherwise, you can downgrade the OS using `ha os update --version 4.xx`. If it used to work, we should figure out which update exactly made the problem appear.

Dinth, 2020-11-03:
Hey. Sorry for lack of updates, i completely forgotten about this ticket. 
Since I have logged it I have migrated my Hass to Ubuntu server and the problem still persisted. I have done some further troubleshooting and pinpointed the issue to EEPROM memory on the stick getting corrupted. 
When the new devices are added to the mesh, at some point the data about last 20-30 devices (not only ones recently added) gets corrupted. Whenever the software wants to query one of those corrupted devices it fails and after few such failures, the stick stops working at all, instead just throws "Not enough space in stream buffer". Similar behaviour can be seen on Windows using Aeotec bundled software - it also fails to query the same nodes which are corrupted in OZW, but i believe there may be an issue with the way OZW is adding the nodes. It's not hardware related problem though, all together I have tried 5 different ZW5 sticks, including the latest RPi4 compatible revision and can see the same behaviour on each of them. So far, my workaround is to backup my stick as often as possible, and restore the EEPROM whenever "Not enough space in stream buffer" error starts showing up.

closed 2021-05-12 by stale[bot]