home-assistant/operating-system #872: Passthrough of PCIe devices on Hyper-V VHDX

tduffy83, 2020-09-19:
**HassOS release with the issue:**

4.13
**Supervisor logs:**

0-09-19 01:57:26 INFO (SyncWorker_0) [supervisor.docker.interface] Stop homeassistant application
20-09-19 01:58:15 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SERVICES' stop 2 add-ons
20-09-19 01:58:15 INFO (SyncWorker_2) [supervisor.docker.interface] Stop addon_core_duckdns application
20-09-19 01:58:16 INFO (SyncWorker_2) [supervisor.docker.interface] Clean addon_core_duckdns application
20-09-19 01:58:16 INFO (SyncWorker_1) [supervisor.docker.interface] Stop addon_core_samba application
20-09-19 01:58:20 INFO (SyncWorker_1) [supervisor.docker.interface] Clean addon_core_samba application
20-09-19 01:58:20 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SYSTEM' stop 1 add-ons
20-09-19 01:58:20 INFO (SyncWorker_0) [supervisor.docker.interface] Stop addon_core_mosquitto application
20-09-19 01:58:21 INFO (SyncWorker_0) [supervisor.docker.interface] Clean addon_core_mosquitto application
20-09-19 01:58:21 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.INITIALIZE' stop 0 add-ons
20-09-19 01:58:21 INFO (MainThread) [supervisor.plugins.audio] Stop Audio plugin
20-09-19 01:58:21 INFO (SyncWorker_2) [supervisor.docker.interface] Stop hassio_audio application
20-09-19 01:58:26 INFO (SyncWorker_2) [supervisor.docker.interface] Clean hassio_audio application
20-09-19 01:58:26 INFO (MainThread) [supervisor.plugins.cli] Stop cli plugin
20-09-19 01:58:26 INFO (SyncWorker_1) [supervisor.docker.interface] Stop hassio_cli application
20-09-19 01:58:30 INFO (SyncWorker_1) [supervisor.docker.interface] Clean hassio_cli application
20-09-19 01:58:30 INFO (MainThread) [supervisor.plugins.multicast] Stop Multicast plugin
20-09-19 01:58:30 INFO (SyncWorker_0) [supervisor.docker.interface] Stop hassio_multicast application
20-09-19 01:58:33 INFO (SyncWorker_0) [supervisor.docker.interface] Clean hassio_multicast application
20-09-19 01:58:33 INFO (MainThread) [supervisor.plugins.dns] Stop CoreDNS plugin
20-09-19 01:58:33 INFO (SyncWorker_2) [supervisor.docker.interface] Stop hassio_dns application
20-09-19 01:58:38 INFO (SyncWorker_2) [supervisor.docker.interface] Clean hassio_dns application
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing... 
[04:27:06] INFO: Update udev information
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[04:27:06] INFO: Start local supervisor watchdog...
20-09-19 04:27:08 INFO (MainThread) [__main__] Initialize Supervisor setup
20-09-19 04:27:08 INFO (MainThread) [supervisor.bootstrap] Initialize Supervisor Sentry
20-09-19 04:27:08 INFO (MainThread) [supervisor.bootstrap] Setup coresys for machine: qemux86-64
20-09-19 04:27:08 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attach to Supervisor homeassistant/amd64-hassio-supervisor with version 245
20-09-19 04:27:08 INFO (MainThread) [__main__] Setup Supervisor
20-09-19 04:27:08 INFO (MainThread) [supervisor.host.info] Update local host information
20-09-19 04:27:08 INFO (MainThread) [supervisor.host.services] Update service information
20-09-19 04:27:08 INFO (MainThread) [supervisor.host.network] Update local network information
20-09-19 04:27:08 INFO (MainThread) [supervisor.host.sound] Update PulseAudio information
20-09-19 04:27:08 INFO (MainThread) [supervisor.host.apparmor] Load AppArmor Profiles: {'hassio-supervisor'}
20-09-19 04:27:08 INFO (MainThread) [supervisor.host.services] Reload local service hassos-apparmor.service
20-09-19 04:27:08 INFO (SyncWorker_0) [supervisor.docker.interface] Attach to homeassistant/amd64-hassio-dns with version 9
20-09-19 04:27:08 INFO (MainThread) [supervisor.plugins.dns] Start CoreDNS plugin
20-09-19 04:27:10 INFO (SyncWorker_0) [supervisor.docker.dns] Start DNS homeassistant/amd64-hassio-dns with version 9 - 172.30.32.3
20-09-19 04:27:10 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
20-09-19 04:27:10 INFO (SyncWorker_0) [supervisor.docker.interface] Attach to homeassistant/amd64-hassio-audio with version 17
20-09-19 04:27:10 INFO (MainThread) [supervisor.plugins.audio] Start Audio plugin
20-09-19 04:27:11 INFO (SyncWorker_0) [supervisor.docker.audio] Start Audio homeassistant/amd64-hassio-audio with version 17 - 172.30.32.4
20-09-19 04:27:11 INFO (SyncWorker_0) [supervisor.docker.interface] Attach to homeassistant/amd64-hassio-cli with version 26
20-09-19 04:27:11 INFO (MainThread) [supervisor.plugins.cli] Start cli plugin
20-09-19 04:27:11 INFO (SyncWorker_0) [supervisor.docker.cli] Start CLI homeassistant/amd64-hassio-cli with version 26 - 172.30.32.5
20-09-19 04:27:11 INFO (SyncWorker_0) [supervisor.docker.interface] Attach to homeassistant/amd64-hassio-observer with version 3
20-09-19 04:27:11 INFO (SyncWorker_0) [supervisor.docker.interface] Attach to homeassistant/amd64-hassio-multicast with version 3
20-09-19 04:27:11 INFO (MainThread) [supervisor.plugins.multicast] Start Multicast plugin
20-09-19 04:27:11 INFO (SyncWorker_0) [supervisor.docker.multicast] Start Multicast homeassistant/amd64-hassio-multicast with version 3 - Host
20-09-19 04:27:11 INFO (MainThread) [supervisor.updater] Fetch update data from https://version.home-assistant.io/stable.json
20-09-19 04:27:11 INFO (MainThread) [supervisor.homeassistant.secrets] Load Home Assistant secrets: 1
20-09-19 04:27:11 INFO (SyncWorker_0) [supervisor.docker.interface] Attach to homeassistant/qemux86-64-homeassistant with version 0.115.1
20-09-19 04:27:11 INFO (MainThread) [supervisor.hassos] Detect HassOS 4.13 / BootSlot A
20-09-19 04:27:12 INFO (MainThread) [supervisor.store.git] Load add-on /data/addons/core repository
20-09-19 04:27:12 INFO (MainThread) [supervisor.store.git] Load add-on /data/addons/git/a0d7b954 repository
[... 16 lines cut ...]
20-09-19 04:27:13 INFO (MainThread) [supervisor.services.modules.mqtt] Set core_mosquitto as service provider for mqtt
20-09-19 04:27:17 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SERVICES' start 2 add-ons
20-09-19 04:27:18 INFO (SyncWorker_2) [supervisor.docker.addon] Start Docker add-on homeassistant/amd64-addon-duckdns with version 1.12.1
20-09-19 04:27:18 INFO (SyncWorker_3) [supervisor.docker.addon] Start Docker add-on homeassistant/amd64-addon-samba with version 9.3.0
20-09-19 04:27:23 INFO (SyncWorker_2) [supervisor.docker.interface] Start homeassistant
20-09-19 04:27:23 INFO (MainThread) [supervisor.homeassistant.core] Wait until Home Assistant is ready
20-09-19 04:27:33 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
20-09-19 04:27:33 INFO (MainThread) [supervisor.homeassistant.core] Detect a running Home Assistant instance
20-09-19 04:27:33 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.APPLICATION' start 2 add-ons
20-09-19 04:27:34 INFO (SyncWorker_2) [supervisor.docker.addon] Start Docker add-on homeassistant/amd64-addon-configurator with version 5.1.0
20-09-19 04:27:34 INFO (SyncWorker_3) [supervisor.docker.addon] Start Docker add-on esphome/esphome-hassio-amd64 with version 1.15.1
20-09-19 04:27:35 INFO (MainThread) [supervisor.api.security] /host/info access from a0d7b954_esphome
20-09-19 04:27:39 INFO (MainThread) [supervisor.misc.tasks] All core tasks are scheduled
20-09-19 04:27:39 INFO (MainThread) [supervisor.misc.hwmon] Started Supervisor hardware monitor
20-09-19 04:27:39 INFO (MainThread) [supervisor.core] Supervisor is up and running
20-09-19 04:27:39 INFO (MainThread) [supervisor.host.info] Update local host information
20-09-19 04:27:39 INFO (MainThread) [supervisor.host.services] Update service information
20-09-19 04:27:39 INFO (MainThread) [supervisor.host.network] Update local network information
20-09-19 04:27:39 INFO (MainThread) [supervisor.host.sound] Update PulseAudio information
**Journal logs:**

**Kernel logs:**

**Description of problem:**
I use a USB Z-Waze stick to control Z-Wave devices in my home.  This stick is plugged into a PCIe USB 3.0 Controller card.  This card is passed through to a Hyper-V VM using the HassOS 4.13 VHDX downloaded from the Home Assistant website.  HassOS does not recognize the PCI card, and thus my USB Z-Wave stick is not available to use in Home Assistant.  I can confirm that the PCI passthrough works using Ubuntu, Debian, and Windows 10, so the issue is with HassOS.  Perhaps the devs did not include PCI or USB support in the VHDX version as they did not anticipate users needing this functionality.  Either way, I want to use a HassOS image for ease of support and updates but this is completely broken for my use case.

agners, 2020-09-20:
> Perhaps the devs did not include PCI or USB support in the VHDX version as they did not anticipate users needing this functionality.

Quite likely. Do you happen to know what driver(s) is/are required? Alternatively, the output of `lsmod` (and maybe also `lspci`) in Ubuntu or Debian when PCI passthrough works might be helpful.

tduffy83, 2020-09-21:
duffy@duffhome:~$ lsmod
Module                  Size  Used by
xt_nat                 16384  29
xt_tcpudp              16384  31
veth                   20480  0
xt_conntrack           16384  2
ipt_MASQUERADE         16384  3
nf_conntrack_netlink    49152  0
xfrm_user              40960  1
xfrm_algo              16384  1 xfrm_user
nft_counter            16384  68
xt_addrtype            16384  2
nft_compat             20480  67
nft_chain_nat_ipv4     16384  32
nf_nat_ipv4            16384  2 ipt_MASQUERADE,nft_chain_nat_ipv4
nf_nat                 36864  2 nf_nat_ipv4,xt_nat
nf_conntrack          172032  6 xt_conntrack,nf_nat,ipt_MASQUERADE,nf_nat_ipv4,x                                                                                                                                                             t_nat,nf_conntrack_netlink
nf_defrag_ipv6         20480  1 nf_conntrack
nf_defrag_ipv4         16384  1 nf_conntrack
libcrc32c              16384  2 nf_conntrack,nf_nat
nf_tables             143360  295 nft_compat,nft_chain_nat_ipv4,nft_counter
nfnetlink              16384  4 nft_compat,nf_conntrack_netlink,nf_tables
br_netfilter           24576  0
bridge                188416  1 br_netfilter
stp                    16384  1 bridge
llc                    16384  2 bridge,stp
aufs                  274432  0
rfkill                 28672  1
overlay               131072  10
cp210x                 36864  2
usbserial              53248  5 cp210x
sb_edac                24576  0
crct10dif_pclmul       16384  0
crc32_pclmul           16384  0
ghash_clmulni_intel    16384  0
xhci_pci               16384  0
xhci_hcd              266240  1 xhci_pci
usbcore               294912  4 usbserial,xhci_hcd,cp210x,xhci_pci
usb_common             16384  1 usbcore
serio_raw              16384  0
nls_ascii              16384  1
nls_cp437              20480  1
vfat                   20480  1
fat                    86016  1 vfat
intel_rapl_perf        16384  0
hv_balloon             24576  0
sg                     36864  0
evdev                  28672  1
pci_hyperv             28672  0
hyperv_keyboard        16384  0
hyperv_fb              20480  1
hv_utils               32768  0
efi_pstore             16384  0
joydev                 24576  0
pcspkr                 16384  0
efivars                20480  1 efi_pstore
efivarfs               16384  1
ip_tables              28672  0
x_tables               45056  7 xt_conntrack,nft_compat,xt_tcpudp,ipt_MASQUERADE                                                                                                                                                             ,xt_addrtype,xt_nat,ip_tables
autofs4                49152  2
ext4                  741376  1
crc16                  16384  1 ext4
mbcache                16384  1 ext4
jbd2                  122880  1 ext4
crc32c_generic         16384  0
fscrypto               32768  1 ext4
ecb                    16384  0
sr_mod                 28672  0
cdrom                  65536  1 sr_mod
sd_mod                 61440  4
crc32c_intel           24576  3
hv_storvsc             20480  3
scsi_transport_fc      69632  1 hv_storvsc
hid_generic            16384  0
scsi_mod              249856  5 sd_mod,hv_storvsc,scsi_transport_fc,sg,sr_mod
hid_hyperv             16384  0
hid                   135168  2 hid_hyperv,hid_generic
hv_netvsc              81920  0
aesni_intel           200704  0
aes_x86_64             20480  1 aesni_intel
crypto_simd            16384  1 aesni_intel
cryptd                 28672  3 crypto_simd,ghash_clmulni_intel,aesni_intel
glue_helper            16384  1 aesni_intel
hv_vmbus              106496  8 hv_balloon,hv_utils,hv_netvsc,hid_hyperv,hv_stor                                                                                                                                                             vsc,hyperv_keyboard,pci_hyperv,hyperv_fb

tduffy83, 2020-09-21:
lspci
8bc5:00:00.0 USB controller: Renesas Technology Corp. uPD720202 USB 3.0 Host Controller (rev 02)

agners, 2020-09-24:
@tduffy83 it seems `pci_hyperv` is the driver required to support pass through PCI(e) devices. Added the relevant kernel config in #872.

tduffy83, 2020-09-24:
Any idea when we might get a release?  I can test if this makes it into a dev version.

agners, 2020-10-12:
@tduffy83 this is in development release 5.3 now. Please reopen in case pass-through still does not work.

closed 2020-10-12 by agners