home-assistant/operating-system #879: Worse performance than in docker

adrianmihalko, 2020-09-26:
**HassOS release with the issue:**

Frontend version: 20200918.2 - latest

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

https://pastebin.com/raw/d9N6ijrk

**Description of problem:**

In the last days I decided to switch from my good old, but now unsupported HA installation (with Frenck's script) to new in a standalone virtual machine. Here come the problem. For some reason with my ESXI the Hass.io official .vmdk is much slower than my old Ubuntu&HA installation in Docker. Bot has 4GB of memory, 2 CPU. 

Comparison:

https://www.youtube.com/watch?v=wSKjI_wDQgY

First tab: Ubuntu&HA
Second tab: Hass.io .vmdk

top is showing 0% cpu use and there is a lot of free memory:

[image]

Theoretically it should be faster, because it has slimmer host OS, but in practice it is much slower and I simply don’t understand why.

balloob, 2020-09-26:
Was your ubuntu installation also inside a VM?

balloob, 2020-09-26:
Your video shows UI which has nothing to do with the backend. That's all rendered in the frontend.

adrianmihalko, 2020-09-26:
Hi Paul,

Yes, the other installation was running on the same ESXI inside a VM:

[image]

The Ubuntu VM furthermore runs bunch of other software as well.

Hass.io VM is using minimal CPU & load is also low:

[image]

[image]

I literally have no idea what's casuing the slowness on the new VM. One more example: when I open up Unused entities, on the Ubuntu installation is fast and takes about 70%CPU (Developer tools - Timelines), but on new Hassi.io VM it takes 110% often page become blank and I get the usual This page is using significant memory. Closing it may improve...

Do you think that this is a front-end problem or a problem with my VM installation?

balloob, 2020-09-26:
Unused entities does not query the backend. It's pure frontend.

adrianmihalko, 2020-09-27:
Thanks, for now I am switching back to supervisored Ubuntu installation. If someone had any idea, please leave here a comment in the future.

closed 2020-09-27 by adrianmihalko