home-assistant/operating-system #883: 1wire adapter DS9490R - not found in Hassio

pepsonEL, 2020-10-01:
HassOS 5.0 and 5.2

Home Assistan Core 0.115.6

After connect to my RPI4 and Odroid N2 my adapter 1Wire it is not found in Supervisior-System-Hardware. In logs of HOST show that i connect it and disconect. But not found to use it in Home Assistant. Why ? Is any solltion for it ?

Logs from HOST on Odroid N2 with operating system v5.2:
[  167.386717] usb 1-1.1.4: USB disconnect, device number 9
[  173.781765] usb 1-1.4: new full-speed USB device number 10 using xhci-hcd
[  175.282277] audit: type=1334 audit(1601569622.414:240): prog-id=12 op=UNLOAD
[  175.289773] audit: type=1334 audit(1601569622.414:241): prog-id=11 op=UNLOAD
[  194.523774] usb 1-1.4: USB disconnect, device number 10
[  205.523813] usb 1-1.4: new full-speed USB device number 11 using xhci-hcd

Logs from HOST on RPI4 4GB with operating system v5.0:
[ 1239.130017] usb 1-1.3: USB disconnect, device number 3
[ 1241.705173] usb 1-1.3: new full-speed USB device number 4 using xhci_hcd
[ 1241.812237] usb 1-1.3: New USB device found, idVendor=04fa, idProduct=2490, bcdDevice= 0.02
[ 1241.812278] usb 1-1.3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 1241.851618] w1_master_driver w1_bus_master1: Attaching one wire slave 81.000000344670 crc 01

pepsonEL, 2020-10-01:
After update OS on RPI4 to v5.2 after connect Adapter 1wire DS9490R to USB in logs show:
[  381.680965] usb 1-1.3: USB disconnect, device number 3
[  385.759500] usb 1-1.3: new full-speed USB device number 4 using xhci_hcd
[  385.866521] usb 1-1.3: New USB device found, idVendor=04fa, idProduct=2490, bcdDevice= 0.02
[  385.866553] usb 1-1.3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[  385.898083] Driver for 1-wire Dallas network protocol.
[  385.917059] usbcore: registered new interface driver DS9490R
[  385.931827] w1_master_driver w1_bus_master1: Attaching one wire slave 81.000000344670 crc 01

but still not found mounted it in section HARDWARE.

agners, 2020-10-08:
Did that work in a older release? I do not have a 1Wire device to test..

pepsonEL, 2020-11-03:
I dont know but on very old release i think that works....

closed 2021-05-12 by stale[bot]

askthomas, 2021-07-28:
It use to works in RPI4 but not PC like Intel Nuc, there are something missing in NUC HAS install, not been able to find what.

OS find USB module, but not mount it
[image]

agners, 2021-07-29:
The DS2490 USB <-> W1 driver is not enabled. However, this is different from what the original report was about: There it was about missing hardware entry in Supervisior-System-Hardware.

@askthomas can you open a new issue?

askthomas, 2021-07-29:
Yes I can/will create a new ticket. Thasnk/Thomas Från: Stefan AgnerSkickat: den 29 juli 2021 22:17Till: home-assistant/operating-systemKopia: AskTHomas; MentionÄmne: Re: [home-assistant/operating-system] 1wire adapter DS9490R - not found in Hassio (#883) The DS2490 USB <-> W1 driver is not enabled. However, this is different from what the original report was about: There it was about missing hardware entry in ***@***.*** can you open a new issue?

tomasenskede, 2022-10-06:
> DS9490R

I am trying to configure my DS9490R with OWserver... but I cant config and start the server... what "device" should I enter?

pepsonEL, 2023-05-04:
Hi
I come back.... Hassos 10 and again attache USB DS9490R and show this logs and still no workingL

```
May 04 20:45:44 homeassistant kernel: usb 1-4: new full-speed USB device number 13 using xhci_hcd
May 04 20:45:44 homeassistant kernel: usb 1-4: New USB device found, idVendor=04fa, idProduct=2490, bcdDevice= 0.02
May 04 20:45:44 homeassistant kernel: usb 1-4: New USB device strings: Mfr=0, Product=0, SerialNumber=0
May 04 20:45:44 homeassistant kernel: usbcore: registered new interface driver DS9490R
May 04 20:45:44 homeassistant kernel: w1_master_driver w1_bus_master1: Attaching one wire slave 81.000000344670 crc 01
May 04 20:45:44 homeassistant kernel: w1_master_driver w1_bus_master1: Family 81 for 81.000000344670.01 is not registered.
```