home-assistant/operating-system #888: hassio.host_reboot not rebooting host (Raspberry Pi 4B 8GB).

JMan7777, 2020-10-07:
Hi,

when trying to reboot the Raspberry PI running my Home Assistant via service:
`hassio.host_reboot`
or via:
`Supervisor -> System -> Host System "Reboot" button`
the Raspberry Pi will just shutdown but not reboot.

This is very inconvenient as you cannot remote restart the whole Home Assistant host,
since you still need to unplug and re-plug the USB power supply to perform the restart.

I'm using:
Raspberry Pi 4B 8GB (2020-09-03 bootloader EEPROM and VLI 0138A1)
Home Assistant 0.115.6
HassOS version: 5.2 (installed on Raspberry Pi attached USB 3.0 SSD with 240GB)
Frontend version: 20200918.2 - latest

Thx a lot.

Cheers,

Markus

agners, 2020-10-08:
Is HDMI showing anything when the system hangs?

Do you happen to know what USB SSD Controller you are using? I noticed with my JMS583 that it behaved rather unreliably when not properly powered (e.g. it worked sometimes, but not other times). Can you reproduce the issue when the SSD is connected via a powered USB 3.0 Hub?

JMan7777, 2020-10-09:
Hi,

thx for the reply.

I will check regarding the HDMI output once I'm back home in a week.

Regarding your other questions:
- I use the Raspberry PI with the original 3A power adapter which should have more than enough power.
- The Raspberry PI is connected via the LAN port (not wireless) and via USB I have the following devices connected:
   - ADATA SU650 240GB in a ORICO 2020U3 USB 3.0 case (with UASP support) connected to a USB 3 port @ the PI
   - Nortek HUSBZB-1 (Zigbee & Zwave) stick connected to a USB 2 port @ the PI

So far the system is very stable (no crashes, etc), so I guess that I don't have any power issues.

Will  come back once I have the HDMI output captured.

Thx again.

diamant-x, 2020-11-11:
Stupid question but, is the 8Gb RAM model supported now? Based on the readme, only up to 4Gb is. https://github.com/home-assistant/operating-system/blob/dev/Documentation/boards/README.md

JMan7777, 2020-11-12:
> 
> 
> Stupid question but, is the 8Gb RAM model supported now? Based on the readme, only up to 4Gb is. https://github.com/home-assistant/operating-system/blob/dev/Documentation/boards/README.md

I have no problem running hassio 5.x (64 bit to make use of the 8GB) on my  Raspberry Pi 4B 8GB. I need the 5.x version as I boot from SSD. 

[image]

I not yet found the time to re-test my issue under version 5.5 (need to move my Pi near a TV to connect via HDMI) but hope I can do it the next days.

JMan7777, 2020-11-14:
I finally could get the HDMI output from the reboot problem.

The Raspberry PI actually reboots but shows then the following boot screen error:

[image]

When I reset the power the Raspberry PI and Home Assistant will boot up normally.

I tested with:

Home Assistant 0.117.6
Supervisor: 2020.11.0
HassOS: 5.5
Frontend version: 20201021.4 - latest

pepe59, 2020-11-14:
I have no idea if this is related to this problem.
I updated HassOs to version 5.5 and after about 2 hours the system froze. I restarted by unplugging the power. HassOs 5.5 and HA 0.117.6 started without errors and froze in the next few hours.
I don't see mistakes anywhere. I returned HassOs to version 5.4, HA 0.117.6, Supervisor 2020.11.0 which runs stably.
Rpi4 4 Gb
Sata board X825
SSD Kingstone 120Gb
Adapter 5V 4A

agners, 2020-11-26:
@pepe59 hm, the 5.5 release does contain a new kernel, so I guess it could be that this is the culprit. So the system is not displaying any error, not even on the console (HDMI)?

pepe59, 2020-11-26:
I bought Rpi4 8Gb. I installed HassOS 5.5 64bit. I restored the backup and since then the system seems stable. I didn't notice any system freezes. The restart is running correctly and everything is working fine.
I have no idea about Rpi4 4Gb, but I'm just wondering if something could have gone wrong when updating the system. I did not use HDMI to check for errors.

JMan7777, 2020-12-01:
I updated to the new [Hass OS 5.6](https://github.com/home-assistant/operating-system/releases/tag/5.6l) and the problem is gone now.

I assume the:

- Add rpi-eeprom firmware [(#939)](https://github.com/home-assistant/operating-system/pull/939) @tmm1

fix also solved the reboot issue.

So I closing this bug now.

closed 2020-12-01 by JMan7777