home-assistant/operating-system #891: DNS server settings ignored for lan hosts.

Zixim, 2020-10-09:
**HassOS release with the issue:**

- Frontend -> Configuration -> Info
```
arch	x86_64
chassis	vm
dev	false
docker	true
docker_version	19.03.11
hassio	true
host_os	HassOS 4.13
installation_type	Home Assistant OS
os_name	Linux
os_version	5.4.63
python_version	3.8.5
supervisor	247
timezone	Europe/Brussels
version	0.115.6
virtualenv	false
```

- Or use this command: hass --version
```
➜  ~ hass --version
zsh: command not found: hass
```

**Journal logs:**
```
Oct 09 08:11:07 homeassistant 85843c26f66f[386]: [INFO] 127.0.0.1:35481 - 43627 "A IN influx.local. udp 30 false 512" NXDOMAIN qr,rd,ra 105 0.090119317s
Oct 09 08:11:07 homeassistant 85843c26f66f[386]: [INFO] 172.30.32.1:48509 - 43627 "A IN influx.local. udp 30 false 512" NXDOMAIN qr,rd,ra 105 0.091012247s
Oct 09 08:11:08 homeassistant 85843c26f66f[386]: [INFO] 127.0.0.1:36749 - 57069 "A IN syncthingx.local. udp 34 false 512" NXDOMAIN qr,rd,ra 109 0.024540873s
Oct 09 08:11:08 homeassistant 85843c26f66f[386]: [INFO] 172.30.32.1:38087 - 57069 "A IN syncthingx.local. udp 34 false 512" NXDOMAIN qr,rd,ra 109 0.025081218s
Oct 09 08:11:08 homeassistant 85843c26f66f[386]: [INFO] 127.0.0.1:35481 - 57327 "AAAA IN syncthingx.local. udp 34 false 512" NXDOMAIN qr,rd,ra 109 0.096807474s
Oct 09 08:11:08 homeassistant 85843c26f66f[386]: [INFO] 172.30.32.1:38087 - 57327 "AAAA IN syncthingx.local. udp 34 false 512" NXDOMAIN qr,rd,ra 109 0.097437652s
```

**Description of problem:**
**_TLDR : After some time (X hours) , HA stops using the user-defined DNS server, thus no longer being able to resolve hosts on the LAN._**

HA is configured to use a local DNS server : 
```
➜  ~ ha dns info
host: 172.30.32.3
locals:
- dns://192.168.11.2
servers:
- dns://192.168.11.2
version: "9"
version_latest: "9"
➜  ~
```

Trying to resolve a local host fails : 
```
➜  ~ nslookup influx.local
Server:         172.30.32.3
Address:        172.30.32.3#53

** server can't find influx.local: NXDOMAIN
```
On the DNS server side, the logs show no request arriving for that lookup.

_Forcing_ the lookup to use the specific DNS server works : 
```
➜  ~ nslookup influx.local 192.168.11.2
Server:         192.168.11.2
Address:        192.168.11.2#53

Name:   influx.local
Address: 192.168.11.134
```
And in this case, the DNS logging indeed confirms name resolution : 
```
Oct  9 08:21:20 dnsmasq[344]: query[A] influx.local from 192.168.11.5
Oct  9 08:21:20 dnsmasq[344]: query[AAAA] influx.local from 192.168.11.5
```

### Doing the same on the HassOS host works without ant kind of issue :
```
# nslookup influx.local
Server:         192.168.11.2
Address:        192.168.11.2:53

Name:   influx.local
Address: 192.168.11.134

Non-authoritative answer:

#
```
### Additional info :
- Doing a `ha dns restart` solves the issue, for a while (X hours), but it always returns to being broken.
- I migrated to HassOS a month or 2 ago, previously been running Hassio for a few years.
- In a Hassio setup, this issue _never_ happened.
- HassOS based setup has had this from my first install.
- Pleading for help on Discord yielded very little response, no useful response at all.

agners, 2020-10-11:
As discussed on Discord, I looked a bit deeper into the Core DNS setup to try to understand what could go wrong.

It really seems as if Core DNS suddenly stops query the local DNS server (in which case this bug probably belongs to a different place, but that is not certain at this point).

One theory I had was that the [forward](https://coredns.io/plugins/forward/#syntax) policy to the local server somehow got stuck on the fallback server. However, it seems that the `forward` plugin with `policy` `sequential` should "reset" itself after every query. I also tested this locally: Disabling the DNS on my router starts to make use of the external DNS, but then re-enabling it immediately moves starts to query the internal DNS again. Could still be a special situation in your case, but in general it seems to work as intended.

Also the `mdns` plugin used should actually still forward queries to your local DNS server even when using `.local` domains.

So I am a bit out of idea what could go wrong in your setup... :-(

Zixim, 2020-10-11:
Thing is, I'm not alone with this. Even it it probably is a margin case.

After our talk on Discord, I have added my router as extra dns server.
```
➜  /config ha dns info
host: 172.30.32.3
locals:
- dns://192.168.11.2
servers:
- dns://192.168.11.2
- dns://192.168.11.1
version: "9"
version_latest: "9"
```
The router has ONLY 192.168.11.2 as its dns server.
It's ufly AF, but if it works...
I'll know more in 12 hours or so.

Zixim, 2020-10-12:
it didn't work ...
the above config, after 14 hours, now yields this : 
```
➜  /config nslookup influx.local
Server:         172.30.32.3
Address:        172.30.32.3#53

** server can't find influx.local: NXDOMAIN
```
and : 
```
➜  /config nslookup influx.local 192.168.11.2
Server:         192.168.11.2
Address:        192.168.11.2#53

Name:   influx.local
Address: 192.168.11.134
```
and : 
```
➜  /config nslookup influx.local 192.168.11.1
Server:         192.168.11.1
Address:        192.168.11.1#53

Name:   influx.local
Address: 192.168.11.134
```