home-assistant/operating-system #894: Beta 5.2 - can't see update notification for 5.3

FaserF, 2020-10-11:
**HassOS release with the issue:**

5.2

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
I am on the beta version 5.2 with my raspberry pi 4. But I think I am unable to update to version 5.3 as I don't receive a notification to update to the new release. But I have a notification to "update" to version 4.13
So it looks like something is wrong with the update detection.

And how should I proceed? Do I have to reinstall the whole os to get the latest version?

agners, 2020-10-11:
The release is still int the process of being built. Takes a while until it is completed and uploaded.

FaserF, 2020-10-12:
Okay thanks but now 10 hours later there still is no notification for 5.3
And I still have the "update" to 4.13 - which is a bug, isn't it?

dracosplace, 2020-10-12:
It jumped from 22 assets to 26 not long ago. 
I'd check back in a few more hours.

FaserF, 2020-10-12:
[image]
Stays the same

closed 2020-10-12 by pvizeli

FaserF, 2020-10-12:
I am sorry if I was too impatient, but why was the issue closed?
The problem still exists, I am not sure how to proceed, and my update notification is wrong.
@pvizeli 

And as I can see the builds are finished and generated, so I should have recieved a correct update notification by now if there was no bug?

ludeeus, 2020-10-12:
Notifications are fetched from <https://version.home-assistant.io/beta.json> which shows 4.x for beta.
dev has 5.x builds <https://version.home-assistant.io/dev.json>

Even if builds are done, they need to be checked before they are promoted to their channel.

FaserF, 2020-10-12:
okay got it, thanks. I wasnt aware of that. Then I will wait for the check to get done
And sorry for my impatience.

moto3001, 2020-10-13:
To piggyback on ludeeus, you can find the stable releases here:
https://version.home-assistant.io/stable.json

FaserF, 2020-10-14:
I have now updated to Version 5.3 via CLI:
ha os update --version 5.3

Thanks to https://github.com/home-assistant/operating-system/issues/746#issuecomment-708183768