home-assistant/operating-system #896: systemd-time-wait-sync.service gets stuck sometimes

tmm1, 2020-10-12:
I'm not sure if this is a systemd bug or kernel issue, but on RPI4 after a reboot I often notice that `systemd-time-wait-sync.service` has become stuck:

```
# systemctl list-jobs
JOB UNIT                                 TYPE  STATE
 69 e2scrub_all.timer                    start waiting
 38 time-sync.target                     start waiting
 67 timers.target                        start waiting
 79 systemd-update-utmp-runlevel.service start waiting
  1 multi-user.target                    start waiting
 37 systemd-time-wait-sync.service       start running
```

```
# systemctl status systemd-time-wait-sync.service
Warning: The unit file, source configuration file or drop-ins of systemd-time-wait-sync.service changed on disk. Run 'sys>
● systemd-time-wait-sync.service - Wait Until Kernel Time Synchronized
     Loaded: loaded (/usr/lib/systemd/system/systemd-time-wait-sync.service; enabled; vendor preset: enabled)
    Drop-In: /etc/systemd/system/systemd-time-wait-sync.service.d
             └─hassos.conf
     Active: activating (start) since Fri 2020-02-07 15:50:52 UTC; 8 months 4 days ago
       Docs: man:systemd-time-wait-sync.service(8)
   Main PID: 117 (systemd-time-wa)
      Tasks: 1 (limit: 2148)
     Memory: 440.0K
     CGroup: /system.slice/systemd-time-wait-sync.service
             └─117 /usr/lib/systemd/systemd-time-wait-sync

Feb 07 15:50:53 server systemd-time-wait-sync[117]: adjtime state 5 status 40 time Fri 2020-02-07 15:50:52.918974 UTC
Oct 12 19:37:06 server systemd-time-wait-sync[117]: adjtime state 5 status 40 time Mon 2020-10-12 19:37:06.002184 UTC
Oct 12 19:38:23 server systemd-time-wait-sync[117]: adjtime state 5 status 40 time Mon 2020-10-12 19:38:23.754433 UTC
Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.
```

This happens even though the `systemd-timesyncd.service` is activated, and even though `/run/systemd/timesync/synchronized` exists:

```
# systemctl status systemd-timesyncd
Warning: The unit file, source configuration file or drop-ins of systemd-timesyncd.service changed on disk. Run 'systemct>
● systemd-timesyncd.service - Network Time Synchronization
     Loaded: loaded (/usr/lib/systemd/system/systemd-timesyncd.service; enabled; vendor preset: enabled)
    Drop-In: /etc/systemd/system/systemd-timesyncd.service.d
             └─hassos.conf, ro.conf
     Active: active (running) since Mon 2020-10-12 19:37:06 UTC; 26min ago
       Docs: man:systemd-timesyncd.service(8)
   Main PID: 252 (systemd-timesyn)
     Status: "Initial synchronization to time server 162.159.200.123:123 (time.cloudflare.com)."
      Tasks: 2 (limit: 2148)
     Memory: 1.2M
     CGroup: /system.slice/systemd-timesyncd.service
             └─252 /usr/lib/systemd/systemd-timesyncd

Feb 07 15:50:55 server systemd[1]: Starting Network Time Synchronization...
Feb 07 15:50:56 server systemd-timesyncd[252]: System clock time unset or jumped backwards, restoring from recorded times>
Oct 12 19:37:06 server systemd[1]: Started Network Time Synchronization.
Oct 12 19:37:06 server systemd-timesyncd[252]: Network configuration changed, trying to establish connection.
Oct 12 19:37:06 server systemd-timesyncd[252]: Network configuration changed, trying to establish connection.
Oct 12 19:38:23 server systemd-timesyncd[252]: Initial synchronization to time server 162.159.200.123:123 (time.cloudflar>
```

```
# ls -alh /run/systemd/timesync/synchronized
-rw-r--r--    1 systemd- systemd-       0 Oct 12 19:54 /run/systemd/timesync/synchronized
```

Further we can see that `systemd-time-wait-sync` is stuck waiting for events:

```
# lsof -nPp 117
COMMAND   PID USER   FD      TYPE             DEVICE SIZE/OFF  NODE NAME
systemd-t 117 root  cwd       DIR                8,3      238  7333 /
systemd-t 117 root  rtd       DIR                8,3      238  7333 /
systemd-t 117 root  txt       REG                8,3    14104  3808 /usr/lib/systemd/systemd-time-wait-sync
systemd-t 117 root  mem       REG                8,3    92368  1512 /usr/lib/libz.so.1.2.11
systemd-t 117 root  mem       REG                8,3   129824  1328 /usr/lib/libgpg-error.so.0.28.0
systemd-t 117 root  mem       REG                8,3   155792  1459 /usr/lib/libpthread-2.30.so
systemd-t 117 root  mem       REG                8,3   628680  1373 /usr/lib/libm-2.30.so
systemd-t 117 root  mem       REG                8,3    31352  1467 /usr/lib/librt-2.30.so
systemd-t 117 root  mem       REG                8,3   374464  1383 /usr/lib/libmount.so.1.1.0
systemd-t 117 root  mem       REG                8,3    88696  1369 /usr/lib/libkmod.so.2.3.4
systemd-t 117 root  mem       REG                8,3    31224  1340 /usr/lib/libip4tc.so.2.0.0
systemd-t 117 root  mem       REG                8,3   773704  1301 /usr/lib/libgcrypt.so.20.2.5
systemd-t 117 root  mem       REG                8,3    22856  1236 /usr/lib/libcap.so.2.27
systemd-t 117 root  mem       REG                8,3   324232  1227 /usr/lib/libblkid.so.1.1.0
systemd-t 117 root  mem       REG                8,3  1413888  1232 /usr/lib/libc-2.30.so
systemd-t 117 root  mem       REG                8,3  2220416  3496 /usr/lib/systemd/libsystemd-shared-244.so
systemd-t 117 root  mem       REG                8,3   159440  1199 /usr/lib/ld-2.30.so
systemd-t 117 root    0r      CHR                1,3      0t0  5418 /dev/null
systemd-t 117 root    1u     unix 0x0000000057e493d7      0t0 10285 type=STREAM
systemd-t 117 root    2u     unix 0x0000000057e493d7      0t0 10285 type=STREAM
systemd-t 117 root    3u  a_inode               0,12        0  5352 [eventpoll]
systemd-t 117 root    4u  a_inode               0,12        0  5352 [signalfd]
systemd-t 117 root    5r  a_inode               0,12        0  5352 inotify
systemd-t 117 root    6u  a_inode               0,12        0  5352 [timerfd]

# strace -ttTp 117
strace: Process 117 attached
20:08:13.919924 epoll_pwait(3,
```

Internally, `systemd-time-sync-wait` will either wait for a clock event from the kernel (which is known to be unreliable), or for an inotify event for `/run/systemd/timesync/synchronized`. However it seems the latter is not working, even if I `touch` the file again or delete and re-create it.

```
# strace -ttTp 117 &
# touch /run/systemd/timesync/synchronized
strace: Process 117 attached
20:11:47.916173 epoll_pwait(3, 
```

Interestingly, if I delete and re-create the the directory and then the file it gets an event but still doesn't activate:

```
# rm /run/systemd/timesync/synchronized ; rmdir /run/systemd/timesync ; mkdir /run/systemd/timesync ; touch /run/systemd/timesync/synchronized
[{EPOLLIN, {u32=907074160, u64=907074160}}], 4, -1, NULL, 8) = 1 <40.990304>
20:13:36.995021 clock_gettime(CLOCK_BOOTTIME, {tv_sec=2149, tv_nsec=284383941}) = 0 <0.000026>
20:13:36.995165 read(5, "\1\0\0\0\0\1\0@\0\0\0\0\20\0\0\0timesync\0\0\0\0\0\0\0\0", 272) = 32 <0.000028>
20:13:36.995273 epoll_pwait(3,
```

If we look at the inotify state of the daemon, it looks like only one watcher is setup:

```
# cat /proc/117/fdinfo/5
pos:	0
flags:	02004000
mnt_id:	12
inotify wd:1 ino:2014 sdev:15 mask:100 ignored_mask:0 fhandle-bytes:c fhandle-type:1 f_handle:824d969c1420000000000000
```

Looking at `mnt_id: 12` in mountinfo shows it is attached to the `/run` filesystem:

```
# cat /proc/117/mountinfo  | grep :12
26 19 0:21 / /run rw,nosuid,nodev shared:12 - tmpfs tmpfs rw,mode=755
40 19 0:21 /machine-id /etc/machine-id ro shared:12 - tmpfs tmpfs rw,mode=755
```

On the systemd project there are some references to similar problems. So it is possible this is a systemd bug, but the check for `/run/systemd/timesync/synchronized` was specifically added to combat this and I think maybe there is some bad interaction with inotify or the kernel version which is to blame.

https://github.com/systemd/systemd/issues/8683
https://github.com/systemd/systemd/issues/15947

tmm1, 2020-10-12:
Looking through the [systemd `time-wait-sync.c` source](https://github.com/systemd/systemd/blob/4b6f74f5a0943e0abfa8e6997811f8f7b7f00a15/src/time-wait-sync/time-wait-sync.c), we see there are two paths which are watched by inotify:

>        r = inotify_add_watch_and_warn(state.inotify_fd, "/run/systemd/", IN_CREATE);

>        sp->run_systemd_timesync_wd = inotify_add_watch(sp->inotify_fd, "/run/systemd/timesync", IN_CREATE|IN_DELETE_SELF);

From fdinfo output above, we know that only one inotify watch descriptor exists. This appears to be the one for `/run/systemd`, which can be verified via strace:

```
# strace -ttTp 117 2>&1 | grep foobar &
# touch /run/systemd/foobar
20:31:01.790573 read(5, "\1\0\0\0\0\1\0\0\0\0\0\0\20\0\0\0foobar\0\0\0\0\0\0\0\0\0\0", 272) = 32 <0.000064>
```

So it appears the root cause is that the inotify watch descriptor for the `/run/systemd/timesync` directory is never registered successfully, possibly because the `systemd-time-wait-sync` service is started before that directory exists.

tmm1, 2020-10-12:
> So it appears the root cause is that the inotify watch descriptor for the `/run/systemd/timesync` directory is never registered successfully, possibly because the `systemd-time-wait-sync` service is started before that directory exists.

I patched my local copy with systemd/systemd#17327 and looks like my hypothesis was correct:

```
# systemctl status systemd-time-wait-sync
...
Feb 07 15:50:53 server systemd-time-wait-sync[113]: Failed to add a watch for /run/systemd/timesync: No such file or directory
Feb 07 15:50:53 server systemd-time-wait-sync[113]: adjtime state 5 status 40 time Fri 2020-02-07 15:50:52.974432 UTC
Oct 12 21:23:42 server systemd-time-wait-sync[113]: adjtime state 5 status 40 time Mon 2020-10-12 21:23:42.000139 UTC
```

What's strange is when I delete and recreate the timesync directory, even though the first watch fires the second watch is not added nor is a new error printed:

```
# rm /run/systemd/timesync/synchronized ; rmdir /run/systemd/timesync ; mkdir /run/systemd/timesync ; touch /run/systemd/timesync/synchronized

strace: Process 114 attached
21:28:32.064674 epoll_pwait(3, [{EPOLLIN, {u32=835775088, u64=835775088}}], 4, -1, NULL, 8) = 1 <2.602661>
21:28:34.668149 clock_gettime(CLOCK_BOOTTIME, {tv_sec=89, tv_nsec=767409552}) = 0 <0.000161>
21:28:34.668571 read(5, "\1\0\0\0\0\1\0@\0\0\0\0\20\0\0\0timesync\0\0\0\0\0\0\0\0", 272) = 32 <0.000147>
21:28:34.668987 epoll_pwait(3,
```

I don't really understand how this is possible. When an inotify event arrives on the first watch descriptor, it should be trying to add the second watch descriptor again [here](https://github.com/systemd/systemd/blob/master/src/time-wait-sync/time-wait-sync.c#L68-L70):

```
static void process_inotify_event(sd_event *event, ClockState *sp, struct inotify_event *e) {
        if (e->wd == sp->run_systemd_wd) {
                /* Only thing we care about is seeing if we can start watching /run/systemd/timesync. */
                if (sp->run_systemd_timesync_wd < 0)
                        update_notify_run_systemd_timesync(sp);
```

I suppose it is possible that `run_systemd_timesync_wd` is non-zero, but how could that be the case when we see the earlier failure due to ENOENT in the logs.

agners, 2020-10-12:
Nice tear-down :-)

I remember looking into this (or similar) as part of #843, I think essentially the issue was that the data partition (/mnt/data) was not mounted properly, which lead to a whole bunch of things going wrong, but typically lead to systemd being stuck on the time sync service. But this was on ODROID N2+, where resizing behaved erratic (fixed with #848).

Which version of HAOS are you using? What is `systemctl status mnt-data.mount` saying?

tmm1, 2020-10-12:
Looks like the version of systemd we're using is missing https://github.com/systemd/systemd/commit/f6f4f5fe5395a57f10dd446c7266c53f0673eaac, which is definitely not helping.

tmm1, 2020-10-12:
> missing [systemd/systemd@f6f4f5f](https://github.com/systemd/systemd/commit/f6f4f5fe5395a57f10dd446c7266c53f0673eaac)

This kinda explains everything. If `sp->run_systemd_wd` is 0, then it makes sense that `process_inotify_event` will never call `update_notify_run_systemd_timesync` and thus never start watching the timesyncd directory.

I am rebuilding locally with that patch to confirm, but I am pretty sure it will fix everything.

agners, 2020-10-12:
Not sure I followed completely, but is the first watch maybe watching another mount/directory?

agners, 2020-10-12:
> I'm using 5.4 off dev branch. mnt-data is fine in this case.

Ok definitely a different issue then. In that case it also make sense that you get a shell. For the other issue I had to use a rescue shell to debug.

tmm1, 2020-10-12:
> Not sure I followed completely, but is the first watch maybe watching another mount/directory?

Basically the issue was that `inotify_add_watch_and_warn` always returns 0 before systemd v246. This means the [following code](https://github.com/systemd/systemd/blob/4b6f74f5a0943e0abfa8e6997811f8f7b7f00a15/src/time-wait-sync/time-wait-sync.c#L226-L230) doesn't work:

        r = inotify_add_watch_and_warn(state.inotify_fd, "/run/systemd/", IN_CREATE);
        if (r < 0)
                return r;

        state.run_systemd_wd = r;

Since `run_systemd_wd` is set to zero, when the inotify event with the actual wd comes in later it doesn't match:

    static void process_inotify_event(sd_event *event, ClockState *sp, struct inotify_event *e) {
        if (e->wd == sp->run_systemd_wd) {

This is a regression from https://github.com/systemd/systemd/commit/27c3112dcbd1b5f171c36c32550d9c6331375b0b

tmm1, 2020-10-12:
> I am rebuilding locally with that patch to confirm, but I am pretty sure it will fix everything.

Confirmed it works now. You can see in the logs it attempts to add the second watch multiple times, every time `/run/systemd` is modified:

```
Feb 07 15:50:52 server systemd-time-wait-sync[116]: Failed to add a watch for /run/systemd/timesync: No such file or dire>
Feb 07 15:50:52 server systemd-time-wait-sync[116]: adjtime state 5 status 40 time Fri 2020-02-07 15:50:52.550728 UTC
Feb 07 15:50:52 server systemd-time-wait-sync[116]: Failed to add a watch for /run/systemd/timesync: No such file or dire>
Feb 07 15:50:56 server systemd-time-wait-sync[116]: Failed to add a watch for /run/systemd/timesync: No such file or dire>
Feb 07 15:50:56 server systemd-time-wait-sync[116]: Failed to add a watch for /run/systemd/timesync: No such file or dire>
Oct 12 22:01:44 server systemd-time-wait-sync[116]: adjtime state 5 status 40 time Mon 2020-10-12 22:01:44.000094 UTC
Oct 12 22:03:00 server systemd-time-wait-sync[116]: adjtime state 0 status 2000 time Mon 2020-10-12 22:03:00.765735 UTC
Oct 12 22:03:00 server systemd[1]: Started Wait Until Kernel Time Synchronized.
```

agners, 2020-10-12:
> Basically the issue was that inotify_add_watch_and_warn always returns 0 before systemd v246.

Argh I see, ugly. I guess we can just patch our systemd with https://github.com/systemd/systemd/commit/f6f4f5fe5395a57f10dd446c7266c53f0673eaac until (if?) they have a maintenance release.

agners, 2020-10-12:
We probably will update to 246 for rel-5 anyways, need to check what buildroot did there.

Hm, in rel-4 we use 244.3, which I guess is affected too since https://github.com/systemd/systemd/commit/27c3112dcbd1b5f171c36c32550d9c6331375b0b came in with 244-rc1.

tmm1, 2020-10-12:
> We probably will update to 246 for rel-5 anyways, need to check what buildroot did there.

Cool! I am very new to all this so I wasn't sure how often buildroot/systemd are updated from upstream.

closed 2020-10-12 by pvizeli

rr19-hub, 2022-04-22:
I had the exact same problem with the success of new installations not completing on new disks. The log (browser) had a completely wrong date and time. At first I suspected DNS and therefore took out the pi-hole as DNS server. But that did not help. Then I read that for NTP also the **UDP protocol** is needed and that was then the error: in my Fritzbox new devices may only 'surf and mail'. After I deleted this restriction for the new Raspberry, everything worked!

agners, 2022-04-25:
> in my Fritzbox new devices may only 'surf and mail'

:joy:

PR #897 (2020-10-12): Fix systemd time-wait-sync inotify bugs
fixes #896 

cc @agners