home-assistant/operating-system #900: Hassos doesn't boot after update hassos 5.3

hdjweb, 2020-10-15:
**HassOS release with the issue:**

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

hdjweb, 2020-10-15:
Hassos doesn't start after update with terminal to hassos 5.3 i give the command with terminal " ha os update --version 5.3"

SilverWx, 2020-10-15:
Hi,

I never updated hassos through CLI but I think the right command is
`ha os update --version=5.3`
but I don't know if this can work since 5.3 is not on the main branch.
Also, you have to provide more details if you want some help I think.

pvizeli, 2020-10-15:
Please fill out the template, otherwise, I close the issue

SilverWx, 2020-10-15:
Update done on vmware installation flawlessly ;)

agners, 2020-10-21:
Closing due to incomplete information.

closed 2020-10-21 by agners