home-assistant/operating-system #902: USB to TTL CH340C don't run on Zigbee2mqtt Coordinator made by CC3530+CC2592

mrIngrid, 2020-10-16:
**HassOS release with the issue:**
 Rpi4 4gb SSD mSata.
HassOS 5.3 beta relase & Home Assistant 0.116.3

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
Zigbee2mqtt coordinator made from CC2530+CC2592 conected by USB 2.0/3.0 with serial usb to ttl CH340C don't run, but with the old CP2102 run perfectly.
[image]

agners, 2020-10-21:
Does this device work on other Linux distributions (e.g. Ubuntu)? Do you know the USB VID/PID of this (can usually be obtained using `lsusb` on Linux devices).

mrIngrid, 2020-10-21:
> Does this device work on other Linux distributions (e.g. Ubuntu)? Do you know the USB VID/PID of this (can usually be obtained using `lsusb` on Linux devices).

Yes. This device works perfectly on HassOS 4.xx Stable (boot on microSD). When i decide to try the developement version 5.2 on my RPI4 , i see only this inconvenient...

mrIngrid, 2020-10-21:
> Does this device work on other Linux distributions (e.g. Ubuntu)? Do you know the USB VID/PID of this (can usually be obtained using `lsusb` on Linux devices).

[image]

riccionese, 2020-10-23:
I've the same issue. Raspberry pi4 + SSD +CC3530+CC2592 and zigbee2mqtt doesn't work

diego88-ctrl, 2020-10-23:
Hi guys. Exactly the same problem here.

agners, 2020-10-26:
I do have a CH340C based device as well, the zzh! CC2652R Multiprotocol RF Stick. It seems to be detected by HAOS fine:
```
...
[  378.317583] usb 1-1.3: new full-speed USB device number 6 using xhci_hcd
[  378.435840] usb 1-1.3: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.63
[  378.451115] usb 1-1.3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[  378.465151] usb 1-1.3: Product: USB2.0-Serial
[  378.480643] ch341 1-1.3:1.0: ch341-uart converter detected
[  378.497352] usb 1-1.3: ch341-uart converter now attached to ttyUSB0
....

```
# lsusb
Bus 002 Device 004: ID 0bda:9210
Bus 002 Device 002: ID 0bda:0411
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 004: ID 1a86:7523
Bus 001 Device 003: ID 0bda:5411
Bus 001 Device 002: ID 2109:3431
Bus 002 Device 003: ID 0bda:0411
Bus 002 Device 001: ID 1d6b:0003
Bus 001 Device 005: ID 0bda:5411
```

Is this maybe an Add-on problem?

(I do have the SSD behind a powered Hub though)

mrIngrid, 2020-10-27:
With the latest bootloader & firmware updates of the RPI, the coordinator is seen correctly by the OS, but it doesn't work well.  Here I opened another issue about it. https://github.com/danielwelch/hassio-zigbee2mqtt/issues/455

ykiv63, 2020-10-27:
I've the same issue. Raspberry pi4 + SSD +CC3530+CC2592 and zigbee2mqtt doesn't work

ciotlosm, 2020-11-01:
I have HassOS 5.4 with latest bootloader for rpi4 from September. I'm using CC2652R (zzh!) and I have m.2 ssd as boot storage (no sd card connected). Everything works correctly. I do have: usb extender for CC2652R and the m2 ssd is connected also using a longer cable so devices are not close to each other.

agners, 2020-11-02:
This seems not to be a OS issue as the CC2652R with CH340C works for me and @ciotlosm.

closed 2020-11-02 by agners