home-assistant/operating-system #903: ProxMox 6.2 and Home Assistant VM not working

Joebuckeye123, 2020-10-16:
**HassOS release with the issue:**

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
When it was announced that HA supervised was going to be deprecated and HassOS was the main platform, I decided to change my setup.  I saw JuanMTech video on ProxMox VM and decided to go in this direction.  ProxMox and HassOS was working great for about a month.  However, one day I restarted the Home Assistant VM and it would not run.  ProxMox 6.2 will not allow Home Assistant VM to work.  I have tried using QCOW2 and Intel NUC images and had the same result.
In reviewing the comments at https://community.home-assistant.io/ , there was a similar issue with ProxMox 6.2 and HassOS 4.8 reported by woody4165 on 6/5/20.  I didn’t see notes in this Blog that problem was being corrected.
I really liked ProxMox for its ability to schedule automatic backup images to USB.  My option is to install Ubuntu 18.04 and HA supervised.  My hope is that Home Assistant is already working on this problem with ProxMox 6.2-4 and HassOS.
[image]

Chili-Man, 2020-10-16:
@Joebuckeye123 I ran into a similar issue trying to run the  QCOW2 image through `virt-install` instead of proxmox; I'm not sure if this the case with you, but problem was that I had forgotten to `gunzip hassos_intel-nuc-4.14.img.gz` remove the compression :disappointed:  which was causing the same thing you're seeing. Hope that helps

Joebuckeye123, 2020-10-17:
Thank you very much Chili-Man.  I downloaded 7-zip file manager and extracted the .gz file.  Home Assistant is in process of building.   Hope this helps others that run into this problem.

closed 2020-10-17 by Joebuckeye123