home-assistant/operating-system #906: Update dropbear package, support ed25519 keys & rsa-sha2 signatures

paravoid, 2020-10-17:
Dropbear is currently at version 2019.78. Current upstream version is 2020.80, which upstream buildroot seems to have bumped the package for: https://github.com/buildroot/buildroot/commit/87c3f71c5d0107cb5e01a3d23b5350193a87d17b

Most notably, version 2020.79 [added support](https://matt.ucc.asn.au/dropbear/CHANGES) for ed25519 keys, as well as rsa-sha2 signatures:
> - Support ed25519 hostkeys and authorized_keys, many thanks to Vladislav Grishenko. This also replaces curve25519 with a TweetNaCl implementation that reduces code size.
>
> - Support using rsa-sha2 signatures. No changes are needed to hostkeys/authorized_keys entries, existing RSA keys can be used with the new signature format (signatures are ephemeral within a session). Old ssh-rsa signatures will no longer be supported by OpenSSH in future so upgrading is recommended.

These are important, among other reasons because OpenSSH will ["in the near-future" disable](https://www.openssh.com/txt/release-8.4) the "ssh-rsa" signature algorithm:
> It is now possible[1] to perform chosen-prefix attacks against the SHA-1 algorithm for less than USD$50K. For this reason, we will be disabling the "ssh-rsa" public key signature algorithm by default in a near-future release.

agners, 2020-11-02:
For `rel-5` we plan to bump to a newer buildroot release, so this will be taken care of soon.

paravoid, 2020-12-14:
This has been fixed with the recent OS Release 5 indeed. Thanks!

closed 2020-12-14 by paravoid