home-assistant/operating-system #908: Firmware for Realtek Bluetooth 5.0 dongle not available

embcla, 2020-10-17:
**HassOS release with the issue:**

4.13

**Supervisor logs:**

none applicable

**Journal logs:**

**Kernel logs:**

[image]

**Description of problem:**
Drivers for the stick are not available, the entire folder /lib/firmware/rtl_bt is missing (check here for the relative debian package https://reposcope.com/package/firmware-realtek )

pvizeli, 2020-10-17:
This is for bugs only. Feature and enhancement requests should go in our community forum: https://community.home-assistant.io/c/feature-requests

closed 2020-10-17 by pvizeli

vBrolin, 2020-10-27:
This is not a feature request, these devices were working before 4.13, then it broke, not sure why the folder is missing but it was probably there before since it was working just fine before that upgrade.